helgeerbe/picframe #24: Make setup more user friendly

helgeerbe, 2021-02-26:
Extend the -i option to configure `configuration.yaml`properly.
E. g. 

- set the correct paths.
- ask for locale

- offer to copy to copy 'picframe.service' to `~/.config/systemd/user` and enable it on boot time.

...

helgeerbe, 2021-02-28:
Adding a quick hack for -i option
Creates `configuration.yaml` 
- set all paths to selected picframe_data directory
- ask for pic dir
- ask for deleted pic dir
- reads current locale and ask for it

Hope this makes setup more easy. Later on, we should move config to the database an make setting persistent.

jgodfrey, 2021-02-28:
Sorry to be late on the requested PR review here... Minor, but I might suggest that the `start.py` script mention where the results of the process are stored (so, point to the `configuration.yaml` file).

closed 2021-02-28 by helgeerbe

helgeerbe, 2021-02-28:
It das a `print("This will configure ", destination)` 
destination is the full path

jgodfrey, 2021-02-28:
Yeah, I see that now. Cool - thanks.