helgeerbe/picframe #25: Enable display on/of on other plattforms

helgeerbe, 2021-02-26:
Use `xset`on other plattforms to turn display on/of.

helgeerbe, 2021-03-03:
@paddywwoof 

I'm using pi3d platform detection. But `pi3d.PLATFORM_LINUX` is unknown, although PLATFORM=3, this is the definition of the linux platform in Pi3d.

https://github.com/helgeerbe/picframe/compare/dev...display_switch

Can you have a closer look?

paddywwoof, 2021-03-03:
Helge, all the classes and properties accessible by ```pi3d.``` are fixed up in [pi3d/__init__.py](https://github.com/tipam/pi3d/blob/master/pi3d/__init__.py) and that doesn't look to include PLATFORM_LINUX for some reason (bug by any other name, which I will fix!) **Good spot**

To get at it you can use ```pi3d.constants.PLATFORM_LINUX```

paddywwoof, 2021-03-03:
On the raspberry pi 4 pi3d will be using the linux setup because it can't run the broadcom legacy drivers. That's why it has to have X running for the display surface. Not sure if that's relevant for this on/off switching

paddywwoof, 2021-03-03:
Maybe the most pythonic (whatever that means) way here is to not bother with the platform detection but do a couple of try: except: blocks, first one for vcgencmd except try xset except give up!

helgeerbe, 2021-03-03:
Testing would take time, because I always have to do subprocess calls. Maybe I will move this part into viewer setup. So it has to be done only one time at start up. On swicht command, I will test only a constant: 
Display-Switch:
- vcgencmd - only available on raspbian (but works without X)
- dpms - the linux standard way, but requires X and dpms enabled
- unsupported

paddywwoof, 2021-03-04:
Helge, I will try patching the revision onto the code I have running on the RPi4. It will be interesting to see how it fares (i.e. it will think it's a linux setup which could well be fine). What I meant about the try catch was that you're already doing try catch so the architecture testing could just be taken out. Something like
```python
    def display_is_on(self):
            try: # vcgencmd only applies to raspberry pi
                state = str(subprocess.check_output(["vcgencmd", "display_power"]))
                if (state.find("display_power=1") != -1):
                    return True
                else:
                    return False
            except Exception as e1:
                try: # try xset on linux, DPMS has to be enabled
                    output = subprocess.check_output(["xset" , "-display", ":0", "-q"])
                    if output.find(b'Monitor is On') != -1:
                        return True
                    else:
                        return False
                except Exception as e2:
                    #### log neither neither work e1 e2      
```

helgeerbe, 2021-03-04:
I see, but I did something similar. Instead of nesting the exceptions. I ignore them, an do the next try block.

The first thing I try is to call "vcgencmd". This is only available on raspbian. (Hope it works as expected on pi 4). 
Second is to run "xset" which requires that X is started with dpms on. I saw in Wolfgang's guides, that he turned it explicitly off, for the pi4 setup. But I hope, tun screen saver off, and no blanc screen should be good enough.

paddywwoof, 2021-03-04:
But because of your architecture testing the vcgencmd block will be skipped on pi 4 (as it's running under X). That's what I mean by just doing the try: you're not really interested in what the architecture is just whether the vcgencmd works, and if it doesn't will the xset work.

paddywwoof, 2021-03-04:
No it doesn't turn off. Will try the 'try' only method and let you know.

paddywwoof, 2021-03-04:
Yes that works OK. I know it's horrible if you're used to avoiding 'intentionally' triggering exceptions or catching them in a controlled way, but it seems to be the standard python system!
```python
    @property
    def display_is_on(self):
        try: # vcgencmd only applies to raspberry pi
            state = str(subprocess.check_output(["vcgencmd", "display_power"]))
            if (state.find("display_power=1") != -1):
                return True
            else:
                return False
        except Exception as e1:
            try: # try xset on linux, DPMS has to be enabled
                output = subprocess.check_output(["xset" , "-display", ":0", "-q"])
                if output.find(b'Monitor is On') != -1:
                    return True
                else:
                    return False
            except Exception as e2:
                self.__logger.warning("Display ON/OFF is not supported for this platform.")
        return True

    @display_is_on.setter
    def display_is_on(self, on_off):
        try: # vcgencmd only applies to raspberry pi
            if on_off == True:
                subprocess.call(["vcgencmd", "display_power", "1"])
            else:
                subprocess.call(["vcgencmd", "display_power", "0"])
        except Exception as e:
            try: # try xset on linux, DPMS has to be enabled
                if on_off == True:
                    subprocess.call(["xset" , "-display", ":0", "dpms", "force", "on"])
                else:
                    subprocess.call(["xset" , "-display", ":0", "dpms", "force", "off"])
            except Exception as e:
                self.__logger.warning("Display ON/OFF is not supported for this platform.")
```

helgeerbe, 2021-03-04:
Oh, there seems to be a bug on pi 4 https://www.raspberrypi.org/forums/viewtopic.php?t=243981

Please test, if display is turned off, with xset  on pi4. I tested it with pi 3 running, X, Ubuntu laptop, and Ubuntu pi 3. If it is working, we should test for xset first. And then for vcgencmd.

helgeerbe, 2021-03-04:
Does that means, "vcgencmd" it throwing an exception? I expected, it returns OK, but ignoring the command. But when your code is working, I'm fine with it.
Do you take care of the change?

paddywwoof, 2021-03-04:
```vcgencmd display_power 0``` runs without exception. OK I will tidy up code that works and push back to the branch

closed 2021-03-04 by helgeerbe