helgeerbe/picframe #27: Enhance filter to use logical AND(&&) and OR (||)

helgeerbe, 2021-02-26:


jgodfrey, 2021-02-28:
The hard part here, I think, is in providing a mechanism to allow _AND_ and _OR_ logic to be constructed interactively. Has anyone given thought to the interaction here? That is, how would you expect a user to define a complex filter?

sapnho, 2021-03-21:
This would indeed be very nice. With the integration in Home Assistant, you can just enter the keywords easily. If we could add "AND" or "OR" this would be a great feature!

sapnho, 2021-03-27:
> The hard part here, I think, is in providing a mechanism to allow _AND_ and _OR_ logic to be constructed interactively. Has anyone given thought to the interaction here? That is, what mechanism will we need to provide to allow a user to define a complex filter?

The "+" sign in the Home Assistant filter wouldn't work as an approach?

helgeerbe, 2021-03-27:
I would keep it simple and using just simple AND and OR conditions: 
When I remember right, we are using internally the like operator? 
So entering something like "san AND fran" would result in
 `SELECT * FROM table
WHERE address LIKE '%san%' and address LIKE '%fran%';`
And 'San Francisco' would be found. But not 'San Cristóbal'

Example for Or:
"londo OR berl"
`SELECT * FROM table
WHERE address LIKE '%londo%' or address LIKE '%berl%';`
would result in "Berlin" and "London"

paddywwoof, 2021-03-27:
At the moment there is a dictionary of 'where clauses' that are joined together with `AND` and used to select from the database. For each kind of filter (location, tags, etc) there is just one entry. It would be possible to construct more complicated clauses but maybe they should be just 'freestyle' i.e. in home assistant the user would create
```SQL
location LIKE '%france%' AND (tags LIKE '%boat%' OR tags LIKE '%sail%')
```
The `LIKE` and `'%%'` parts could be left off and added back in picframe. But I think field name and the AND, OR and (critically) braces would need to be entered by the user
```txt
location france AND (tags boat OR tags sail)
```
Obviously this will produce **completely** different results from
```txt
location france AND tags boat OR tags sail
```

helgeerbe, 2021-03-27:
In home assistant you have two query fields for the search. I would combine them always with AND
````sql
(location) AND (tags)
````

paddywwoof, 2021-03-27:
At the moment picframe Model holds the info as `{'location':'england', 'tags':'rachel'}` which gets made into the query
```SQL
location LIKE '%england%' AND tags LIKE '%rachel%'
```
I think attempting anything more complicated than single terms ANDed together will be very difficult. However if we had an additional freestyle entry as shown above the dictionary might be
```python
{'location':'england', 'freestyle':'(location bradford and tags rachel) or (location sheffield and (tags exhibition or tags rachel))'}
```
which would generate the SQL
```SQL
location LIKE '%england%' AND ((location LIKE '%bradford%' AND tags LIKE '%rachel%') OR (location LIKE '%sheffield%' AND (tags LIKE '%exhibition%' OR tags LIKE '%rachel%'))
```
I suppose the simpler but less flexible option would be to just allow the clause to apply to `location` and a different one for `tags` so
```python
{'location':'england and (bradford or sheffield)', 'tags':'rachel or exhibition'}
```
which would generate SQL
```SQL
(location LIKE '%england%' AND (location LIKE '%bradford%'  OR location LIKE '%sheffield%')) AND (tags LIKE '%rachel%' OR tags LIKE '%exhibition%' )
```
Which is different but will probably generate similar results  most of the time.

helgeerbe, 2021-03-27:
Yes, the last one was what I was thinking of. I'm fine with that.

paddywwoof, 2021-03-27:
I will try out some `( AND OR` parsing later today!

paddywwoof, 2021-03-27:
This simple parser seems to work
```python
    def __build_filter(self, val, field):
        if val.count("(") != val.count(")"):
            return None # this should clear the filter and not raise an error
        leave = ("(", ")", "AND", "OR")
        val_split = val.replace("(", " ( ").replace(")", " ) ").split() # so brackets split from words
        filter = [(
                s if s.upper() in leave else "{} LIKE '%{}%'".format(field, s)
            ) for s in val_split]
        return " ".join(filter)

```
However the slq in image_cache is not in a try block so it's quite easy to make the app crash by entering something like `england and priestthorpe road` or `england and not bradford` which produce malformed SQL
```sql
 `location LIKE '%england%' and location LIKE '%priestthorpe%' location LIKE '%road%'
 `location LIKE '%england%' and location LIKE '%not%' location LIKE '%priestthorpe%'
```
I think the rule is that, ignoring brackets, there must be an 'and' or 'or' in all the odd numbered places and non of the even positions i.e.
`x or y and (z or w and p) or (u or v and q)` => `x or y and z or w and p or u or v and q` which is OK but
`x or y and not (z or w road and p) not (u or v and q)` => `x or y and not z or w road and p not u or v and q` which fails.

The users need to know that they can't put in NOTs and multi word search strings.

jgodfrey, 2021-03-27:
Since I don't have a Home Assistant instance running (maybe I should do that sometime soon), I'm having a hard time visualizing how/where these SQL-like statements are created by a user?

paddywwoof, 2021-03-27:
I'm sending them using the http interface. (Which needs updating for tags filters, I'm only testing location) Using mqtt the topic is device_id/tags_filter or location_filter with payload string.

paddywwoof, 2021-03-27:
In my opinion the easiest way to control picframe is http

sapnho, 2021-03-28:
> In my opinion the easiest way to control picframe is HTTP

@paddywwoof Is there already a wiki on how this works?

paddywwoof, 2021-03-28:
Wolfgang, I don't think there is. I will put something together but it will be little more than:
```text
in a browser (safari, firefox, edge, brave, opera even chrome) enter in the address bar `IP:port` of your Raspberry Pi. i.e.
192.168.0.235:9000
You will have found the IP for using ssh and the port number can be changed if needed in the configuration.yaml file
(default is 9000)

It is a super simple web page with no sliders, validation etc. 
sometimes it looses track of the values so you have to refresh the page.
```

sapnho, 2021-03-28:
Ah, I didn't know that this exists! Thanks!

Location Filter didn't work for me but it's a nice simple interface for people who don't want to bother with MQTT.

[image]

paddywwoof, 2021-03-28:
Pushed some changes to a new branch 4f40d66f4e81a6cca7a42bd581050cd379afe687 if anyone wants to try this. I've run tests on this laptop but not really tried it on the RPi yet.

sapnho, 2021-03-28:
@paddywwoof Happy to test it if you can let me know what commands to use to switch onto your branch.

paddywwoof, 2021-03-28:
Wolfgang, If you've followed the official instructions then `picframe` will be properly installed in `/home/pi/.local/lib/python3.7/site-packages/picframe` so it might be a bit "hacky" but the easiest way to test a small change to a single file (like this) is to just rename the existing `controller.py` to `controller_bkp.py` and copy into the folder the modified version.

I can do this over the SMB network very easily: On my laptop I already have copy of the git repo, but it would be the same on the RPi via ssh, open a terminal, cd to wherever you cloned the repo then pull the latest changes and checkout the enhanced_filter branch. git magic will then mean the version of the file you copy from this folder will be modified one. So with the original cloning at the beginning
```shell
$ cd ~ # or wherever you keep development stuff
$ git clone https://github.com/helgeerbe/picframe.git
$ cd picframe
$ git pull
$ git checkout enhance_filter
```
Changing an installed package in this way isn't really recommended but so long as you make sure you have a copy and replace file afterwards it's OK. If it all goes wrong you can just pip re-install.

paddywwoof, 2021-03-28:
PS I put an additional `commit` on this branch that allows use of NOT i.e. you can do
```
france or spain or (germany and not berlin)
```

helgeerbe, 2021-03-28:
Works good. 
Just found out that you have to search for 

'New York'

````
new and york
````

The space seems to be a wildcard and all images are found.

sapnho, 2021-03-28:
Thanks, @paddywwoof. Exchanging files is no problem but there is no picframe directory  in my `/home/pi/.local/lib/python3.7/site-packages/`
[image]

helgeerbe, 2021-03-28:
Did you the installation as root? Check `/usr/lib/python3.7/site-packages`

sapnho, 2021-03-28:
Yes, it's a Pi4. However, in Samba I can only access the user directory. I'll wait then until you deploy it to prod.

paddywwoof, 2021-03-28:
Ah Pi4. All different (annoyingly)

The function basically assumes single words, ands, ors and brackets and it checks it complies. So 'new york' fails and clears the filter. I think it could work the other way round and group everything between any of those tokens into one string. Might end up more than a single list comprehension line...

I will think a bit more about it

paddywwoof, 2021-03-28:
This bd6f86f8a227d6b85061fc76628a0e9cdb9008b5 should cope with spaces in things you want to search for. It's still possible to enter malformed search requests such as `berlin not and not` but I've put a `try: except:` block 8ca01c8896a99cff9e514ebe888214c4474976d3 for, essentially, the whole method in ImageCache.query_cache

helgeerbe, 2021-04-15:
Guess, we are done here?!

closed 2021-04-15 by helgeerbe