helgeerbe/picframe #28: Add improved support for auto-matting on non-HD resolutions

jgodfrey, 2021-02-26:
Currently, the auto-matting capabilities of PictureFrame have been designed for HD resolutions. While the matting will work for non-HD resolutions, it won't look exactly as intended. For example, on a 4k display, the matting decorations (bevels, shadows, ...) will be narrower than intended.

The fix here will involve a number of things, including:
 * Provide 4k versions of the ninepatch files used to create the decorations
 * Provide a mechanism to access the correct set of ninepatch files.
 * Calculate an internal "modifier" that can be used to auto-adjust appropriate matting properties based on the current resolution.

jgodfrey, 2021-02-28:
Has anyone viewed the auto-mat results on a 4K display? While I can imagine what they look like (generally, too _narrow_ I assume), I've actually never seen them. I do have a 4K TV I could hook up to I guess - it's just not in a convenient location for dev work...

jgodfrey, 2021-03-09:
Still curious here - has anyone tried the matting on a 4k display. As described above, I can imagine what it'll look like, but I've yet to see it.

paddywwoof, 2021-03-09:
I too don't have a 4k screen.. but you might be able to specify the dimensions in the config file wider than the screen. I will try that out now on the RPi4

jgodfrey, 2021-03-09:
I *do* have a 4k tv I could hook up to. So far, I've just been too lazy to do it... :)

closed 2024-08-16 by helgeerbe