helgeerbe/picframe #29: Support accurate text overlay information for portrait pairs

jgodfrey, 2021-02-26:
Currently, the text overlay features of the application only support a single image. When images are presented as portrait pairs (either matted or unmatted), the text information shown pertains only to the _left-hand_ image in the pair. No information is presented for the _right-hand_ image.

It's trivial to access the text information for both images in this situation. The more difficult part is determining how to best present the information for 2 images. For instance, it could be presented in two separate blocks - one on each side of the screen. Or, it could be presented in multiple lines, with the first line(s) representing the left image and the remaining line(2) representing the right image.

None of this should be hard. We just need to decide on how to best present the info.

paddywwoof, 2021-02-27:
I will have a go at setting up the side by side system as that seems more modular and visually logical.

helgeerbe, 2021-02-27:
Is this for the release? Then we should assign it to the Projects

paddywwoof, 2021-02-27:
Yes. Might as well as it effectively works now. Just needs vertical alignment tidying.

paddywwoof, 2021-02-27:
Vertical alignment trickier than I thought; but easy enough once I'd figured out where the info was stored. I managed to do it without having to modify the pi3d.TextBlock code.

I've pushed this into a new branch https://github.com/helgeerbe/picframe/commit/0ffa97c42cd4afe9f91a8447ae4650fd3b1d64eb as there are quite a few alterations and it's a core file. If anyone with access to a RPi fancies running it we can then merge it and cross it off.

jgodfrey, 2021-02-27:
Hi Paddy,

I just took a quick look at this. Generally, it seems to work as expected. However, it looks like there are still some vertical positioning issues for certain cases. I've seen a number of times where paired images render one of the text blocks higher (by a line or two) than is necessary. 

I haven't yet poked around for a cause...

jgodfrey, 2021-02-27:
Also, I noticed that we're not using the configured text size. Looks like it's hardcoded.

This:

```python
self.__text = pi3d.PointText(font, camera, max_chars=200, point_size=50)
```

Should be:

```python
self.__text = pi3d.PointText(font, camera, max_chars=400, point_size=self.__show_text_sz)
```

paddywwoof, 2021-02-28:
Oops, don't know when the fixed point size crept in!

The char_offset array with the vertex locations for each char in the TextBlock gets reused and the locations of unused characters are not reset to zero (because the characters are not actually going to be drawn). So what I have done is look at the y position of the first ```len(final_string)``` and use the min as the offset. However ```\n``` will be in the final_string but actual drawn characters so it's probably picking up y positions for previous characters. I will try to figure out a way to stop this - but want to avoid having to modify and re-issue pi3d (though obviously if that's the best solution I will do it)

paddywwoof, 2021-02-28:
Hopefully fixed all those issues https://github.com/helgeerbe/picframe/commit/2708e85fa4ee7f1d9fbfe913fc93fb635a4b8a9f. I've not seen any vertically misaligned text testing on this laptop but I will hold off merging in case anyone sees anything odd.

jgodfrey, 2021-02-28:
I haven't seen any additional vertical alignment issues, so I think you've taken care of this one. Thanks!

closed 2021-02-28 by paddywwoof