helgeerbe/picframe #30: Add a schema version record to the database

jgodfrey, 2021-02-26:
Before a public release, we really should add a "schema version" record to the database. While this won't be immediately useful, it will allow us (in the future) to automatically _migrate_ the database from one schema version to another. 

Currently, the ability to auto migrate the data isn't necessary as the database only contains auto-generated cache info. With that, the database can simply be deleted and it will be automatically rebuilt in its entirety. However, at some point, I'd like to use the database to store manually added data (for example, maybe a "favorites" flag). At that point, removing the database will cause a loss of data that can't be automatically added back.

With a schema version stamp in the db, it becomes possible to migrate the data between versions instead of simply deleting it.

jgodfrey, 2021-02-27:
PR for this ticket is here:

https://github.com/helgeerbe/picframe/pull/36

Not sure whether you want me to merge it myself or not. Let me know for future work...

helgeerbe, 2021-02-27:
Jeff, I'm fine if you two merge directly to dev. But make sure, dev is working after merge. That reminds me, to think about some Unit-Test to run after each merge. Will put it on the issue list

helgeerbe, 2021-02-27:
Merged

closed 2021-02-27 by helgeerbe