helgeerbe/picframe #39: Allow auto-matting of only certain images, based on specified aspect ratio tolerance

jgodfrey, 2021-02-27:


helgeerbe, 2021-02-27:
Is this relevant for our first release?

jgodfrey, 2021-02-27:
> Is this relevant for our first release?

I'd say _Yes_. I just marked it as such. Are we up against any particular deadline here?

helgeerbe, 2021-02-27:
Not really, but I see that we have lots of great ideas. And could improve for ever. Guess Wolfgang is eager to publish is article. So I try to sort out, what is important for our first release.

jgodfrey, 2021-02-27:
Sure, that makes sense. Currently, I see most of the open issues on the _to be done before release_ list as some sort of _cleanup_ for features that have been added recently. While they might not be 100% required, some of the features sort of feel unfinished without the changes. And, I don't think anything on that list is very big...

helgeerbe, 2021-02-27:
No, there are actual only 2 issues open. In the meanwhile, I can work on the wiki. 😄

jgodfrey, 2021-02-27:
Just merged these changes.

[#852f2c5](https://github.com/helgeerbe/picframe/commit/852f2c53fc2f4088094c3e10f6f7a59353b1b6be)

closed 2021-02-27 by jgodfrey