helgeerbe/picframe #40: Implement some unit test, which have to run on any merge to dev.

helgeerbe, 2021-02-27:
These test have to ensure, that the dev branch is always green and working. Failed test should result in further commit stops, until the branch is back on green.

paddywwoof, 2021-03-08:
I had hoped to make a kind of test that would run the slideshow with a standard size and save a portion of the screen (using pi3d.masked_screenshot()) For comparison with a standard array of pixels. Unfortunately, in order for masked_screenshot() to work, for some reason, I have to call it within the ```ViewerDisplay.slideshow_is_running()``` method. So that involves a whole lot of code inserted to trigger the capture (i.e. either an additional method argument or a class property to trigger the capture with something in Controller to set that and something in start to set that, also to stop Controller a short time afterwards.) All a bit messy.

But what I have done is make three versions of AlleExif.JPG one 960x600 and two 480x600 which total 300k then a configuration_test.yaml also in the test folder. In the parent folder to ```./test``` I made a little file ```run_test.py```
```python
import os
from picframe import start
os.system("rm ./test/pictureframe.db3")
start.main()
```
i.e. just do a visual check that it seems to be running OK, so I'm not sure it's much better than running on whatever setup we already have.

helgeerbe, 2021-03-09:
Hi Paddy, have a look at https://stackoverflow.com/questions/42397189/dependency-injection-in-unit-test-with-python
Maybe you could use this idea bypass `ViewerDisplay.slideshow_is_running()` and create your own function. That would keep testcode out.

We could think of installing also the unit test folder. So everybody, would be able to run all tests after installation.

paddywwoof, 2021-03-09:
Helge, that's a good idea. i.e. have a testing class inheriting from ViewerDisplay with a modified init and slideshow_is_running. I'll check if it works (i.e. pi3d.masked_screenshot() produces actual pixels), we might regret making *all* the properties private with double under scores! (Of course it's not hard to get round python mangling: by refering to ```v._ViewerDisplay__display.width``` etc)

helgeerbe, 2021-03-09:
Hm, where it makes sense, we could change functions to use only a single underscore. I like the idea to have an 'official' interface for classes and hide private functions (that might be less fail safe, because they have an internal knowledge of the way they are called). But for testing functions I'm fine with single underscore. In my understanding, they are recommendations not to be used outside of a class. But in wrapped testclasses, that might be fine.

paddywwoof, 2021-03-09:
OK. The pi3d.masked_screenshot() has to occur *after* the various pi3d.Shape.draw() calls and *before* the pi3d.Display.loop_running() call, which swaps the buffers over (otherwise the screen shot is just filled with background!). So if I move the loop_running() up to the top of the method then I can wrap the method and save the sample.

It doesn't test all the functionality by any means (i.e not mqtt, http, keyboard, mat k-means etc etc) but it will fall over if many of the main components have a bug so probably worth doing. I'm not sure of the best way to tie this into a testing system so will just post in a new branch and let you experiment with it. (I will put the movement of loop_running() into dev branch)

paddywwoof, 2021-03-09:
Here is the commit. c9bf82ff17ac90b1744e7e73ff157e2356ed4836 hopefully it will make sense. Let me know if I need to give a bit more explanation.

I realize now I look that it really needs to be run from the directory parent of ```test/``` which might need tidying up

paddywwoof, 2021-03-10:
Just been trying the run_test.py script on the RPis and getting a big difference. One aspect was using int which made a file twice as big on my 64 bit laptop! But that wasn't the cause of the difference - it was because the portrait pairs were in a different order. At the moment the non-shuffle sort is ordered by exif datetime (which if absent uses the file mod time) which is the same for the two images so the order is random! I will see if I can edit the exif date somehow.

helgeerbe, 2021-03-10:
Well, changing geolocation to be optional. Then I think we are ready for a first release. At least we should sure how to setup on pi 4. The rest I would target to a next release. I think most people would set up PictureFrame once, and will never touch it again.

paddywwoof, 2021-03-10:
Helge, I've patched the run_test.py so it passes the assertion with the same pixel array from laptop. I've set the order of the DateTimeOriginal in the three images.

I will update the configuration_example.yaml and upload it.

paddywwoof, 2021-03-10:
Helge, I've pushed some changes up to the wiki: changed the instructions for Pi 4 for the time being.

helgeerbe, 2021-03-10:
If you're done, I will release PictureFrame

closed 2025-03-31 by helgeerbe