helgeerbe/picframe #42: Provide ability to change text justification

jgodfrey, 2021-02-27:
Currently, the text overlay is _left_ justified within its given area (either the entire screen width or 1/2 the width in the case of portrait pairs). I'd much prefer the text to be _center_ justified within the same areas. But, to each his own, so I think we should make this a configuration option and allow any of _left_, _right_, or _center_ text justification.

paddywwoof, 2021-02-28:
Jeff, justification is easy enough to fix. I will do that now.
Paddy

jgodfrey, 2021-02-28:
Thanks Paddy, that'd be great.

paddywwoof, 2021-02-28:
should work now see #59 but needs checking on other setup

jgodfrey, 2021-02-28:
The new text justification seems to be working as expected - nice!

I will say that it seems to wrap prematurely at times (unrelated to the justification changes I assume). I see there's a `text_width` property that controls that - what does does that value actually represent?

paddywwoof, 2021-02-28:
It's a number of characters, see https://github.com/tipam/pi3d/blob/03e4e11ebd11ee05d907ae66e640e649761c680b/pi3d/util/TextBlock.py#L309 which is obviously less satisfactory than using the actual char glyph widths... but the process of chopping up text always produces uneven line lengths because of word sizes, so I thought it wouldn't matter too much.

It basically means that you have to tweak the text_width value to get a result that fits nicely with your screen and chosen character size. We should probably put a note in the user instructions!

jgodfrey, 2021-02-28:
Ah, OK. Yeah, that's unfortunate as it makes it a bit fiddley to configure. From a user-perspective, it'd be nice to just specify a buffer-zone at the edges to avoid (though, that's probably not even necessary and could just be fixed).

Ultimately, it'd be nice if we could auto-calculate an appropriate `text_width` value and not require the user to figure it out experimentally...

paddywwoof, 2021-02-28:
The only simple(ish) way I can think of doing that at the moment is to use a rough rule of thumb pixels per char based on the char size setting. Then as each text block is drawn the pixels per char can be calculated from the actual x positions of the characters and that could be fed into the line length calculation for subsequent text blocks.

The alternative would need the pi3d.TextBlock methods to work in a different way (possible but a more major project)

Maybe this should go in the todo list rather than sort it out for this release.

jgodfrey, 2021-02-28:
> Maybe this should go in the todo list rather than sort it out for this release.

Yeah, completely agree. There's no rush on this. Just commenting as a possible, future improvement.

closed 2021-02-28 by paddywwoof