helgeerbe/picframe #44: Reduce darkened area for text overlay

sapnho, 2021-02-28:
When matting and text overlay is activated, the darkened area stretches far above the text area. Can we limit this to the text area? It's probably independent of the matting feature but with matting on, it is very visible.

helgeerbe, 2021-02-28:
Was not sure, if this is a bug or enhancement request.

sapnho, 2021-02-28:
I am not sure how Jeff intended it but I would propose to limit the darkening to the text area.

jgodfrey, 2021-02-28:
Yeah, I agree that the text scrim area is probably "taller" than necessary. Though, the idea is to 1) provide a smooth gradient that ultimately blends into "nothing", and yet 2) have an area that's dark enough near the bottom that the text is readily visible regardless of background.  

So, the gradient has to have some amount of "fall off" distance beyond the dark area that's required for the text. Looking at the code today, the gradient extends 1/3 the height of the screen (though, much of that isn't visible). Reducing that to 1/4 screen height looks OK to me, though that requirement will definitely vary with the number of lines of text that are displayed.

The scrim height is fixed and not related to the height of the displayed text string.

jgodfrey, 2021-02-28:
One other possibility here... Paddy reworked the gradient construction to be based on a `numpy` array. AFAIK, that gradient is entirely linear in nature. That is, it goes form max-dark at the bottom edge to completely transparent at the top edge. It might help to base the gradient on a curve that tends to front-load the darker color (toward the bottom in our case).

Anyway, not sure what that might take in the numpy construction code. 

In the mean time, I think we can 1) shorten the scrim area and 2) make it a bit lighter (maybe)?

I'll try to commit something soon unless someone beats me to it.

paddywwoof, 2021-02-28:
You can make any curve you want really. Change the line
```python
        text_bkg_array[:,:,3] = np.linspace(0, 170, bkg_ht).reshape(-1, 1)
        # to something like
        text_bkg_array[:,:,3] = (np.linspace(0, 13, bkg_ht) ** 2).reshape(-1, 1)
```
or if it needs to be more complicated (a sigmoid type of curve) make the x values first ```x = np.linspace(0, x_max, bkg_ht)``` then use that in a formula.

jgodfrey, 2021-03-01:
Just merged [d3e62ae ](https://github.com/helgeerbe/picframe/commit/d3e62ae415b243c919f42855bce6c7eeb550efec)to fix (?) this. I reduced both the height and opacity of the scrim texture. I played with making the gradient based on several curves, but (to my eye) the linear gradient looks better (though, maybe we just need a better curve formula).

I'll leave the ticket open until others have had a chance to validate...

paddywwoof, 2021-03-02:
I think the new version looks fine. (We can always ease back later if we went too far)

closed 2021-03-02 by paddywwoof