helgeerbe/picframe #45: Image info should only show when image transition has finished

sapnho, 2021-02-28:
I have set fade time to 10, and transitions to every 20 sec. Immediately when crossfading starts, the info of the new photo is shown, long before the new one is properly visible. Can we only show this information when the fade-in has been completed?

Paddy had fixed that in the old version.

sapnho, 2021-02-28:
Maybe we can link this to the darkening. The background darkening only appears when the new image has fully appeared.

paddywwoof, 2021-02-28:
I can't reproduce this. I have ```show_text_tm: 8.0```, ```time_delay: 20.0```,  ```fade_time: 10.0``` but I'm running on laptop. If I use the next button the name flashes up straight away then fades out and int and out again. (Which is a bit odd but actually quite useful) Do you think it is related to the size of the images, which could be making the FPS drop? Do you get the same effect with longer time_delay or shorter fade_time?

sapnho, 2021-03-01:
My files are all 1920 x 1200px, about 2-3MB, so nothing major. But I will check different fade times when the database is updated. Currently, I can't even start anymore because of missing files links..

helgeerbe, 2021-03-01:
I have seen, that Jeff merged a partial fix to the database issue. I will create a new package, so you can test this and the dB issue.

sapnho, 2021-03-01:
I just installed the latest version, and now it works as Paddy described.

jgodfrey, 2021-03-01:
> I just installed the latest version, and now it works as Paddy described.

That's expected - until we appropriately handles missing images while the slideshow is running.

paddywwoof, 2021-03-02:
This should be fixed now in dev branch. Re-open it if the problem suddenly appears again.

closed 2021-03-02 by paddywwoof