helgeerbe/picframe #46: Make config changes persistent and changeable at runtime

helgeerbe, 2021-02-28:
Maybe migrate config to database and offer a webfrontend for changing config.

paddywwoof, 2021-03-02:
I'm not sure about this. It seems a fairly major change to the way that config variables are used.

helgeerbe, 2021-03-04:
Maybe you are right, normal use would be configure it to your needs onetime and run it forever.

paddywwoof, 2021-03-10:
Rich, I'm not sure the best way to approach this one. I suppose there are two classes of config at the moment - ones that specify initial values for things that can be altered live with mqtt or http, and ones that are only used in setup, but in this last type are ones that, realistically, are fixed (such as http port number) and ones that could be altered (such as portrait_pairs or mat_images) but there isn't any functionality to do that yet. There would need to be a lot of changes to make all these latter config values 'live-changeable'.

The second aspect is saving the live changed values back into the config file which,  as mentioned above sounds like a db job. But quite a bit of work to implement - plus it really needs an easy input and management system that ideally dosn't require the user to install sqlitebrowser or suchlike.

So your idea of an html form to modify the config file might be a reasonable compromise. There would need to be a system of saving the previous versions in a series of backups that could be easily reverted if something got scrambled. Presumably the form would be serialized to json in javascript then convert to yaml in python easily engough, and vise versa though I imagine verification and filtering might be complicated.

closed 2025-04-15 by github-actions[bot]