helgeerbe/picframe #47: Stopping picframe with CTRL C doesn't always work

sapnho, 2021-02-28:
Working on a headless Pi via ssh. When I start picframe, I can sometimes not stop it with CTRL C.

sapnho, 2021-02-28:
Note: When I wait for like a minute or so, it will eventually stop.

helgeerbe, 2021-02-28:
Is keyboard true working? Stopping with ESC?

sapnho, 2021-02-28:
I don't have keyboard attached, so ESC never works. But I think I know what it is. To test the system thoroughly, I put 25k images in Pictures. And it just takes a while to get going I guess. Once it gets going, CTRL C works when typed twice.

What is weird is when I started picframe for the very first time, the slide show started almost immediately.

helgeerbe, 2021-02-28:
Just tested it. Running picframe over ssh with keyborad enabled. I can make use of the keyboard. ESC is quitting picframe.
Killing wit Ctrl-c does indeed to have a problem.

ToDo for Helge check:
Configure all threads as daemons, e.g.:

thread = threading.Thread(target=thread_msg, args=args, kwargs=kwargs)
thread.daemon = True
thread.start()
These threads will exit when the main thread exits, without any additional intervention needed from you.

paddywwoof, 2021-02-28:
Yes I considered making the threads daemons buts that's a bit like turning your computer off by cutting the powere. Better to have threads stop in a controlled way if possible. The keyboard thread assumes it's being stopped by Esc. Other threads have a Boolean so they can be stopped remotely.

Ctrl C is a bit like cutting the power but is possible to catch in python and trigger measured shutdown.

Picture should start showing very quickly but if the file caching is running flat out it probably adds big latency on Ctrl C. Http should work with v little setup and allow killing system in controlled way.

helgeerbe, 2021-02-28:
import signal
import sys

def signal_handler(sig, frame):
    print('You pressed Ctrl+C!')
    controller.stop

this in controller.loop
signal.signal(signal.SIGINT, signal_handler)

helgeerbe, 2021-02-28:
I'll give it a trial.

helgeerbe, 2021-02-28:
https://github.com/helgeerbe/picframe/compare/dev...ctrl-c seems to work, but is a bit harsh to controller.loop

paddywwoof, 2021-02-28:
I'm not really sure why you set ```self.__shutdown_complete = True``` rather than just call self.stop(). By setting __shutdown_complete you stop the loop blocking until the processes in the loop have stopped. Was that necessary for some reason?

helgeerbe, 2021-02-28:
It's the main thread. So when the handler is called, the loop isn't running anymore. Without setting `self.__shutdown_complete = True`, stop will wait forever.

paddywwoof, 2021-02-28:
Ah, so it doesn't intercept the ctrl c, just allows other functions to be called afterwards. There probably isn't much in the controller loop to worry about anyway. When I was trying to get rid of the seg faults etc it was mainly the display loop that had to close down last, and the image_cache be allowed to flush to db correctly.

helgeerbe, 2021-02-28:
I could try to set a lock.
interruptLock = threading.Lock()

paddywwoof, 2021-02-28:
I don't think it's worth putting too much effort in. Normal operation won't use starting from ssh and stopping with ctrl-c. Same with the keyboard. Usually several ctrl-c will stop it... and if it fails: pull the plug!

helgeerbe, 2021-02-28:
OK, I will merge it. The idea with the lock was nonsens. This is for multithreading. The controller loop is hard interrupted.

Ascendin81, 2022-01-14:
I am using a mini Bluetooth keyboard with Pi4, and I cannot exit the slideshow with ESC, or Ctrl-C, or anything.  I can however do Alt-Tab and go to the terminal, and then stop it, but sometimes it messes up the database.

helgeerbe, 2022-01-14:
I know there are issues with Ctrl-C. I would propose to control picframe by systemd. 
https://github.com/helgeerbe/picframe/wiki/Running%20As%20Service

helgeerbe, 2022-01-20:
Fixed in 2022.01.20

closed 2022-01-20 by helgeerbe