helgeerbe/picframe #48: Starting large Pictures folder takes a few minutes

sapnho, 2021-02-28:
Starting picframe with 25000 images will take 2m15s. Is there a way to speed this up? Paddy knows this request from earlier and had found a way if I remember correctly.

jgodfrey, 2021-02-28:
That definitely doesn't sound right. The startup should be relatively painless regardless of the number of images involved. Something sounds wrong.

paddywwoof, 2021-02-28:
Agree; that sounds like a bug. it's not supposed to wait more than a few seconds. And if it does a 'no pictures' picture should show.

jgodfrey, 2021-02-28:
The caching system _does_ build a list of new/modified files and folders on each call. So, if the 25K images were all new, building that list will all happen up front - but that should be *really* fast (even with that many images). I certainly wouldn't expect it to take 2+ mins...

sapnho, 2021-02-28:
The first time after a fresh install it was quick. When I removed a few images it became very slow.

jgodfrey, 2021-02-28:
@Wolfgang. So, to reproduce this, it sounds like I should...

- Remove my db
- Add lots of images to my Pictures folder
-  Start the app (this seemed OK above)
- Stop the app
- Remove a few image files at random
- Start the app again (this was slow)

Is that correct, as far as you can remember?

jgodfrey, 2021-03-01:
So, I've poked around here a bit. No solutions yet, but recording my findings for now...

For some reason a file that's _NOT_ found in the db in `__get_modified_files()` is _way_ faster than a file that is found in the db. On first blush, that appears to be due to the 2nd part of this line:

`if not found or found['last_modified'] < mod_tm:`

So, the `not found` part is blazingly fast, but the `found['last_modified'] < mod_tm` seems *much* slower. To a certain extent, that's to be expected I guess, but the difference here seems really excessive.

What that means in practice is that the very first look up of images (when none exist in the db) is _MUCH_ faster than later lookups when most of the images are already in the db and changes have occurred.

This manifests itself each time `update_cache()` is called and has no _in memory_ images to process. At that point, it checks the disk for changes. If changes are found (additions or deletions), it processes them as it has time. But, that pass to find the changes is much slower - especially if it's in a folder that contains lots of images. 

@sapnho - Are your 25k images spread between a bunch of folders, or are they in a relatively flat structure? If your removed files were in a folder where there are lots of other images, this slowdown is hit...

Where that becomes a problem is if the call to `update_cache` is still running when `query_cache` is called. In that case, we enter `query_cache` call, but it doesn't complete until after the `update_cache` call completes. I'm not sure what's causing this interleaved behavior, but it feels like `query_cache` may be running into a db lock and waiting for it to be freed (though, not sure).

So, a few things to look at. 

1. Why is the `OR` block of the above-mentioned code line so much slower? Perhaps that query could be more clever...
2. Why does the `query_cache` call stall if `update_cache` is already running?

jgodfrey, 2021-03-01:
Related to the above, I think I've made the query faster (still some testing to do). So that may take care of item #1 above. Still need to look into item #2.

jgodfrey, 2021-03-01:
Just merged [f5ecc09 ](https://github.com/helgeerbe/picframe/commit/f5ecc09ff4acd5a176161d4b2598062887be334f)as (at least) a partial fix here...

This makes the look-up of missing or modified files much faster (so, should greatly reduce the reported delay), but doesn't address the interleaved nature of `query_cache()` and `update_cache()`.

sapnho, 2021-03-01:
> Wolfgang - so, to reproduce this, it sounds like I should...
Remove my db
Add lots of images to my Pictures folder
Start the app (this seemed OK above)
Stop the app
Remove a few image files at random
Start the app again (this was slow)
Is that correct, as far as you can remember?

I had about 20 folders in my /Pictures directory, each holding 1250 images. No subdirectories below that.
Then I moved 19 folders out of /Pictures.
When I restarted PF, I started getting these error messages.

paddywwoof, 2021-03-01:
If the revised __get_modified_files has fixed it (for the 25,000 file case) then that's probably fine. If there's still a delay we could do a similar thing to the file update function. something along the lines of
```python
    def __get_modified_files(self):
        ...
        stop_tm = time.time() + 2.0
        while time.time() < stop_tm:
            (dir,date) = self.__modified_folders.pop(0)
```

jgodfrey, 2021-03-01:
> When I restarted PF, I started getting these error messages.

I assume the error message you're referring to are more related to the other issue here, right?

https://github.com/helgeerbe/picframe/issues/49

The (partial) work I did here was to improve the delay you reported. The errors (the other issue) are simply because the db still contains references to now-missing images and the slideshow doesn't really check for that ATM. That will be fixed.

jgodfrey, 2021-03-01:
> If the revised __get_modified_files has fixed it (for the 25,000 file case) then that's probably fine. If there's still a delay we could do a similar thing to the file update function. something along the lines of

I think it'll be more complicated than that since (currently) the folder modified times aren't updated until all of the files have been added. So, that'll cause the folder scans to be rerun each time - which will likely expose the same issue...

And, I think there *is* still something that needs some attention here (the apparent _lock wait_ between the mentioned methods)

paddywwoof, 2021-03-01:
Hmm, I'm not sure because as soon as any file gets added to self.__modified_files the folder checking should stop, and the suggested __get_modified_files() mod should always do complete folders. If I've not figured the logic out correctly (it's getting convoluted) but maybe something with a list of 
```python
            ...
            self.__modified_folders = [x for x in self.__get_modified_folders() if x not in self.__processed_folders]
            ...

    def __get_modified_files(self):
        ...
        stop_tm = time.time() + 2.0
        while time.time() < stop_tm:
            (dir,date) = self.__modified_folders.pop(0)
            ...
            self.__processed_folders.append((dir,date))
...
#eventually processed_folders gets emptied.
```
but done in a more elegant way. But as I say I feel you are locking the re-scans by adding anything to the files list, so the issue to watch out for is only processing half of folder.

jgodfrey, 2021-03-01:
> Hmm, I'm not sure because as soon as any file gets added to self.__modified_files the folder checking should stop, and the suggested __get_modified_files() mod should always do complete folders. If I've not figured the logic out correctly (it's getting convoluted) but maybe something with a list of processed folders

Yeah, you're right. We could break the file look-up down to work within only 1 folder at a time. Though, if a folder is sufficiently large, I think the problem still exists. So, while that would help, and maybe be enough for most cases, it doesn't seem to address the real problem. Though, I'm still not sure what's causing the real problem...

I'll try to poke around again this evening. Though, definitely feel free to give it a go if you'd like...

jgodfrey, 2021-03-02:
After having submitted the first round of changes (the query optimization discussed above), I haven't been able to reproduce any significant delay in a number of circumstances, including:

- Stopping and restarting the app before the cache is completely built
- Adding files (with the app both running and stopped)
- Removing files (with the app both running and stopped)

Do others see any significant issues here now?

Related, I've been reading about running SQLite in [WAL mode](https://sqlite.org/wal.html), which also might be interesting for this project. Though, I still have some more research to do before I'd actually consider switching it over...

paddywwoof, 2021-03-02:
No. But... I haven't tried adding and deleting tens of thousands of images yet. A few tens of images added or deleted doesn't seem to take a noticeable time.

WAL might be good as it claims to speed up writing though I imagine that the majority of the time is spent doing the exif work. I suppose it would be worth doing a cprofile run during the cache build and investigating further if the db is a bottleneck at any stage.

paddywwoof, 2021-03-09:
I think this one is also OK to close. If we get feedback that the first run is very slow for large collections of images then we can try making it run one folder at a time, but I think this is pretty fast as it is.

closed 2021-03-09 by paddywwoof

jgodfrey, 2021-03-09:
So, what's the current consensus on this one? Is there still a problem here, or is this one good? While I don't expect unusually large changes to the folder structure to be gobbled up **instantly** by the caching system, I also don't expect it to be very painful either. And, while I don't want to make the caching more complex than necessary, if it still needs work, so be it...

jgodfrey, 2021-03-09:
Oh, I see Paddy closed this just as I submitted the above... :)

paddywwoof, 2021-03-09:
;)