helgeerbe/picframe #49: Bug when removing images from directory

sapnho, 2021-02-28:
I just renamed the old Pictures directory with many images and created a new Pictures with only a thousand. Upon restart of picframe I get:

> picframe
INFO:start.py:starting ['/home/pi/.local/bin/picframe']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
WARNING:viewer_display.ViewerDisplay:Can't create tex from file: "/home/pi/Pictures/Fnatenrahmen Pictures Kopie 2/2018. 04.15 - 15.01. 42 Fnatenrahmen.jpg" or "None"
WARNING:viewer_display.ViewerDisplay:Cause: [Errno 2] No such file or directory: '/home/pi/Pictures/Fnatenrahmen Pictures Kopie 2/2018. 04.15 - 15.01. 42 Fnatenrahmen.jpg'
Traceback (most recent call last):
  File "/home/pi/.local/bin/picframe", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/start.py", line 144, in main
    c.loop()
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/controller.py", line 259, in loop
    if self.__viewer.slideshow_is_running(pics, time_delay, fade_time, self.__paused) == False:
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 368, in slideshow_is_running
    new_sfg = self.__tex_load(pics, (self.__display.width, self.__display.height))
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 231, in __tex_load
    im = self.__check_heif_then_open(pics[0].fname)
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 151, in __check_heif_then_open
    image = Image.open(fname)
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2634, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Pictures/Fnatenrahmen Pictures Kopie 2/2018. 04.15 - 15.01. 42 Fnatenrahmen.jpg'
^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt

sapnho, 2021-02-28:
Could it be that there are issues refreshing the database? This file

> FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Pictures/Fnatenrahmen Pictures Kopie 2/2018. 04.15 - 15.01. 42 Fnatenrahmen.jpg'

is not included in the new Pictures Folder.

jgodfrey, 2021-02-28:
Hmmm... Yeah, I guess I can understand why. When we update the cache, we don't actually clear "missing" images (your case) from db until we've first added all of the new images. But, the slideshow is allowed to run during that time. So, it looks like the show has been built with references to missing images.

Probably the easiest fix would be to simply skip images that are found to be missing (that should be handled gracefully for other reasons anyway). They *will* get removed from the db, just not in time in this case.

So, we could simply skip missing images at display time - though that could cause a portrait pair to only have a single image available.

Or, we could verify the existence of each image as it's added to the presentation script. Though, by the time we get around to displaying a given image, it could be gone for other unknown reasons anyway.

My vote would be to just ignore missing images at display time and let the existing db cleanup to work as is...

jgodfrey, 2021-02-28:
Yeah, looking at the code, the image loading function looks pretty dangerous ATM... It does nothing to verify the image it was asked to open actually exists on disk...

paddywwoof, 2021-02-28:
Yes. Short term stop the error and just skip missing files. Longer term it would be nice to try and figure out a directory name change or a bulk move of images into a new subdirectory. It feels like it wouldn't be too difficult with the db structure the way it is and would save the reload especially of nominatim data

sapnho, 2021-02-28:
What we need to anticipate is that people will move/remove images both during PF being active and when stopped.

jgodfrey, 2021-02-28:
> What we need to anticipate is that people will move/remove images both during PF being active and when stopped.

In theory, either should be fine - though there are obviously a few bugs that need to be squashed.

paddywwoof, 2021-03-01:
The ViewerDisplay.__tex_load method needs to be sorted out a little more generally than just testing if a file exists as it's possible for the try block to fail for other reasons (corrupt images were an issue I had so I enabled LOAD_TRUNCATED_IMAGES). At the moment the expected returned pi3d.Texture is set to None and this just prevents the existing image showing, but that means a) the previous image will just show for a repeated 200s until a valid image is found b) if this is the first image the screen will be blank. Probably it should fast forward if __tex_load returns None but the slide change machinery is held in Controller so this would need a signal to be returned.

At the moment the ViewerDisplay.slideshow_is_running() method returns a simple boolean from pi3d.loop_running() but I'm not sure that useful now as the Controller loop is being broken directly (rather than second hand by stopping the display in some way). Maybe we should change the return value to mean a bit more, including image didn't load - move on to next.

I'm happy to tinker with this but I'm going to be out for a few hours so it won't happen till later today.

jgodfrey, 2021-03-01:
> Probably it should fast forward if __tex_load returns None

Agreed - this feels like the right approach. It'd likely be easier to fix the issue earlier in the process (like, when the initial list of db id's is being built for the slideshow), but then we'd really need to verify it again at "access" time (as availability could have changed). So, since we have to do that anyway, there's really no need for an additional check upstream.

paddywwoof, 2021-03-01:
I'm home now so I can knock something up and test it

paddywwoof, 2021-03-01:
6e25887c9643d32181cf2cd823aee44459dc7718 copes with files disappearing and appearing. If the subdirectory filter is on then the new images added to a subdirectory won't appear until you run Controller.subdirectory() to rebuild the list with sqlite query.

Have a play and see if there are any other issues. It's on a new branch ```skip_images``` (At some stage we will have to start tidying some of this multitude of single issue branches!)

helgeerbe, 2021-03-01:
I could enable auto delete after successful merge of pull requests.

paddywwoof, 2021-03-01:
Probably not worth it. This probably just an initial rush of blood to the head and it's easy enough to delete them on the remote server I think (It's been a while as it's fairly rare but I remember is was easy)

jgodfrey, 2021-03-01:
> I could enable auto delete after successful merge of pull requests.

I've just been deleting mine after merge.

closed 2021-03-02 by paddywwoof