helgeerbe/picframe #50: enhance configure does not expand relativ path

helgeerbe, 2021-02-28:
If using `picframe -i .` the path is not expanded to absolute path. So it will work only, if current directory is pi's home directory

helgeerbe, 2021-02-28:
pip install picframe==2021.2.27a3 fixed it

closed 2021-02-28 by helgeerbe

sapnho, 2021-02-28:
What is the best way to update for users?

helgeerbe, 2021-02-28:
never tried this `pip install picframe --upgrade`
but with the pre releases you have to uninstall and install
`pip uninstall picframe`
`pip install picframe`

helgeerbe, 2021-02-28:
picframe_date untouched. but when you run `picframe -i .` save your configuration.yaml. this will be overwritten.

sapnho, 2021-02-28:
>  pip uninstall picframe
Skipping picframe as it is not installed.

helgeerbe, 2021-02-28:
ensure you use python3 so use `python3 -m pip uninstall picframe`