helgeerbe/picframe #52: Alternative way of calculating line length automatically

paddywwoof, 2021-02-28:
The only simple(ish) way I can think of doing that at the moment is to use a rough rule of thumb pixels per char based on the char size setting. Then as each text block is drawn the pixels per char can be calculated from the actual x positions of the characters and that could be fed into the line length calculation for subsequent text blocks.

The alternative would need the pi3d.TextBlock methods to work in a different way (possible but a more major project)

Maybe this should go in the todo list rather than sort it out for this release.

_Originally posted by @paddywwoof in https://github.com/helgeerbe/picframe/issues/42#issuecomment-787542295_

paddywwoof, 2021-03-02:
I've put up a version as a new branch 855986dbc702cd78e3ed90623a18cba4e8677cc4 It has to make two compromises i) it uses an average pixels per character from the first line of text being shown on screen ii) it makes a guess at the extra characters beyond the expected line length before the cut-off is reached to move the word onto the next line instead of appending to the current one.

Even to do this in the pi3d code accurately is moderately non-trivial. As the TextBlock is built up character by character the loop could check for the x position of the last character however it would then have to back-track to the last space then re-allocate the characters to the next line. At the pi3d level each character is vertex in GL_ARRAY_BUFFER with the uv components mapped to the locations of characters in a Texture so 're-allocating' is a matter of changing np.array values etc. TLDR; not impossible but probably a bit messy, and tying ourselves into the latest version of pi3d might not be ideal. So, as this isn't urgent we can try it out with different screens, file and town names and see it it will do.

paddywwoof, 2021-03-05:
I've made an [alteration to pi3d.TextBlock](https://github.com/tipam/pi3d/commit/86e7fc0acd5b1b33cd003792c70eec020248546a) to allow wrapping on pixel width. At the moment it's in the develop branch, is it worth requiring the latest version of pi3d? If anyone wants to test it I could make a picframe branch with revised version of viewer_display but you would have to clone pi3d onto your RPi then pick up that version at the start of viewer_display (which only requires a ```sys.path.insert(1,'/home/pi/dev/pi3d')``` at the start)

jgodfrey, 2021-03-06:
> At the moment it's in the develop branch, is it worth requiring the latest version of pi3d?

Since the current work feels like a pretty major update to the picture frame app, I don't have any particular issue with it requiring the latest version of pi3d.

helgeerbe, 2021-03-06:
Me, too. We could change in setup.py the release number of pi3d. In that way the update is done automatically while installing PictureFrame.

paddywwoof, 2021-03-06:
OK I will put out a release of pi3d today.

paddywwoof, 2021-03-06:
Do people want to try this out as a different branch 1096431ae2e5fb7a7814878b421fce89cdca2cbd or should I merge it into dev. I've been running it a while on RPi3 and 4  and it seems OK. Well there isn't any obvious difference apart from there is now no text_width to set in config it just uses the full screen width less 50px each side (or half if pairs)

jgodfrey, 2021-03-06:
The logic sounds reasonable to me. While I don't have any real need to try it out ahead of a merge, if you'd prefer that it get some more testing first, I can probably take a look this evening, though it'll 6-8 hrs before I'm in a position to do so.

paddywwoof, 2021-03-06:
I'm reasonably happy with it (famous last words!) so will merge it anyway. If there's something wrong it will almost certainly be an additional tweak rather than reverting to before the merge.

closed 2021-03-06 by paddywwoof

jgodfrey, 2021-03-07:
Hi Paddy,

I installed the latest package this morning and (once I finally got it running...), the text length calculation looks really good to me. I even played with the `display_x`, `display_y`, `display_w`, and `display_h` parameters to see if I could trip it up, but it seems to work as expected based on my  limited testing. Nice job!

paddywwoof, 2021-03-07:
Actually there are a few quirks to the calculation that I've fixed for the next release of pi3d, but the errors are small so not bad as it is - mainly a simplification of the way it keeps track of either end of each line.

jgodfrey, 2021-03-07:
A potential enhancement here might be to make some attempt to evenly divide the text between the required number of lines. I just saw a worst-case scenario go by where I had one completely full line of text plus a second line with only `UK` on it. That would probably look more tidy if the text were more evenly split across the two (required) lines.

paddywwoof, 2021-03-07:
Hmm interesting. Might be feasible with a slightly different approach to the wrap position calculation.