helgeerbe/picframe #54: Changing the Pictures folder still shows images from previous folder

jgodfrey, 2021-03-01:
After building up a cache DB of valid images, if the `Pictures` folder is redefined in the config to point to somewhere else entirely, images from the original folder are still shown.

While we probably don't want to delete valid records that point to a different folder, we shouldn't show them in the viewer. Probably, we want to augment the query in `image_cache.query_cache()` to include a `fname LIKE {picdir}/%` to anchor all selected images to the current pic dir... Looking, I guess very similar to the way we handle the _subdirectory_ query today.

paddywwoof, 2021-03-02:
This branch 847078babebc33a797cd6c9255372044f0c3402e is a relatively simple change as per your suggestion seems to do the job. I'm not sure how much it slows the select down for, say, 25000 files. It will have some impact as pic_dir is expanded to /home/pi/Pictures/% but it's probably measured in ms rather than s.

PS annoyingly I branched while in recalc_text_width so those changes are also in this branch. There's probably a git method for removing a previous commit but it may be easier to see if both lots of changes are OK to incorporate into dev!

paddywwoof, 2021-03-05:
This should be fixed as of where_pic_dir merge

closed 2021-03-05 by paddywwoof