helgeerbe/picframe #57: very slow running on raspberry pi 4 with xinit

paddywwoof, 2021-03-03:
When running picframe (or any other pi3d applications) from the desktop it runs fairly smoothly at about 30FPS however with no desktop loaded starting from the command line using ```xinit``` there is a warning ```libEGL warning: DRI2: failed to create dri screen``` and the app runs at about 6FPS. If ```startx``` is used instead of ```xinit``` then the DRI2 works fine and the application runs full speed. However if ```startx``` is used in a systemd service or via ssh then DRI2 fails and the FPS drops to single figures.

It looks like a permission issue. I don't remember this happening before, though it could have done as most testing would be from the desktop but in the past the dependencies were installed for all users including root and xinit was always run with sudo (which still fixes this issue)

jgodfrey, 2021-03-03:
Ugh - that's a good (but unfortuante) find. I just spun up 2 new rpi4's last night, though I still have some config work to do on them. If there's anything specific you'd like me to test or verify there, let me know.

paddywwoof, 2021-03-03:
Jeff, If you set one up following Helge's instructions, just to make sure the problem is universal and not something stupid I've done at some stage, that would be good. It might be worth trying to run without installing ```xserver-xorg-legacy```. But I think you will have to run as root, which will mean installing modules so they're accessible by root. For instance if I try to run picframe with ```sudo xinit...``` it fails because the yaml module was installed in /home/pi/.local/ etc.

It's not the end of the world if we have to run on the RPi4 as root as that's what's happened with PictureFrame in the past... just that it's not regarded as good practice and Helge seemed to have found a way round it!

helgeerbe, 2021-03-03:
@paddywwoof where will I see this messages? In debug log of DEBUG:pi3d.... I can't see anything unusual. But I'm running a pi 3 on ubuntu. I could also try Pi3 on rasbian.

jgodfrey, 2021-03-03:
> eff, If you set one up following Helge's instructions, just to make sure the problem is universal and not something stupid I've done at some stage, that would be good.

I'll try to run through the motions this evening.

sapnho, 2021-03-03:
How do you start picframe on a Pi4? Just made a fresh install on a Pi4. Still booting to console? GL Driver KMS? 
picframe gives me:

> pi@Pi4withPF2021:~ $ picframe
INFO:start.py:starting ['/home/pi/.local/bin/picframe']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
X11 needs to be running
Traceback (most recent call last):
  File "/home/pi/.local/bin/picframe", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/start.py", line 124, in main
    c.start()
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/controller.py", line 269, in start
    self.__viewer.slideshow_start()
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 348, in slideshow_start
    display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=self.__background, use_glx=self.__use_glx)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/Display.py", line 538, in create
    display = Display(tkwin, use_pygame)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/Display.py", line 146, in __init__
    self.opengl = DisplayOpenGL()
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/util/DisplayOpenGL.py", line 81, in __init__
    assert False, 'Couldnt open DISPLAY {}'.format(display_name)
AssertionError: Couldnt open DISPLAY None
^X^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):

paddywwoof, 2021-03-03:
@helgeerbe you won't see it on Pi3 I don't think (though maybe it will happen if you use the fake KMS driver) It happens with the various message starting X after running xinit... from the command prompt. I'm using the local repo where_pic_dir branch so doing it like
```
pi@raspberrypi:~/picframe $ startx /usr/bin/python3 /home/pi/picframe/picframe/start.py /home/pi/picframe/configuration.yaml

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Current Operating System: Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1280x1024M@60 smsc95xx.macaddr=DC:A6:32:03:FE:0C vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=51db2599-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 01 December 2020  05:59:57PM
xorg-server 2:1.20.4-1+rpt2+deb10u2 (https://www.debian.org/support) 
Current version of pixman: 0.36.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Wed Mar  3 16:30:49 2021
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
INFO:start.py:starting ['/home/pi/picframe/picframe/start.py', '/home/pi/picframe/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe/configuration.yaml:
ERROR:model.Model:error trying to set locale to en_US.utf8
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
```
@sapnho are you starting it via ssh? In which case I think [Helge's instructions](https://github.com/helgeerbe/picframe/wiki/Running%20As%20Service#running-as-service-on-pi4-console-only) will work but type the ExecStart line onto the command line. i.e. you need to give the full path to python, the start.py script and the configuration.yaml file

helgeerbe, 2021-03-03:
running over ssh, I had to export Display first.
`export DISPLAY=:0`
`xinit /home/pi/.local/bin/picframe /home/pi/picframe_data/config/configuration.yaml`

sapnho, 2021-03-03:
I get this:

> pi@Pi4withPF2021:~ $ export DISPLAY=:0
pi@Pi4withPF2021:~ $ xinit /home/pi/.local/bin/picframe /home/pi/picframe_data/config/configuration.yaml
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Current Operating System: Linux Pi4withPF2021 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=DC:A6:32:59:A6:EC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=465c112a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 15 January 2021  02:03:57PM
xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support) 
Current version of pixman: 0.36.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Wed Mar  3 17:00:57 2021
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

paddywwoof, 2021-03-03:
those look like the errors you get running without sudo when you haven't installed xserver-xorg-legacy and setup /etc/X11/Xwrapper.config

just do
```bash
$ cat /etc/X11/Xwrapper.config
```
and post the results

sapnho, 2021-03-03:
> pi@Pi4withPF2021:~ $ cat /etc/X11/Xwrapper.config
> allowed_users=anybody
> needs_root_rights=yes

sapnho, 2021-03-03:
Ah, didn't read the "apt-get install xserver-xorg-legacy" part. I need new glasses...

sapnho, 2021-03-03:
Ok, now it starts but the fade is indeed superslow.

jgodfrey, 2021-03-04:
> > Jeff, If you set one up following Helge's instructions, just to make sure the problem is universal and not something stupid I've done at some stage, that would be good.
> 
> I'll try to run through the motions this evening.

Apologies - I did not get to this tonight.

paddywwoof, 2021-03-04:
No, problem. Wolfgang has probably confirmed that the vanilla RPi4 instructions have an issue. We should probably find the simplest way to get round that. It might just work if you put ```sudo``` before the install line, skip the xserver-xorg-legacy and do ```sudo xinit...``` on the command line and systemd service. However the install location for root won't be ```/home/pi/.local``` but something like ```/usr/lib/python3/dist-packages``` which is not ideal. Normally python looks after the sys.path tangle which is what would happen if we launched a file rather than a module. Maybe we can do that by having a file in our /home/pi/picframe_data called start.py along the lines of
```python
from picframe import start
start.main()
```
then on the command line or systemd service
```
sudo xinit /usr/bin/python3 /home/pi/picframe_data/start.py /home/pi/picframe_data/config/configuration.yaml -- -s off -dpms -s noblank
```

Because I've been wanting to change and check things I've been cloning the repository then ```python3 -m pip install -e .``` but that probably isn't for standard instructions. I assume the ```python3 -m pip install picframe==2021.2.27a4``` will be just missing some of the bug fixes and improvements of the past few days so you could try that (with the above sudo prepended and start.py shell or whatever)

I will try another clean SD install this morning if I get chance.

Paddy

paddywwoof, 2021-03-04:
OK I've burnt a new SD and done the normal setup (and put some pictures in ~/Pictures, forgot the first time!) then:
```bash
$ cd ~
$ sudo python3 -m pip install picframe==2021.2.27a4
$ picframe -i . # using all defaults
$ nano picframe_data/config/configuration.yaml # change ~/Pictures and ~/DeletedPictures to /home/pi/Pictures etc
$ nano picframe_data/start.py # and put in the two lines as above. This could be in ~/ I suppose
  from picframe import start
  start.main()
$ sudo xinit /usr/bin/python3 /home/pi/picframe_data/start.py /home/pi/picframe_data/config/configuration.yaml
```
It seems to run normal speed. I've not been able to track down why the DRI2 function call doesn't work for ssh or systemd. Will keep looking.

sapnho, 2021-03-04:
@paddywwoof I can confirm that your instructions work. I'll let it run for a while.

paddywwoof, 2021-03-04:
Wolfgang, goodness me - I only posted that a few minutes ago. You must have had everything set up ready to go and implemented it instantly!

sapnho, 2021-03-04:
I still had the Pi4 running, removed the old install, and followed your instructions. Ever since I use an HDMI stick on my iMac instead of using a separate monitor for the Pi, things are a lot easier on a full desk. :-)

sapnho, 2021-03-04:
I also noticed that there is no "GL something" error message anymore, or was that only on the Pi3?

sapnho, 2021-03-04:
It would be great if there was a way that we can include the start.py content in the default configuration. One less step for happy users.

helgeerbe, 2021-03-04:
> I also noticed that there is no "GL something" error message anymore, or was that only on the Pi3?

I can confirm, that this error is still on a pi 3

paddywwoof, 2021-03-04:
Wolfgang: Yes making the super start.py file would be easy enough to do in the -i step I would have thought.

It really is time to find what causes the glError message. It seems harmless but it's been there for ages and I just can't find where it comes from. I will make that this afternoon's task. The actual error number means that a GL constant has been passed to a function outside the range of permitted values - but doesn't give any indication of which value or which function!

helgeerbe, 2021-03-04:
What happens when you start a xterm first?

`xinit xterm - e "/usr/bin/python3 /home/pi/picframe_data/start.py /home/pi/picframe_data/config/configuration.yaml" `

One other thing. Open to ssh sessions.
1. first just start `xinit` this should start X and opens a xterm
2. on the second ssh `/usr/bin/python3 /home/pi/picframe_data/start.py /home/pi/picframe_data/config/configuration.yaml`

Is this working? Maybe we have a timing error, if something is not ready yet, when starting X. 

That way also curses works and you could use keyboard, e. b. over bluetooth, to control the program.

paddywwoof, 2021-03-04:
Helge, OK I will try that. Some of the dependencies might be installed in /usr/local/lib/python.. but it might be interesting to see what happens.

paddywwoof, 2021-03-04:
Well it changed the warning messages. I got a complaint about ```xterm``` not being a program so had to do
```
xinit /usr/bin/x-terminal-emulator -e /usr/bin/python3 /home/pi/picframe_data/start.py /home/pi/picframe_data/config/configuration.yaml
```
but still fails to load dri
```
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
```
If I start x-terminal-emulator in one ssh (it gives me the same libGL errors as above so I'm not hopeful, but..) in the next ssh window I get the error about the display already being in use. I feel that if I try to use a different display (if that's possible) it will undo the benefits of starting xterm on the first ssh.

helgeerbe, 2021-03-04:
Well, I should buy me a pi 4 I guess.
I found that, https://forum.openframeworks.cc/t/raspi-4-failed-to-load-driver-vc4/36979.
So there is a chance, that it will work, starting as a service.

paddywwoof, 2021-03-04:
Helge, well done, that's *exactly* our issue so, as you say, there's hope of getting a fix. Though I'm not sure what the cure of removing user/group from systemctl involves - if it boils down to running the service as root then we're already there!

helgeerbe, 2021-03-04:
It's been a while, that I worte X11 code. But there might be some user rights / permission issues. I would try to run it systemd as normal user and see, if it is starting properly. If not, we have to start it as root.
One other solution, would be to do autologin to X and start it as normal user in systemd.

paddywwoof, 2021-03-05:
I made some changes to start.py so it creates a little program to run start if starting sudo xinit. It also checks that picframe -i isn't being run as sudo (which would put the picframe_data in /root/...) and, while I was at it, I made the two directory definitions still in configuration.yaml with ~/ at the beginning into absolute paths.

helgeerbe, 2021-03-05:
Paddy, are you sure, you changed the latest start.py? picframe -i requires a path. Not taking home dir. And it's creation configuration.yaml from configuration_example.yaml. Searching for ~/..... and repaces this with absolut path.

paddywwoof, 2021-03-05:
Helge, no I'm not certain I've changed the right thing. I will check. This is the diff 118bd4ed53f4c97fc6352826d2475ed729118472 The version I modified changed the ~/picframe_data substring but didn't change ~/Pictures or ~/DeletedPictures If you're running picframe as pi that doesn't matter but if you have to run as root then the paths don't work.

helgeerbe, 2021-03-05:
Your are right, this will work. I hadn't in mind to run picframe as a different user (root). Guess this will do the job.

paddywwoof, 2021-03-05:
Hopefully we will find a way. But in the short term if everything else is working, we can release the module with special instructions for RPi4 as there were with the old version.

I've searched a little but I'm not really sure what the suggestion ```I am able to get the systemd unit to startx properly if I remove the user/group that I had previously set there``` means. Does it make sense to you?

helgeerbe, 2021-03-06:
Well, I don't know. Startx is a starter script for xinit and does a little bit more. Maybe it takes care of opengl, but that is more hoping.  
I tested some systemd user services which requires X to be running. So I started startx from console, in Xinitrc I had a one liner which triggerd my service to start. Before that I had authentification problems in X, because my scripts started to early and X was not up and running.

paddywwoof, 2021-03-06:
Yes I did notice that xinit running from attached keyboard and screen had the DRI problem but startx didn't. However startx via SSH had the same issue as xinit. I'm not sure if that's useful info to diagnose a cure

closed 2021-03-09 by paddywwoof

jgodfrey, 2021-03-10:
So, do we have a clean, proven way of installing this (in my case, from code) on a RPi4 now? It seems that every time I start over with the intent of getting a clean install, I end up spending an hour or so hacking and slashing my way through the various suggestions and snippets that are scattered about. Ultimately, I always manage to get things running, but it feels far from the "clean slate" that was my original intention.

paddywwoof, 2021-03-10:
Jeff, this should work. I will try to update the wiki. The first 5 lines would be replaced by just ```sudo pip3 install picframe``` for a vanilla install from pypi.
```bash
$ cd /home/pi/dev/picframe # or wherever your repo is kept
$ git pull
$ git checkout dev
$ cd ~
$ sudo python3 -m pip install /home/pi/dev/picframe --upgrade

$ picframe -i . # NB that period for current folder
$ nano picframe_data/config/configuration.yaml # change nominatim geo_key [compulsory] and use_http [optional] +any others
$ sudo xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml
```
and in ```picframe.service``` without sudo but as root
```bash
...
User=root
ExecStart=xinit /usr/bin/python3 /home/pi/picframe/run_start.py /home/pi/picframe_data/config/configuration.yaml
...
```

sapnho, 2021-03-10:
Wonderful!

jgodfrey, 2021-03-11:
> > Jeff, this should work. I will try to update the wiki.
> 
> Thanks Paddy. I'll give this a try this evening.

Paddy - your instructions worked on a clean RPi4 installation. Thanks!