helgeerbe/picframe #59: Error in georeverse coding

sapnho, 2021-03-04:
For the second time I got this error message:

> ERROR:geo_reverse.GeoReverse:lat=17.606900, lon=66.512500 -> 'features'

Possibly some Exif issues but maybe we can implement a "try" command or something similar to avoid a crash?

It doesn't say which file, unfortunately, so I cannot check the image itself.

jgodfrey, 2021-03-04:
Looking at the code, the error you mention *is* the result of a `try/except` block now. I'd guess the lookup of that `lat/lon` failed, which will trigger the `except` block and the error you mention. It looks like that should bubble up to the original call and simply _not_ fill out the geo location data for the image in question. On the surface, that all appears to be handled ok.

Is it actually causing a crash in your case?

Related, at the level where that error is generated, the code doesn't have access to the image name. It's just blindly looking up a passed `lat/lon` reference. Though, the image name was certainly known higher in the call structure and could be logged from there.

sapnho, 2021-03-04:
Good to know. The thing is that my now screen goes dark after a while, it seems like my old method of deactivating the screen saver doesn't work anymore, so I can't see if it stopped or not.

sapnho, 2021-03-04:
> Related, at the level where that error is generated, the code doesn't have access to the image name. It's just blindly looking up a passed lat/lon reference. Though, the image name was certainly known higher in the call structure and could be logged from there.

If it doesn't cause a crash, I wouldn't worry about the image name.

paddywwoof, 2021-03-04:
putting the lat/lon into nominatim returns ```

```error | "Unable to geocode"``` and the location looks to be in the  middle of the Indian Ocean, so that might narrow it down (plane, boat?). As Jeff says it should be isolated by try: except:

My wife's picture frame was originally set up following your instructions although I've put this new picframe on to show the slides now, but the screen stays on and can be switched on and off with the ViewerDisplay method (via Controller, via http or mqtt etc) I can't remember which instructions I followed, just started at the beginning and worked my way through - all very straightforward!

Also I noticed that picframe==2021.2.27a4 doesn't have the bug fixes for moving images between directories or switching to another root from ~/Pictures so best not to try any of that using the pip install from pypi.org (I expect Helge will update it over the next day or so).

sapnho, 2021-03-04:
> My wife's picture frame was originally set up following your instructions although I've put this new picframe on to show the slides now, but the screen stays on and can be switched on and off with [...]

Yes, but you are using a Pi3 in your wife's frame, whereas my issues are on the Pi4. Last time I checked, the Pi3 was fine as well.

jgodfrey, 2021-03-04:
@sapnho - My original RPi4 setup work was done directly from your article here:

https://www.thedigitalpictureframe.com/how-to-set-up-your-raspberry-pi-for-your-digital-picture-frame/

And, while it's been in "development" mode ever since, I've routinely left it running all day long and I've never seen it go dark on its own.

helgeerbe, 2021-03-04:
New alpha is released. Changelog is here: https://github.com/helgeerbe/picframe/discussions/35#discussioncomment-431012

helgeerbe, 2021-03-04:
option -dpms should not set, when starting with xinit. This turns dpms off. So the frame, couldn't turn display off. The other arguments are fine.

paddywwoof, 2021-03-09:
I think this is OK to close - just a debug message really.