helgeerbe/picframe #60: Screensaver is activated on Pi4

sapnho, 2021-03-04:
I used to do this to avoid the screen from going dark

```
sudo nano /etc/xdg/lxsession/LXDE-pi/autostart and then
@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xset s off
@xset -dpms
@xset s noblank
```

This doesn't seem to work anymore.

paddywwoof, 2021-03-08:
Wolfgang, I think the screen stays on for my setup... I have two things set up, and as it's a reasonably fresh SD, I don't think I've altered anything else. I'm not sure that the command line options for xinit aren't sufficient:
```bash
ExecStart=xinit /usr/bin/python3 /home/pi/picframe/run_start.py /home/pi/picframe_data/config/configuration.yaml -- -s off -dpms -s noblank
```
But I've also followed the [raspberrypi.org instructions here](https://www.raspberrypi.org/documentation/configuration/screensaver.md) and used raspi-config
[image]
Let me know if I need to do any more digging on this.
Paddy

sapnho, 2021-03-08:
Thanks, @paddywwoof : That solved the puzzle! I had never seen that option in raspi config but the setup was changed quite a bit a few months ago. But whatever, for a Pi 4 this is the solution.

paddywwoof, 2021-03-08:
Excellent. I will close this issue. Did you try the screen rotation? There was a question someone asked about having two monitors running from the same Pi in different orientations, which seemed to be quite tricky, though possibly feasible to solve. However if you change your time a the minimum wage it's almost impossible to justify not buying another RPi as the cheapest solution!

closed 2021-03-08 by paddywwoof