helgeerbe/picframe #61: glGetError 0x500

helgeerbe, 2021-03-05:
Well its more likely a pi3d error, but I dig a little into it.

Error occurs in Buffer.py line 195:

````python
if opengles.glGetError() == GL_OUT_OF_MEMORY:
      LOGGER.critical('Out of GPU memory in Buffer._load_opengl')
````

I can't debug into `glGetError()` but this call results to the `glGetError 0x500` output.

paddywwoof, 2021-03-05:
Helge, thanks for finding that. (I was going to dig into it Yesterday but got diverted onto something else) Will have a look at it now.

helgeerbe, 2021-03-05:
Found this is just an old Error.

Originally raised ist the Error here "DisplayOpenGl.py" line 131. by this call: `opengles.glEnable(GL_PROGRAM_POINT_SIZE)`

helgeerbe, 2021-03-05:
Old error means, that you call `opengles.glGetError()` the first time in Buffer.py. But `opengles.glEnable(GL_PROGRAM_POINT_SIZE)` is the call, which creates the errr.

helgeerbe, 2021-03-05:
GL_PROGRAM_POINT_SIZE
If enabled and a vertex or geometry shader is active, then the derived point size is taken from the (potentially clipped) shader builtin gl_PointSize and clamped to the implementation-dependent point size range.

Errors
GL_INVALID_ENUM is generated if cap is not one of the values listed previously.

So maybe your definition does not match gl.h?
GL_PROGRAM_POINT_SIZE = 0x8642

helgeerbe, 2021-03-05:
Indeed, in mine  `/opt/vc/include/GLES/gl.h `  there is no `PROGRAM_POINT_SIZE`
Found, that it requires openGl > 2.0, don't know, what raspbian is using.

paddywwoof, 2021-03-05:
Ah Helge, I wish I'd checked my emails 2 hours ago! Anyway I also tracked the error down to there and hopefully fixed it. ffdb3ba099d125c4e5dfcbd3b167d37487899fe0 The problem is essentially that the OpenGL 'specification' ```is more what you'd call "guidelines" than actual rules``` so things that work on OpenGLES2.0 might not work on OpenGL3.1 and definitely not visa versa!

helgeerbe, 2021-03-05:
Good to know, that you fixed it. Guess this requires a new pi3d version? Actual the minimal version for pi3d is 2.42. Do we have to change it in setup.py if you release a new version. Or was this a more cosmetic bug?

paddywwoof, 2021-03-05:
The RPis up to v. 4 used OpenGL ES2.0 and that's what the broadcom 'legacy' driver libGLESv2 uses. The RPi4 can't use the broadcom drivers and instead uses the mesa driver that's been improved and allows OpenGL2 and 3 'emulation' in so far as the actual GPU is essentially the same. It's not really emulation like a virtual machine as much of the code (shaders) is compiled, however some things do run *slower* on the RPi4.

I think this is a cosmetic bug as it's existed essentially since the RPi4 came out.

I would like to know what system you used to track down the source of the error - I knew it was coming from some previous libGLESv2.so function call but had to find it by putting in print statements to bracket it down!

I will put out a new version of pi3d in the next few days. I might see if I get any inspiration on the text justification in TextBlock and include that as well.

Thanks again

Paddy

helgeerbe, 2021-03-05:
I'm using visual studio code, over the marketplace you can install a lot of plugins.
- Of course python (let you switch between different python versions)
- pylint for well formed python code
- remote host (over ssh I develop directly on my frame on the wall, like I'm using it directly)
- The debugger is excellent and let you also debug imported library (with some setting, but in that way, I don't need sys path, to debug pi3d).

To track the error down, was quite easy. Set a break point before the first call to pi3d. And step through the loop. When the error occured. I set an other breakpoint and went into the calll. After all, I had to restart PictureFrame about 5 times, to find the first call to glError.

After identifying openglens I tracked, where it was instantiated. Then simply going step by step thourgh the code.

paddywwoof, 2021-03-05:
I use code already - I will have to add a few more plugins, certainly remote host and debugger would be handy.

closed 2021-03-05 by paddywwoof