helgeerbe/picframe #64: Portrait Mode

sapnho, 2021-03-05:
One request that I get quite frequently is PictureFrame portrait mode.

This doesn't seem to be as easy as adding

 `Display_rotate=1` in /boot/config.txt

This feature would be quite welcome by a number of dedicated users.

paddywwoof, 2021-03-07:
Following instructions from https://pimylifeup.com/raspberry-pi-rotate-screen/

On RPi3 and earlier the ```/boot/config.txt``` can be edited and a line inserted ```display_hdmi_rotate=1``` added.

On RPi4 you need to run a command line instruction *after* Xserver has started but *before* pi3d.Display.create(). The easiest way to do that is to edit the automatically created ```run_start.py``` file like this
```python
from picframe import start
import os                                                    # << add
os.system("DISPLAY=:0 xrandr --output HDMI-1 --rotate left") # << add
start.main()
```
These seemed to work OK for me, let me know if there are issues and I will investigate further.

One thing that we might constructively add (but I don't think we should aim for this with this release) is that a portrait screen should have the pairing vertically not horizontally. What we should do with the text in these circumstance I'm not sure.

Paddy

jgodfrey, 2021-03-07:
> One thing that we might constructively add (but I don't think we should aim for this with this release) is that a portrait screen should have the pairing vertically not horizontally. What we should do with the text in these circumstance I'm not sure.

Right - I've been thinking about this too. And, I assume the above description means that, with the frame in `portrait` mode, we should really pair `landscape` format images, in a `vertical` fashion (so, one above the other). And, yeah, that does bring some text overlay challenges. I guess the arguably correct thing here would be to simply place the text along the bottom edge of each image's 1/2 screen. 

Though, that could be problematic when pairing images with different aspect ratios as (currently) the matting layout function centers the "pair" on the screen. So, they wouldn't necessarily fall neatly into their 1/2 screen space in all circumstances...

paddywwoof, 2021-03-07:
Yes the text position might be slightly approximate but it might not matter if the text only showed for a few seconds - a bit annoying if it obscured part of the picture and the text was showing for most of the slide duration.

sapnho, 2021-03-08:
Wonderful. I just tried it on a Pi 4 and it works very nicely.

sapnho, 2021-03-08:
The only thing that I just noticed is that the darkened area behind the text is a lot taller than it needs to.

jgodfrey, 2021-03-08:
Yeah - the scrim area is 1/4 of the screen height (until recently, it was 1/3 the screen height), so that probably is overkill for a portrait orientated screen. Perhaps it should just be _X_ pixels high...

paddywwoof, 2021-03-08:
Yes, probably sensible. Maybe proportional to the font point show_text_sz. I suppose there might be more lines with it being narrower, but it's probably not going to wrap up higher than the portrait pair hight. So, yes, X pixels will be fine!

jgodfrey, 2021-03-08:
Though, a hard-coded pixel value will probably require modification for vastly different resolutions (4k for instance). Maybe better would be to use 1/4 of the smallest screen dimension. That would eliminate the need for resolution specific calculations.

paddywwoof, 2021-03-09:
change in dev branch to use min(width, height) // 4

closed 2021-03-09 by paddywwoof

paddywwoof, 2021-03-09:
Yes, I think I will do a fresh start too. I don't think there should any material changes now for bugs... improvements can wait for the next issue!

sapnho, 2021-03-15:
Because I get the question quite often, I wrote a short article: https://www.thedigitalpictureframe.com/raspberry-pi-digital-picture-frame-portrait-orientation/

The matting is even more spectacular in portrait mode!

paddywwoof, 2021-03-15:
Excellent article, along with all your other instructions. One thing I noticed: a backspace seems to have crept in `"run\_start.py"`

sapnho, 2021-03-15:
> One thing I noticed: a backspace seems to have crept in `"run\_start.py"`

Thanks, Paddy, already corrected"

jgodfrey, 2021-03-16:
I haven't read the article yet, but noticed that this:

_So you would like to use your digital frame in portrait instead of the more common portrait orientation?_

Should probably read:

_So you would like to use your digital frame in portrait instead of the more common **landscape** orientation?_