helgeerbe/picframe #68: sqlite3.OperationalError: no such column: None

sapnho, 2021-03-14:
Well, testing is about doing strange things, isn't it?

Here is what I just did (Pi 4). I renamed the /Pictures directory to /Picturesx and started PF. The "No Pictures Selected" came up as expected. Then I renamed /Picturesx back to /Pictures and started. This came up:

> INFO:start.py:starting ['/home/pi/picframe_data/run_start.py', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Traceback (most recent call last):
  File "/home/pi/picframe_data/run_start.py", line 2, in <module>
    start.main()
  File "/usr/local/lib/python3.7/dist-packages/picframe/start.py", line 157, in main
    c.loop()
  File "/usr/local/lib/python3.7/dist-packages/picframe/controller.py", line 249, in loop
    pics = self.__model.get_next_file()
  File "/usr/local/lib/python3.7/dist-packages/picframe/model.py", line 276, in get_next_file
    pic_row = self.__image_cache.get_file_info(file_ids[0])
  File "/usr/local/lib/python3.7/dist-packages/picframe/image_cache.py", line 133, in get_file_info
    row = self.__db.execute(sql).fetchone()
sqlite3.OperationalError: no such column: None

sapnho, 2021-03-14:
Deleting pictureframe.db3 will fix it but maybe there is an easy way to do this automatically in such a case.

jgodfrey, 2021-03-14:
Hmmm. Looking at the code on my phone, I don't really understand the error. I'll try to reproduce this, and see if I can fix it, today.

jgodfrey, 2021-03-14:
Ah. Still on my phone, but I think we must be passing a `file_id` of `None` to the function in this case. Will need to investigate further when I have access.

paddywwoof, 2021-03-14:
I thought I'd caught the file_id being None but there's nothing explicit. The assumption is that all the entries in Model.__file_list are genuine ids, but somehow ImageCache.query_cache() is returning a list with nulls in. Not really sure how - though it gets the ids from all_data so maybe that can return nulls is part of the records have been deleted but not others - changing the name of the root directory will start the process of deleting all the db records and re-building them, so a query half way through might well be scrambled.

jgodfrey, 2021-03-14:
@paddywwoof - Since `get_file_info()` can already legitimately return `None` (and you've handled that on the caller's side) an easy fix would be to just add:

`if not file_id: return None`

 early in `get_file_info()`. 
 
 Sound reasonable?

paddywwoof, 2021-03-14:
Yes, sounds a good fix (normally I'm a bit [over] cautious about coercing variables to boolean - probably because it reminds me of the mess that was javascript < ES6. However in this case file_id should always be an integer greater than zero, so ```if not file_id``` is perfect.)

jgodfrey, 2021-03-14:
Sounds good. While testing this, I stumbled into an image with an invalid EXIF date format that's causing some heartburn. I'll fix that up too and commit both soon...

jgodfrey, 2021-03-14:
Committed fix to `dev` branch via https://github.com/helgeerbe/picframe/commit/5c074afd2f6f456e1f6ae584fb5758898830b5be.

closed 2021-03-14 by jgodfrey