helgeerbe/picframe #69: pi3d DisplayOpenGly.py assert error

helgeerbe, 2021-03-14:
Paddy, when I try to debug the actual dev branch I run into an assert error.

Line 119 of DisplayOpenGL.py

```` python
        context_attribs = (EGLint * 3)(EGL_CONTEXT_CLIENT_VERSION, 2, EGL_NONE)
        if numconfig.value > 0:
          self.config = poss_configs[0]
          self.context = openegl.eglCreateContext(self.display, self.config,
                                                  EGL_NO_CONTEXT, context_attribs)
          if self.context != EGL_NO_CONTEXT:
            break
      assert self.context != EGL_NO_CONTEXT and self.context is not None
````

Maybe only the indent is wrong? I've got context is not an attribute of self. As you can see self.context will be defined only under a condition.

helgeerbe, 2021-03-14:
It only happens when running from remote.

`export DISPLAY=:0` would fix it. 

So maybe not really an error.

paddywwoof, 2021-03-14:
Helge, yes the code is a bit lazy there and I've change it for the next release
```python
        if numconfig.value > 0:
          self.config = poss_configs[0]
          self.context = openegl.eglCreateContext(self.display, self.config,
                                                  EGL_NO_CONTEXT, context_attribs)
          if self.context != EGL_NO_CONTEXT:
            break
        else:
          self.context = EGL_NO_CONTEXT
      assert self.context != EGL_NO_CONTEXT and self.context is not None
```
But, effectively if numconfig is not greater than zero it's just not going to work. Maybe I should add a ```print()``` or ```logging.error()``` to give people more clue as to what might have gone wrong: basically egl can't find a display context.

closed 2021-03-14 by paddywwoof