helgeerbe/picframe #71: Pi4 Autostart

sapnho, 2021-03-14:
Has anyone tried the autostart on a Pi 4? https://github.com/helgeerbe/picframe/wiki/Running%20As%20Service#running-as-service-on-pi4

I guess the line

> ExecStart=xinit /usr/bin/python3 /home/pi/picframe/run_start.py /home/pi/picframe_dat$

should be

> ExecStart=xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_dat$

but even then the autostart doesn't happen. I enabled the script and refreshed the daemon.

paddywwoof, 2021-03-14:
Yes, it does work
```bash
sudo nano /etc/systemd/system/picframe.service
```
```bash
[Unit]
Description=Pi3D on Pi4
After=multi-user.target

[Service]
Type=idle

User=root
ExecStart=xinit /usr/bin/python3 /home/pi/picframe/run_start.py /home/pi/picframe_data/config/configuration.yaml

#Restart=always

[Install]
WantedBy=multi-user.target
```
Assuming you did the default init with ```-i .```

paddywwoof, 2021-03-14:
Yep. That's a bug in the wiki due to copy pasting from nano. That line that ends with a `$` should have the full path to configuration.yaml

I will fix that now!

sapnho, 2021-03-14:
Thanks, Paddy!

But shouldn't "/home/pi/picframe/run_start.py" still be changed to  "/home/pi/picframe_data/run_start.py"?

paddywwoof, 2021-03-14:
Yes. It's a good job you're testing all this! The [instructions here](https://github.com/helgeerbe/picframe/wiki/Running%20As%20Service#running-as-service-on-pi4) should be OK now

sapnho, 2021-03-14:
Yes, indeed, we need to make them foolproof. 

So, I got Pi4 autostart but the Pi3 version doesn't yet work. The instructions are a tad confusing.

> Create ~/.config/systemd/user/picframe.service

means

> sudo nano ~/.config/systemd/user/picframe.service

Because that gives me the error "[ Directory '/home/pi/.config/systemd/user' does not exist ]"

sapnho, 2021-03-14:
These instructions [here](https://www.thedigitalpictureframe.com/how-to-set-up-your-raspberry-pi-for-your-digital-picture-frame/#For_the_Raspberry_Pi_2_and_3-2) worked fine but I did not manage to change them to picframe.

sapnho, 2021-03-14:
BTW, here is the [updated article](https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d).

But the Pi3 autostart still needs fixing. :-)

paddywwoof, 2021-03-14:
Yes the instructions are confusing. Don't use sudo for any of the Pi 3 stuff as it will put files in the wrong location. i.e.~ for sudo is converted to /root/
```Create ~/.config/systemd/user/picframe.service``` should be read as:
```bash
$ mkdir ~/.config/systemd/user/ -p
$ nano ~/.config/systemd/user/picframe.service
```
I will change the wiki for that too.

I admit that I've not actually done a clean install of the Pi 3 as that's Judith's picture frame running on the wall in the kitchen. However I will probably try one now!

sapnho, 2021-03-14:
Still doesn't work for me. The system is enabled, daemon reloaded but it doesn't autostart.

paddywwoof, 2021-03-14:
what do you get with `journalctl` I find systemd completely impossible to debug without using that!

sapnho, 2021-03-14:
> Mar 14 16:31:28 Pi3withPF2021 systemd[619]: picframe.service: Failed to execute command: No such file or directory
Mar 14 16:31:28 Pi3withPF2021 systemd[619]: picframe.service: Failed at step EXEC spawning /usr/bin/picframe: No such file or directory
Mar 14 16:31:28 Pi3withPF2021 systemd[1]: picframe.service: Main process exited, code=exited, status=203/EXEC
Mar 14 16:31:28 Pi3withPF2021 systemd[1]: picframe.service: Failed with result 'exit-code'.

sapnho, 2021-03-14:
I can wait until you have done your Pi3 fresh install. 

But whatever happens, don't mess with Judith's frame! :-)

paddywwoof, 2021-03-14:
That looks a bit odd. Wonder why it's trying to run `/usr/bin/picframe` rather than `/home/pi/.local/bin/picframe` as specified in picframe.service I did also notice in the wiki that it specifies the wrong location for the configuration.yaml file. It should be in `/home/pi/picframe_data/config/configuration.yaml`

I will try a pip install on the Pi 3 in the kitchen and get back to you. (I will make sure I don't break it)

Paddy

helgeerbe, 2021-03-14:
That's odd. What does your `picframe.service` look like?

sapnho, 2021-03-14:
[Unit]
Description=PictureFrame on Pi3

[Service]
ExecStart=/home/pi/.local/bin/picframe ~/picframe_data/config/configuration.yaml
Restart=always

[Install]
WantedBy=default.target

helgeerbe, 2021-03-14:
That's exactly, what I have on my living room frame.

What do you get when typing on console: `which picframe`
I would expect: `/home/pi/.local/bin/picframe`

helgeerbe, 2021-03-14:
On pi3 you you are using systemd user service. This requires (only on pi3) the  `-- user`  option!!
- Enable service `systemctl --user enable picframe.service`
- Start service `systemctl --user start picframe.service`

paddywwoof, 2021-03-14:
OK, I've gone through this with clean install and it's running... it was picframe/config/configuration.yaml in the wiki which I just copy/pasted (having talked about that only a few seconds before!) I've fixed that (and corrected wiki) and it's running OK

Wolfgang, the fact that journalctl is showing /usr/bin/picframe makes me think that maybe a sudo crept in at some stage. Try disabling (with sudo) any picframe.service that might be in `/etc/systemd/system/` just in case there's something there from before.

sapnho, 2021-03-14:
Yes, it works now, thank you all.

I ended up making a fresh install, double-checking my instructions in the process, and I think in the end it may have been
> systemctl --user enable picframe.service

Great!

sapnho, 2021-03-14:
I believe this one for Pi4 in the Wiki

> ExecStart=xinit /usr/bin/python3 /home/pi/picframe/run_start.py /home/pi/picframe_data/config/configuration.yaml

also needs cleaning up to "picframe_data/.

paddywwoof, 2021-03-14:
Thanks. Hopefully that's all of them!

jgodfrey, 2021-07-05:
Can this one be closed?

sapnho, 2021-07-05:
It works fine now, yes.

closed 2021-07-05 by helgeerbe