helgeerbe/picframe #74: USB Stick

sapnho, 2021-03-15:
This just came in from a reader:

> Great software but I wonder whether you might have come across this problem?
 
> I set up a simple photo frame using your recently revised instructions and put my photos onto a USB Stick.
> I set up a mount for the stick as /media/USBPhotos and set this to be automatically mounted on reboot
 
> Having done this, I was pretty happy and rebooted a couple of times to see that it was all working.
 
> I then made the mistake of pulling the USB stick out of the Pi while the PI and the software were running. OK, it loaded the default ‘Picture missing’ stuff but didn’t start reading from the USB stick when I plugged it back in. I have since rebooted several times, checked all the settings for mounting the USB stick and everything else I can think of and it still does not load the photos from the USB stick.
 
> What is stranger is that I can read the files from the USB stick. I have copied one of them over to the Pi and I have full access to the stick, as far as I can see.
 
> Is it possible that the database is corrupt?
> Where can I find a log file to see if there is a warning that I specific file is corrupt?

I suppose that is the same bug that Jeff is currently fixing, is it?

jgodfrey, 2021-03-15:
Hmmm... The issue you reported (Wolfgang) should be fixed now, though only in dev. That said, I'm not sure whether this is the same issue or not. I guess you could advise the user to rename the db file (not delete) so it creates a new one. If that works, I wouldn't mind getting access to the offending db file if possible...

jgodfrey, 2021-03-15:
One other thought would be to set the `log_level` all the way down to `DEBUG` to see if anything interesting is kicked up. Though, I guess that'll be output to a console by default.

sapnho, 2021-03-15:
I am pretty sure that this is the same issue. As soon as it is deployed to Production, I'll check.

paddywwoof, 2021-03-15:
And it might be interesting to look at the actual db and see what it 'thinks' the situation is w.r.t. files and file locations. Maybe you (Wolfgang) could ask your reader to take a copy and, ideally, send it in.

paddywwoof, 2021-03-15:
I might simulate the same thing here and see what I get!

paddywwoof, 2021-03-15:
Hmm, I can't reproduce this. I've:
```shell
$ cd ~
$ mv Pictures xPictures
$ cp picframe_data/data/pictureframe.db3 pictureframe1.db3 # etc at various intervals
```
However the 'no pictures' image doesn't load - it just sticks with the images that was showing. And the db doesn't seem to be cleared out. When I stop and restart picframe it does show 'no pictures' and the db gets cleared (all tables apart from location). But if I mv `xPictures` back to `Pictures` it goes back to normal at the next `Controller.next()` So maybe the db has become corrupted in some way, perhaps the directory has to disappear at exactly the right moment during the ImageCache processing.

One thing that might be relevant: I always shut the app down in a controlled way (http interface stop) and avoid using CtrlC (or maybe `systemctl stop`) as that could well occasionally corrupt things if there is db writing going on. Is is worth finding out from the reader how the program was stopped?

sapnho, 2021-03-15:
Hi @paddywwoof, yes the shutdown is via CTRL-C.

paddywwoof, 2021-03-15:
It could be related to that then, though Helge's system to catch CtrlC ought to make it close in a controlled way. I only have a relatively small number of images so the indexing and deleting probably happen very quickly which might be a factor it the corruption is caused by something tripping the db up mid-process.

jgodfrey, 2021-03-19:
Thinking more about this, I don't see any connection between the `Pictures` folder disappearing (for what ever reason) and the database becoming _physically_ corrupted. If the original issue really was database related (which, I guess we don't know for sure), I'd think it's more related to a bug in our caching code than an actual corruption in the underlying database.

paddywwoof, 2021-03-19:
I agree. I don't think the folder disappearing should corrupt anything but it might take a longish time for the records to be deleted from, and re-added to the database, during that time stopping the program by CtrlC might do something unexpected... Maybe!

sapnho, 2021-03-19:
I had an email earlier from reader Michael telling me that the indexing of the images never completes. He should post here later today.

jgodfrey, 2021-03-19:
A completely wild guess, but I seem to remember the issue I fixed [here](https://github.com/helgeerbe/picframe/issues/72) was also causing the indexing to never complete. I wonder if it could be related?  

Helge - I'm not sure where you are with packaged releases but is that fix available yet?

sapnho, 2021-03-19:
That may be a solution. But I suppose that hasn't been pushed to production yet, has it?

jgodfrey, 2021-03-19:
> That may be a solution. But I suppose that hasn't been pushed to production yet, has it?

Yeah, I'm not sure. That's the reason I asked this above...

> Helge - I'm not sure where you are with packaged releases but is that fix available yet?

paddywwoof, 2021-03-19:
The version installed on my RPi is '2021.03.10' (latest on pypi.org) and it doesn't have this change to image_cache. It's probably worth accumulating a few fixes then upgrading the version.

michl270, 2021-03-19:
Dear All,

at first I want to thank you guys very much for your great work. I relly appreciate it :-)

Wolfgang already mentioned my issue regarding the indexing of my images which will not complete.

I have just performed a quick test to check if the exif data are causing the issue. I have removed the exif information from my image, renamed the database-file on my RPI and rebootet afterwards. Now indexing is working properly.

Thank you guys for your assistance!

Have a nice weekend!

helgeerbe, 2021-03-20:
> Helge - I'm not sure where you are with packaged releases but is that fix available yet?

Sorry, was away a couple of days. I will release a new version today.

helgeerbe, 2021-03-20:
Just released version `2021.03.20` [changelog](https://github.com/helgeerbe/picframe/wiki/Changelog).

closed 2021-04-15 by helgeerbe