helgeerbe/picframe #75: MQTT command to just show "date location"

sapnho, 2021-03-16:
How would I best send an MQTT command to show me the date and location info for 30 seconds?

I usually turn off the info text but would like to call it up, typically via Alexa command, to show me the date and location.

In the old 2020 version, I used a workaround (which wasn't very reliable and took 5+ seconds), where I first sent a command to show the date and then a second to show the location.

I wonder if there is a better way in this great new version?

helgeerbe, 2021-03-16:
Just send the mqqt  [commands.](https://github.com/helgeerbe/picframe/wiki/Control%20Interfaces) PictureFrame blends it slowly in. You wouldn't notice that you're sending a couple of commands. After 30 seconds send the all text off command.

From Homeassistant this works well for me. I just use alexa to turn the display off, or to do text to speach for image location.

sapnho, 2021-03-27:
Yep, we can close this issue. Sending two commands is a lot faster now indeed, so @helgeerbe 's solution works just fine.

closed 2021-04-08 by helgeerbe