helgeerbe/picframe #76: Another USB stick question

sapnho, 2021-03-16:
How could we make it so that when a USB stick is inserted, PictureFrame would only show the images on that stick and after it has been removed, would go back to the images on the SD card?

I get this question every now and then and maybe the new architecture allows this?

paddywwoof, 2021-03-16:
I suppose the new architecture might allow an additional interface class to monitor the drive and change the pic_dir held in Model. I've not looked to see what would be involved but you're right it might be easier to do now.

sapnho, 2021-03-26:
I got another request for this feature from another reader recently. The use case was PictureFrame in a hotel lobby and putting the images on a USB stick would make things a lot easier. I also had it for an in-shop display.
I think it would indeed be quite nice to have this integrated into PF2021.

paddywwoof, 2021-03-26:
I will have a look later at how this might be done. I've not  used it but pyudev looks relevant. We also need to sort out the db cache system so the root pic_dir can be changed without triggering a clear out of files. Obviously, as it stands, if a USB drive is removed then all the file paths will no longer exist. I suppose a USB drive might have the same name as another, even with some folder and file names being the same but actually different images. I don't know sophisticated we need to make our system.

Paddy

jgodfrey, 2021-03-26:
To support this, I think the cache system will need to stop being so aggressive about removing _apparently missing_ images... Maybe we could add an `is_available` (or similar) boolean field to the `folder` table that could be easily switched as folders become available / unavailable. Then, rather than removing data for files associated with _missing_ folders, we could update the query code in `query_cache()` to only return files where the containing folder `is_available`. 

With that, we could still get a _reasonably accurate_ (for most conditions) playlist from `query_cache()` as folders appear / disappear and rely on the existing _does this file exist_ check at display time to handle any temporary discrepancies. Also, it'd allow us to keep data for files we've already collected, and "restore" it when its folder became available again.

We'd probably still want to provide a way to actually remove cache data for files that really are missing. I guess that could be triggered by some sort of maintenance command line switch, or an MQTT message, or ???

And, I really don't think we need to be overly concerned about the db containing stale data for some temporary period of time (or permanently, really). It'll definitely handle it well...

None of this seems too difficult - though it changes the cache build logic... 

And, it doesn't speak to the issue of detecting volume availability over time (though, maybe Paddy's `pyudev` mention is relevant here)...

sapnho, 2021-03-27:
> To support this, I think the cache system will need to stop being so aggressive about removing apparently missing images...

This could be also very interesting for users who connect their NAS to PictureFrame as we had it frequently.

helgeerbe, 2021-03-27:
So 
- folders get a `last_seen` tag
- we define a garbage collector duration, or set it to Null to keep the folders for ever
- and implement a clean cache command with a date_before (argument)

Just wonder if it really necessary to change the root directory? Wouldn't it be easier to link/mount usb devices, Nas/smb devices below our picture root? So we can handle them as normal folders? Changing root should result in a clean cache then

jgodfrey, 2021-03-27:
> Wouldn't it be easier to link/mount usb devices, Nas/smb devices below our picture root? So we can handle them as normal folders?

Yeah, that sounds like a reasonable restriction. I wonder what others think?

paddywwoof, 2021-03-27:
I think that would be nicer: create `~/Pictures/usbdrive` and automatically mount/umount the drive when plugged in or ejected. (Possibly have usbdrive1, usbdrive2 etc)

I'm imagining that a lodge in a safari park might have a picture frame running and the guide would get the tourists' best pictures each day and put them on a USB. Each evening over dinner they can marvel at the amazing shots which appear first then all the previous images on the SD card. The standard photo naming makes it very unlikely that the same image name will crop up more than once. Every now and then the database would get a spring clean to clear out the non-SD card image records.

sapnho, 2021-03-27:
That sounds like an excellent idea!

jgodfrey, 2021-03-27:
> I think that would be nicer: create `~/Pictures/usbdrive`

Agreed, this sounds good. And, really, it's just a specific (and automated) case of what Helge mentioned above. That is, to always mount external devices / folders as subfolders of the known, root Pictures folder.

paddywwoof, 2021-03-27:
I suppose it would also be possible to add and remove symbolic links to different folders on the SD card in the same way for temporarily showing some pictures not inside the standard root directory structure. Mainly for development and testing I suppose.

helgeerbe, 2021-03-27:
@paddywwoof  it is.
Just checked, that it is possible to write a systemd service, which creates a link, after mounting of a device. So this can be handled outside of PictureFrame. Don't know if all sticks or just specific (with a given UUID) should be linked. I will test this.

helgeerbe, 2021-03-28:
OK, this worked for me.
Precondition is boot to console (booting to Desktop will lead to an automount of usb sticks)

````
systemd-escape --suffix=mount --path <mountpoint> 
````

Shows you how to name your systemd files. This naming scheme is important!

My mount point is '/home/pi/Pictures/usb'

````
systemd-escape --suffix=mount --path /home/pi/Pictures/usb
home-pi-Pictures-usb.mount
````

Create 'sudo nano /etc/systemd/system/home-pi-Pictures-usb.mount'

````
[Unit]
Description=Mount USB

[Mount]
What=/dev/sda
Where=/home/pi/Pictures/usb
Type=vfat

[Install]
WantedBy=multi-user.target
````

`sda` is my usb stick. You can check it with `lsblk`

Create 'sudo nano /etc/systemd/system/home-pi-Pictures-usb.automount'

````
[Unit]
Description=Automount USB device

[Automount]
Where=/home/pi/Pictures/usb
#unmount if longer than 20 sec unavailable
TimeoutIdleSec=20

[Install]
WantedBy=multi-user.target
````

Run `sudo systemctl daemon-reload`

Enable automount `sudo systemctl enable home-pi-Pictures-usb.automount` and start it `sudo systemctl start home-pi-Pictures-usb.automount`

If you plug in your stick, a folder under `/home/pi/Pictures/usb` appears, which contains the data on the stick. If you unplug the stick the 'usb' folder remains, but is not accessible. So every time you plugin a stick, the images would be under the 'usb' folder.

**Remark**
It's important, that the '.mout' service is not enabled. Otherwise the raspbery tries to mount the stick on boot. That is OK, if you plan to use it as a permanent device. But '.automount' needs '.mount'.  And for permanent mounted devices, you wouldn't need of course the '.automount' file.

Would be good, if you can test this.

closed 2021-04-15 by helgeerbe