helgeerbe/picframe #77: Fading mosaic transitions

sapnho, 2021-03-16:
@paddywwoof: You mentioned some time ago that Pi3D may be able to do Mosaic transitions (e.g. like here https://www.youtube.com/watch?v=9b8VaJUXD30).

Now that we have the grand new architecture, would new transitions be thinkable without too much trouble?

paddywwoof, 2021-03-16:
Hi, I'm not sure the new architecture will make a difference, but I was actually thinking about new transitions a few days ago. The simple fade from one image to another doesn't really use much of the OpenGL capacity - there are, after all only four vertices and two polygons to render!

The two ideas I had were 1) use a third greyscale texture for the transitions i.e. the second image would appear like rain or snow flakes gradually falling more opaquely. 2) use second Shape to draw the appearing image in front of the old image - like the mosaics. This option would require two objects to be drawn with a simple shader rather the present system of one object using a slightly more complicated shader.

I might do an experiment if I get chance later this week.

sapnho, 2021-03-17:
> The simple fade from one image to another doesn't really use much of the OpenGL capacity - there are, after all only four vertices and two polygons to render!

You make it sound so easy! For me, it's still a miracle!

> The two ideas I had were 1) use a third greyscale texture for the transitions i.e. the second image would appear like rain or snow flakes gradually falling more opaquely. 2) use second Shape to draw [...]

Sounds mesmerizing! I guess what would be helpful is to preserve the smoothness of Pi3D's transitions so that there isn't too much disturbance in the room coming from a sudden change in brightness or color.

paddywwoof, 2021-03-18:
If you put this in a test.py file in the pi3d_demos folder it will show the beginnings of the second idea
```python
import numpy as np
import random

import demo
import pi3d

class Shard(pi3d.Shape):
    """ 3d model inherits from Shape, differs from Sprite in being two sided"""
    def __init__(self, camera=None, light=None, name="", corners=((0.0,0.0),(0.0,1.0),(1.0,1.0),(1.0,0.0)),
                uvs=((1.0,1.0),(1.0,0.0),(0.0,0.0),(0.0,1.0)),
                x=0.0, y=0.0, z=0.0,
                rx=0.0, ry=0.0, rz=0.0,
                sx=1.0, sy=1.0, sz=1.0,
                cx=0.0, cy=0.0, cz=0.0):
        """uses standard constructor for Shape extra Keyword arguments:

        *corners*
            x,y locations of corners
        *uvs*
            texture coords matching each corner
        """
        super(Shard, self).__init__(camera, light, name, x, y, z, rx, ry, rz,
                                    sx, sy, sz, cx, cy, cz)

        verts = [list(corners[i]) + [0.0] for i in (0, 1, 2, 3, 0, 1, 2, 3)]
        norms = [[0.0, 0.0, -1.0] for _i in range(8)]
        texcoords = [list(uvs[i]) for i in (3, 2, 1, 0, 3, 2, 1, 0)]
        inds = [[3, 0, 1], [1, 2, 3], [7, 6, 5], [5, 4, 7]]

        self.buf = [pi3d.Buffer(self, verts, texcoords, inds, norms)]

        self.xr = 90
        self.yr = 90 - random.random() * 180
        self.zr = 90 - random.random() * 180
        self.rotateToX(self.xr)
        self.rotateToY(self.yr)
        self.rotateToZ(self.zr)
        self.xend = x
        self.yend = y
        self.xstart = x * (1.0 + 2.0 * random.random())
        self.ystart = y * (1.0 + 2.0 * random.random())
        self.positionX(self.xstart)
        self.positionY(self.ystart)

    def regroup(self, factor=0.0):
        self.rotateToX(self.xr * factor)
        self.rotateToY(self.yr * factor)
        self.rotateToZ(self.zr * factor)
        self.positionX(self.xend + (self.xstart - self.xend) * factor)
        self.positionY(self.yend + (self.ystart - self.yend) * factor)

class Mosaic:
    def __init__(self, w=500.0, h=300.0, divisions=4, x=0.0, y=0.0, z=5000.0):

        n = divisions + 2
        spread = 1.0 / (divisions + 1.0)
        xpts = np.linspace(0.0, 1.0, n).reshape(-1, 1) + (np.random.random((n, n, 1)) - 0.5) * spread
        xpts[:,0,0] = 0.0
        xpts[:,-1,0] = 1.0
        ypts = np.linspace(0.0, 1.0, n).reshape(-1, 1, 1) + (np.random.random((n, n, 1)) - 0.5) * spread
        ypts[0,:,0] = 0.0
        ypts[-1,:,0] = 1.0
        self.coords = np.concatenate((xpts, ypts), axis=2) # or do other way round
        offsets = ((1,1),(0,1),(0,0),(1,0))

        self.empty = pi3d.Triangle(corners=((0,0,0),(1,1,0),(2,0,0)), x=x, y=y, z=z)

        for j in range(divisions + 1):
            for i in range(divisions + 1):
                uvs = [list(self.coords[j+joff, i+ioff]) for (ioff, joff) in offsets]
                corners = [[(u - 0.5) * w, (0.5 - v) * h] for (u, v) in uvs[::-1]]
                x = sum(c[0] for c in corners) / 4.0
                y = sum(c[1] for c in corners) / 4.0
                corners = [[cx - x, cy - y] for (cx, cy) in corners]
                s = Shard(corners=corners, uvs=uvs, x=x, y=y)
                self.empty.add_child(s)

display = pi3d.Display.create(frames_per_second=20)
camera = pi3d.Camera(is_3d=False)
tex = pi3d.Texture("textures/photosphere.jpg")
shader = pi3d.Shader("uv_flat")

m=Mosaic(w=tex.ix, h=tex.iy, z=5000, divisions=6)

for s in m.empty.children:
    s.set_draw_details(shader, [tex])

for i in range(400):
    display.loop_running()

    m.empty.draw()
    if i <= 300:
        for s in m.empty.children:
            s.regroup((300.0 - i) / 300.0)
```
How it fits in with the border mat, text or anything else I haven't thought!

Paddy

paddywwoof, 2021-03-19:
Like this ([youtube video](https://youtu.be/ZgnhI1Ek-7g))

sapnho, 2021-03-19:
That's a good start! Would it be possible to crossfade these mosaics?

paddywwoof, 2021-03-19:
So the alpha value gradually increased? Like
```python
    ...
    def set_alpha(self, alpha):
        for s in self.empty.children:
            s.set_alpha(alpha)

display = pi3d.Display.create(frames_per_second=20)
camera = pi3d.Camera(is_3d=False)
...
    m.draw()
    if i <= 300:
        factor = (300.0 - i) / 300.0 # goes from 1.0 to 0.0
        m.regroup(factor)
        m.set_alpha(1.0 - factor)
```
[image]
[image]

sapnho, 2021-03-19:
I need to reinstall the old PictureFrame2020 version again to try this. Is alpha the transparency setting?

paddywwoof, 2021-03-19:
Wolfgang. That's just a stand-alone script. If you don't have anything installed from pi3d_demos it will still work, and you can put it anywhere, but you need to change the path in `tex = pi3d.Texture("textures/photosphere.jpg")` i.e to `/home/pi/Pictures/IMG_20210106_121033.jpg`

closed 2025-04-15 by github-actions[bot]