helgeerbe/picframe #78: Issue with Home Assistant integration

jeffalperin, 2021-03-19:
Following the instructions in the wiki, I'm mostly successful in integrating picframe with Home Assistant, but am seeing some errors (even though picframe appears to be functioning correctly):

 After initiating picframe from the command line with this command:
`export DISPLAY=:0 && /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml`
I get errors in the console which look like this:
`192.168.7.48 - - [19/Mar/2021 13:07:02] code 400, message Bad request version ('\x10Y9\x90ÜÓ\x06\x07ï¡2âx}')
192.168.7.48 - - [19/Mar/2021 13:07:02] code 400, message Bad request version ('£ÌbQÉ}¦\x0e;_\\Ár\x89ºÅHºJR*\x03\x01©\x16')
192.168.7.48 - - [19/Mar/2021 13:07:02] code 400, message Bad request version ('ç\x81\x00$\x13\x01\x13\x03\x13\x02À+À/Ì©Ì¨À,À0À')`
(The .48 IP address is for the Windows desktop PC which is accessing Home Assistant via browser.)

Any ideas of what to look at to resolve this?

helgeerbe, 2021-03-20:
Hm, looks for me the url encoding in home assistant is not working probably. 

Do you see the image in home assistant?

Under developer tools, there is a template editor in home assistant. When you paste `![Image](http://192.168.7.48:9000/{{ states('sensor.picframe_image') | urlencode }})` What do you  get for an url? Can you copy this url to your browser. Does the image show up?

One other thing. Check if you are using ssl. If you are using ssl you have to use `https`  otherwise 'http'

jeffalperin, 2021-03-20:
I do **not** see the image in Home Assistant.  (I was going to inquire about that.)  Also, I am **not** using SSL.  

When I paste the template code into the template editor, the resulting URL `![Image](http://192.168.7.48:9000/Canadian%20Rockies_DSC_0614_3840x2160.jpg)` does **not** work, but that's because 192.168.7.**48** is the IP of my Windows desktop where I'm running HA in a browser.  The rPi4 running `picframe` is at 192.168.7.**50**.  When I change the IP to .50, I **do** see the image in the browser.

So, where is the .48 IP coming from?  (The error messages being thrown use that IP, too.)

Thanks for your help!

helgeerbe, 2021-03-20:
Ah OK, that makes sense. .48 is logged where the request comes from. So to sort it out for me.

1. mqtt is up and running, because you see the image name in home assistant.
2. In the markdown card you have set in the yaml code `http://192.168.7.50:9000`. The ip from your picframe
3. home assistant **does not use ssl** (`http://myhomeassistant:8123/lovelace`). If you set up home assistant to use ssl it will always use ssl to query the image. No matter if you use 'http' or 'https' in your image url.
4. ssl is off on picframe

Can you post your yaml code? Shoud look like this:

````YAML
cards:
  - type: markdown
    content: >-
      ![Image](http://192.168.7.50:9000/{{
      states('sensor.picframe_image') | urlencode }})
````

I deleted the rest in the card.

jeffalperin, 2021-03-20:
Your 1 -4 are all correct.

I discovered the problem preventing the image from showing in Home Assistant:  I had http**s** instead of just http for the url in the yaml.  I've corrected that so the current yaml is:

```yaml
cards:
  - type: markdown
    content: >-
      {% if state_attr("sensor.picframe_image", "IPTC Object Name") -%}

      ## {{ state_attr("sensor.picframe_image", "IPTC Object Name") }}
       {%- else -%}
      ## {{ states("sensor.picframe_image") }}

      {%- endif %}

      {% if state_attr("sensor.picframe_image", "IPTC Caption/Abstract") -%}
         {{ state_attr("sensor.picframe_image", "IPTC Caption/Abstract") }}
      {% endif %}

      ![Image](http://192.168.7.50:9000/{{ states('sensor.picframe_image') |
      urlencode }})

      {% if state_attr("sensor.picframe_image", "location") != None %}
        {{ state_attr("sensor.picframe_image", "location") }}
      {%- else -%}
        No GPS Data.
      {%- endif %}
  - type: glance
    entities:
      - entity: switch.picframe_back
        tap_action:
          action: toggle
      - entity: switch.picframe_paused
        tap_action:
          action: toggle
      - entity: switch.picframe_next
        tap_action:
          action: toggle
      - entity: switch.picframe_shuffle
        tap_action:
          action: toggle
      - entity: switch.picframe_delete
        tap_action:
          action: toggle
    show_name: false
    show_state: false
    state_color: true
    show_icon: true
  - type: entities
    entities:
      - entity: sensor.picframe_image
        name: Filename
      - entity: sensor.picframe_image_date
        icon: 'mdi:calendar-clock'
        name: Date
      - type: attribute
        entity: sensor.picframe_image
        attribute: EXIF ExposureTime
        unit: sec
        icon: 'mdi:camera-timer'
        name: Shutter speed
      - type: attribute
        entity: sensor.picframe_image
        attribute: EXIF FNumber
        unit: f
        icon: 'mdi:camera-iris'
        name: Aperture
      - type: attribute
        entity: sensor.picframe_image
        attribute: EXIF ISOSpeedRatings
        icon: 'mdi:film'
        name: ISO
      - type: attribute
        entity: sensor.picframe_image
        attribute: EXIF FocalLength
        unit: iso
        icon: 'mdi:signal-distance-variant'
        name: Focal length
      - type: attribute
        entity: sensor.picframe_image
        attribute: Image Model
        unit: mm
        icon: 'mdi:camera'
        name: Camera
      - type: attribute
        entity: sensor.picframe_image
        attribute: IPTC Keywords
        icon: 'mdi:tag'
        name: Tags
  - aspect_ratio: '3:2'
    dark_mode: false
    default_zoom: 11
    entities:
      - entity: sensor.picframe_image
    type: map
    hours_to_show: 0
type: vertical-stack

```
I'm still getting errors like this, though:
`
192.168.7.48 - - [20/Mar/2021 12:58:10] code 400, message Bad request version ('3\x10\x00$\x13\x01\x13\x03\x13\x02À+À/Ì©Ì¨À,À0À')
192.168.7.48 - - [20/Mar/2021 12:58:10] code 400, message Bad request version ('à\x17ßUÀ\x00$\x13\x01\x13\x03\x13\x02À+À/Ì©Ì¨À,À0À')
192.168.7.48 - - [20/Mar/2021 12:59:17] code 400, message Bad request version ('\x16ùÊ\x00QõM¨Q¶VOÓ\x92=¹îh¬"6?\x91,\x8b0e\x7f\x00$\x13\x01\x13\x03\x13\x02À+À/Ì©Ì¨À,À0À')
192.168.7.48 - - [20/Mar/2021 12:59:17] code 400, message Bad request version ('\x9dµ\x07\x01\x12È\x7f±¾ØÁßßgm\x86Ë}èås±¨Ùñ*o\x12°\x91\x00$\x13\x01\x13\x03\x13\x02À+À/Ì©Ì¨À,À0À')
`

jeffalperin, 2021-03-20:
Oh, and is there supposed to be an entity created for the MQTT command:

`stop: picframe/stop, message not used - just stops`

or is there documentation for how to implement?  (I assume this exits the current slideshow?)
I can write this up as a separate issue if you'd like me to.

helgeerbe, 2021-03-20:
For me it looks like,  you are doing a ssl request (see [discussion](https://github.com/helgeerbe/picframe/discussions/5#discussioncomment-395004)). I don't now why. This happend on my homeassistant setup, because I enabled ssl in 'configuration.yaml'

sapnho, 2021-03-20:
Same error messages here after activating the HTTP server:

> INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
192.168.178.85 - - [20/Mar/2021 17:05:19] code 400, message Bad request version ('jj\x13\x01\x13\x02\x13\x03À+À/À,À0Ì©Ì¨À\x13À\x14\x00\x9c\x00\x9d\x00/\x005\x01\x00\x01\x93ÊÊ\x00\x00\x00\x17\x00\x00ÿ\x01\x00\x01\x00\x00')
192.168.178.85 - - [20/Mar/2021 17:05:19] code 400, message Bad HTTP/0.9 request type ('\x16\x03\x01\x02\x00\x01\x00\x01ü\x03\x03\x99=[M\x0ey\x8f<¶½U')

helgeerbe, 2021-03-20:
`stop: picframe/stop, message not used - just stops`
No, there is no entity for that. That will top PictureFrame. If you like to use it, you can just publish the topic manually from the mqtt integration. Just use topic: 'picframe/stop' and as message anything you like.

helgeerbe, 2021-03-20:
@sapnho Thanks for the info. I will do a clean setup and check, what's going on.

What version of Home Assistant are you all using? I'm running the latest version `Home Assistant 2021.3.4`

sapnho, 2021-03-20:
I am running core-2021.1.5 because I had issues with Z-Wave when I updated last time so I went back to the previous version.

sapnho, 2021-03-20:
@helgeerbe I also sent you some info via our Slack channel.

jeffalperin, 2021-03-20:
@helgeerbe I'm running **core-2021.2.3** and can't upgrade due to some sort of incompatibility reported by the Check Home Assistant configuration add-in.

helgeerbe, 2021-03-20:
Well I think, the version doesn't matter. When I disable ssl I see the Bad Request in PictureFrame. But the image is loaded in home assistant and shows up.  If I load the image directly from the browser I don't see the Bad Request.

sapnho, 2021-03-21:
Could it have something to do with the fact that I am accessing HA through Nabu Casa? That is a secure connection. On the iPad app, the image is shown.

helgeerbe, 2021-03-21:
Wolfgang, Thanks for the feedback. The image will work only work in your local network. Not sure how Nabu Casa is working. But I doubt, that the local address is resolved correctly by Nabu Casa, if you are not connected to your local network. 

When calling from the iPad do you see the Bad Request on PictureFrame?

sapnho, 2021-03-21:
Ha, that solved the problem! As soon as I used the "http://homeassistant.local:8123/lovelace/6" URL, the picture came up! Thanks, Helge!

helgeerbe, 2021-03-21:
Great. I'm just wondering if you see still any bad requests on the pictureframe output?

helgeerbe, 2021-03-21:
To make it working with nabu casa, you could use a camera entity in home assistant. So the image is pre-rendered by home assistant. But this will have a lack, when the image changes. So you wouldn't get feedback, when pushing next or prev image button.

sapnho, 2021-03-21:
> Great. I'm just wondering if you see still any bad requests on the pictureframe output?

Nope. No more bad requests.

helgeerbe, 2021-03-21:
Good to know. Thanks for testing.
I turned all encryption off, but still get encrypted requests that result in Bad Requests. I think there is some caching in Home Assistant or the Browser. But I guess, this not worth  to analyse this further.

helgeerbe, 2021-03-21:
Do a lot of people use Nabu Casa? I just open an vpn from my smartphone and use the home assistant app in my local environment. So I don't have the need to use any cloud service.

helgeerbe, 2021-03-21:
@jeffalperin do you use something like Nabu Casa?

sapnho, 2021-03-21:
> Do a lot of people use Nabu Casa? I just open an vpn from my smartphone and use the home assistant app in my local environment. So I don't have the need to use any cloud service.

I think Nabu Casa is very popular but as long as it remains in the home network, there is no real difference, so you can just use the normal URL.

I use it because it allows me to use Amazon Alexa with HA.

helgeerbe, 2021-03-21:
Ah OK. I wrote my own  home automation skill for Alexa integration and run it in within my developer account on AWS. But mainly to find out, how things work.

helgeerbe, 2021-03-21:
@sapnho To use it within nabu casa in you lokal network. You could use a selfsigned certificate and enable ssl in the server. But this would be a complete new article on your blog 😄

jeffalperin, 2021-03-21:
@helgeerbe I am very new to Home Assistant and am trialing Nabu Casa for remote access and access to Alexa.  From my reading in reddit and elsewhere, it seems to be quite popular.

I can confirm that when I access from outside my local network (via Nabu Casa), I do **not** see the image, but I **do** see the image when accessing from within the local network.

Now the strange thing:  When I started picframe this morning, I'm no longer getting the Bad Request messages in the console, and I am seeing the images in Home Assistant.  I haven't changed anything, so I'm puzzled.

helgeerbe, 2021-03-21:
@jeffalperin Thanks for the feedback. Glad to hear, that it works now. I think, that there might be some caching issue within the browser. 

For remote access within Nabu Casa,  the only solution I can think off, is to setup a template camera entity in home assistant's configuration.yaml.  So the picture frame appears like an ip camera to home assistant.  But there is delay when the image change. So you wouldn't have the direct feedback when pushing the next or prev button.

sapnho, 2021-03-21:
> @sapnho To use it within nabu casa in you lokal network. You could use a selfsigned certificate and enable ssl in the server. But this would be a complete new article on your blog 😄

I am all up for it! :-)

sapnho, 2021-03-21:
Thanks, Helge! https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant

helgeerbe, 2021-03-22:
I would close this issue, if everything else is up and running?

closed 2021-03-22 by helgeerbe