helgeerbe/picframe #79: Quits with segmentation fault

KDPate, 2021-03-19:
Running on a Pi 4 pulling pics off off of a samba share on another pi. It works fine for a few pics (unclear how many at this point, maybe 10?) and the quits with a "segmentation fault" Below is literally everything that is shown in the terminal.

pi@raspberrypi4:~ $ picframe ~/picframe_data/config/configuration.yaml
INFO:start.py:starting ['/usr/local/bin/picframe', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Segmentation fault

Any ideas what is happening or for further troubleshooting? Need mode details?
Thanks!

jgodfrey, 2021-03-19:
I don't know that it'll be helpful, but do you see any more info in the terminal if you turn on `DEBUG` level logging? To do that, edit the `configuration.yaml` file and set the `log_level` property to `"DEBUG"`.

Additionally, I wonder what happens if you limit the set of images to something that you know displays correctly. So, for instance, the 10 or so you see before it seg-faults. Is it able to show that set of images repetitively or does that also seg-fault?

jgodfrey, 2021-03-19:
One  more thought, does it work correctly if the images are on the local RPi? Really, just trying to narrow down the problem...

paddywwoof, 2021-03-19:
seg fault not good. Can you tell what size the images are that might cause the crash. Aslo what memory split you have GPU/CPU

KDPate, 2021-03-19:
A 2G Raspberry Pi 4, 256M assigned to GPU.
Seg faults on a different file each time.
The size of the pics causing the fault is not consistent. One 2 meg file did it, another 600k file did it. File size averages 2 meg, biggest is 5 meg.

Logging show this:
........INFO:interface_mqtt.InterfaceMQTT:Send state: {'directory': 'picframe', 'image_counter': '45', 'image': 'IMG_4028-COLLAGE.jpg', 'date_from': -2147356800, 'date_to': 2145945600, 'time_delay': 30.0, 'fade_time': 8.0, 'brightness': 1.0, 'location_filter': '', 'tags_filter': ''}
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0xaddc5970>
DEBUG:pi3d.Texture:Loading ...PIL.Image (RGB)
Segmentation fault

Will try local files only. Will try limited file set.
Thanks

paddywwoof, 2021-03-20:
It sounds like an issue Jeff had a while ago https://github.com/pi3d/pi3d_demos/issues/52 are any of those features relevant? Critically turning off resize

KDPate, 2021-03-20:
That was exactly the problem. 
auto_resize: True 
and the seg faults have stopped. Thanks for the pointer and I should have done a better job of searching for previous reports of the issue!

closed 2021-03-20 by KDPate

paddywwoof, 2021-03-20:
I'm a bit behind hand with the FAQ - there are a few things that need to go in and maybe that would help questions pop up on google.

Anyway, glad you sorted it. But it does mean there is a lack of flexibility in so far as you either need to *just* show 4k images that have been resized to the correct dimensions with resize off or else have mixed images and keep resize on. Maybe I should make it so that you can do something like:
```yaml
  auto_resize_if_width_not: 3840                       # default=1920, set this width of 4K resolution on Raspberry Pi 4 *if* using an Pi 4 *and* you have resized 4k images to match this width.
```
Need to think about this.