helgeerbe/picframe #81: Change auto_resize from simple boolean

paddywwoof, 2021-03-20:
At the moment if auto_resize is set to False to allow 4k images to render at full resolution then all the images in the slideshow must be resized to the correct 4k size supported by the Raspberry Pi. Possibly this should be done automatically by detecting the type of Pi and if Pi4 adding 3840 to the end of `pi3d.Texture.WIDTHS`

This change might be best done in pi3d and just take the auto_resize option out of config

jgodfrey, 2021-03-21:
Yeah, agreed. I think this is a bit opaque to an end user right now anyway, so eliminating it would be a win I think.

helgeerbe, 2021-03-21:
I like that.

paddywwoof, 2021-03-23:
I think [these changes to pi3d](https://github.com/tipam/pi3d/compare/5120414..3c40ef2) make the need to turn automatic resize off, effectively disappear. I will put this into the next release of pi3d quite soon, possibly with the FixedString pixel width justification mod and after relevant testing.

jgodfrey, 2021-07-05:
Can this be closed?

paddywwoof, 2021-07-05:
Yes, it has been superseded by the change to pi3d

closed 2021-07-05 by paddywwoof