helgeerbe/picframe #82: Is Location indexed at the start?

sapnho, 2021-03-21:
I was wondering how the location filter works? I noticed that when I entered "France" it showed me only a few of my "France" images and this morning it was many more.

I guess the location is only added to the database once it has been shown. So for the location filter to work properly, you have to have a least one complete run, right?

sapnho, 2021-03-21:
And are keywords parsed in all files at the start or is it the same logic as locations?

paddywwoof, 2021-03-21:
The exif info is read quite quickly but not instantaneously on first start - maybe by the time two or three images have shown the keywords, title, lat, lon etc will be available. However the select list will only be refreshed once all the images have been shown [n times]. As you say the location description is only downloaded from Nominatim when the image is actually shown, however this is based on lat, lon to four decimal places so you could well have multiple images that all get location description at the same time as soon as one of them is looked up.

Two things occur to me:
a) we could have an additional category in `ImageCache.__update_cache()` of records with lat, lon but not yet description. These could be 'ticked off' one at a time as `__update_cache` runs once every two seconds and Nominatim have a maximum query rate of one per second. 
b) we could re-use the k-means clustering method used for the mat colour to group and order images by proximity

sapnho, 2021-03-21:
> b) we could re-use the k-means clustering method used for the mat colour to group and order images by proximity

That sounds intriguing. So, we could specify a radius around a location?

sapnho, 2021-03-21:
> a) we could have an additional category in ImageCache.__update_cache() of records with lat, lon but not yet description. These could be 'ticked off' one at a time as __update_cache runs once every two seconds and Nominatim have a maximum query rate of one per second.

Sounds like a solution.

closed 2025-04-15 by github-actions[bot]