helgeerbe/picframe #83: Remote control matting on/off

sapnho, 2021-03-21:
It would be very neat if matting on/off was one day included in the mqtt remote control settings.

paddywwoof, 2021-03-21:
I don't think this would be difficult to do. It just needs a method  `ViewerDisplay.set_mat_image_control_values()` (or maybe just modify and reuse the exisiting `__get_mat_image_control_values()`) and a method in `Controller`. And methods in interface_mqtt and interface_http! So, easy to do but quite a few files updated, maybe roll it into an update with other things and make sure it's tested.

sapnho, 2021-03-21:
Great, thanks. Obviously not urgent, but nice to have.

sapnho, 2021-03-22:
And to build on this enhancement wishlist, being able to remotely change the matting style via a dropdown menu in Home Assistant would be swell! :-)

helgeerbe, 2021-11-04:
Included in release 2021.11.04

closed 2021-11-04 by helgeerbe