helgeerbe/picframe #84: Matting Proportions

sapnho, 2021-03-21:
Maybe it's a feature that I don't yet understand and not a bug but I was wondering how the proportions in matting work. 

I have a 1920x1200px frame and an 1920x1200px image. When I activate matting with double_float, I expected that the matting was proportional to the image size. But it looks like this where the distances to the edge are uneven:
 
[image]

paddywwoof, 2021-03-21:
If the distances to the edge were the same top v. sides then you would have to crop or fill. i.e. 50 pixels all round would leave 1820x1100 so your image would be scaled to either 1820x1137.5 (cropped) or 1760x1100 (gaps)

jgodfrey, 2021-03-21:
Yep, Paddy's correct. If you have an image that matches the aspect of your display, and then add a consistent border around it, the resulting image no longer matches the display's aspect ratio. That leads to the "uneven" borders. So, not a bug.

sapnho, 2021-03-21:
Yes, math is hard to beat! :-)

sapnho, 2021-03-21:
> PS did you specify that mat colour - it doesn't look like one that would be chosen by the algorithm?

No, I didn't specify it. Sometimes it has an artistic mind of its own!

jgodfrey, 2021-03-21:
> No, I didn't specify it. Sometimes it has an artistic mind of its own!

What a great result though. That looks *really* nice!

sapnho, 2021-03-21:
It does. The matting feature is superb!

jgodfrey, 2021-03-22:
Seems like this issue can be closed?

closed 2021-03-27 by jgodfrey