helgeerbe/picframe #85: How often is file count updated?

sapnho, 2021-03-21:
I tried the following: I deleted an image and subsequently moved it back to the Pictures folder. 

When the image was deleted, the image count was immediately reduced by one. When I added it back, it stayed at the lower number. Only when I hit e.g. "directories" in Home Assistant was updated to the correct number.

helgeerbe, 2021-03-21:
In Home Assistant all attributes are updated on image change. But from memory I can't tell, when it's internally updated.

jgodfrey, 2021-03-21:
Looking at the code, it looks like the image count is updated only when the "playlist" is rebuilt from the image_cache. That happens based on a number of different situations.

Related, what is the image counter intended to represent? On the surface, I'd expect it to be the count of images in the Pictures folder. However, it looks like it's a count of the items in the internal playlist. So, a set of paired portraits would only count as 1 it seems. So, if my Pictures folder consisted of only 12 portrait images and they were all paired together, I think the image counter would report 6.

sapnho, 2021-03-21:
It is indeed the count of the items in the internal playlist - which is great because when you set a filter in Home Assistant you can immediately see, how many images correspond to that query.

sapnho, 2021-03-21:
I only have landscape images, so in my case, the number is the same.

jgodfrey, 2021-03-21:
> It is indeed the count of the items in the internal playlist - which is great because when you set a filter in Home Assistant you can immediately see, how many images correspond to that query.

My point is though, with paired portraits, it won't be as you describe above (at least I don't think so from looking at the code). If your filter returned 12 images that were all paired together into 6 displayed "frames", would you expect the image count to be 12 or 6? I think it'd be 6 in that case...

I guess either is arguably correct. You would see 6 individual frames displayed, but they'd be made up of 12 images. I'd think 12 would be the more expected result here.

sapnho, 2021-03-21:
I guess we would just have to make it clear what the playlist count is. One could argue that either way is "correct".

jeffalperin, 2021-03-24:
My preference would be that the image count be the number of **images**, not the number of display screens.  (With my photographer hat on, I think in terms of the images I've created, rather than how they might displayed on screen.)  So, 12 in @jgodfrey example above, not 6.  And I agree with Wolfgang that it's very handy to see how many images correspond to a given set of filters; it's a great feature.

paddywwoof, 2021-03-25:
Getting the number of images would be a simple alteration. However the deletion is a bit woolly when dealing with picture pairs, i.e. the left picture only is moved in the file system but the tuple with both db refs is removed from the list. Not sure if anything more intelligent is feasible.

sapnho, 2021-03-25:
I think it's fine as it is with all images in a playlist being counted. The question was just about the update frequency but I guess as soon as the filter is changed, a new image will be counted.

paddywwoof, 2021-03-25:
The function just needs changing like this
```python
    def get_number_of_files(self):
        #return self.__number_of_files
        return sum(1 for pics in self.__file_list for pic in pics if pic is not None)
        # or, logical to my mind
        #return sum(sum(1 for pic in pics if pic is not None) for pics in self.__file_list)
```
The value this method returns will change the instant the image is deleted but, as Jeff says, there is no mechanism in Model now to check for changes to the db that are not reflected in the last selection (as I think used to happen in PictureFrame2020) Maybe this needs to be added?