helgeerbe/picframe #86: Other than Latin characters in text

sapnho, 2021-03-22:
Would it be possible to show other than Latin languages in the text?

Currently, we are limited to the Codepoints but if Japanese, Chinese, Indian, etc. users wanted to use PictureFrame they could not customize it to their language.

paddywwoof, 2021-03-22:
Wolfgang, I think it will work fine with any set of unicode character in the config file `codepoints` string. That variable is put in the config file precisely so users in different countries can change it to suit their needs.

The limitations are that all the characters are rendered onto a 1024x1024 texture so if you had 10,000 characters in codepoints each would only have 10x10 pixels which might get a bit grainy - probably still readable though. Also there might be a tiny additional delay when the display text is built, as each character in the displayed text is checked to ensure it's in codepoints.

[These 1500 characters](https://sensiblechinese.com/wp-content/uploads/2015/06/Hanzi_WallChart_1500_MostCommonCharacters_SIMPLIFIED.pdf) cover over 95% of Chinese text and would result in resolution 26x26 pixels i.e. each character would be like this
[image]

sapnho, 2021-03-22:
Ah, I didn't know that. Thanks for the clarification, Paddy!

helgeerbe, 2021-03-22:
It works. I have some images from Abu Dhabi. Nice Arabic description beside the English text.

sapnho, 2021-03-22:
I can always see in Home Assistant what text is missing on the frame. But we just have too many characters in the world to add them all..

paddywwoof, 2021-03-22:
If we had used the pi3d.FixedString then all the characters in the font would have been available. If this becomes an issue in future we could switch over.

sapnho, 2021-03-22:
Interesting option! What would change using the pi3d.FixedString?

paddywwoof, 2021-03-22:
At the moment I don't think it does line wrapping automatically. It uses the PIL ImageDraw so I think it would be possible to add in the line length in pixels system, but at the moment it doesn't do that - you have to insert the line breaks manually.

paddywwoof, 2021-03-24:
branch d808059fa58f1b6b9f22a1a92704d47c3f4c31a6 using `pi3d.FixedString` It needs the pi3d develop branch which has additional functionality to set the width of the text.

This method will repeatedly create pi3d.Texture objects, which should have the memory released by the adapted garbage collector system in pi3d but probably worth monitoring this. Also I've only tried this on the relatively small screen of this laptop so I don't know if that's an issue.

Paddy

PS I've not taken out the code that checks character in codepoints but that isn't needed any longer. Will fix that later.

paddywwoof, 2021-03-25:
This has been running for a while now without issue (though I think the char size might be slightly larger?). Also I realized that the whole picframe system relies on creating and destroying Texture instances so that obviously isn't a concern.

sapnho, 2021-03-27:
Just out of curiosity: Will this make it into the next release? :-)

paddywwoof, 2021-03-27:
I would think so. It's been running here for a few days without any problems.

paddywwoof, 2021-03-30:
I've updated pi3d to v2.44 on pypi.org and github and changed the requirement in setup.py I've tested this along with the enhanced filter on different platforms and seems OK so far. Are people happy for me to merge these two branches into dev?

The fixed_string change gets rid of the requirement to specify codepoints - it will just use whatever is in the font supplied (or substituted by the user) and it gets rid of the limits on numbers of characters. In addition the Texture resizing option has changed in pi3d so auto_resize isn't needed. (just noticed that I haven't changed the configuration for those two constants - will fix that now)

I haven't been able to able to test the fixed_string system on a 4k or other odd sized monitor which would be nice to see. It should still work but the Texture used for the 'sprite' to render the fixed string will be forced to width of the [next greater (or equal) power of two below the max size returned by OpenGL](https://github.com/tipam/pi3d/commit/f998c41529c1a1547e093b7e6b32b29d4bcdee41) so there might be a bit of stretch or compression.

jgodfrey, 2021-03-30:
> Are people happy for me to merge these two branches into dev?

I haven't had a chance to look at / try any of the above, but it seems like a reasonable add to the `dev` branch to me. We should probably make it a point to explicitly kick the tires a little harder / longer before merging it into production though...

paddywwoof, 2021-03-30:
Having done "vanilla" pip installs for the released version on the Pi3 and Pi4 (because that's the most important thing to not go wrong) it's a bit messy to switch in other branches. I've been doing it by just copying over the `site-packages/picframe` files but you have to fix a couple of lines in configuration.yaml and, for this change, you have to `python3 -m pip install --upgrade pi3d`

OK I'll merge them, let me know if you spot anything.

helgeerbe, 2021-03-30:
Paddy, I will test it. But to do a clean install from source. Just checkout the branch. Than do `python3 pip install /path/to/picfraem/sources --upgrade`. This will do a clean install. Just from the sources and not from pypi.

helgeerbe, 2021-03-30:
And installing from the sources has the advantage, that pi3d should be automatically updated to meet the requirements.

paddywwoof, 2021-03-30:
Ah yes that's neat. I've merged the two branches into dev now.

helgeerbe, 2021-03-30:
hm, when I search for location `athen` and tags `playmobil` this works. Also `istanbul` and `playmobil`.  I get only image with tag playmobil in the searched city. But  `athen and istanbul` with tags `playmobil` seems not to work. Looks like tag search is ignored.

paddywwoof, 2021-03-30:
Helge, thanks for testing this.

I thought at first I must have copied something and left a "tags" in where it should have been "location" or something, but it looks OK as far as I can tell.

Just to get this straight:
```text
location = "athen", tags = "playmobil" => works as expected
location = "istanbul", tags = "playmobil" => works as expected
location = "athen and istanbul", tags = "playmobil" => wrong results.
```
I would expect zero hits for that last query as it would require a location containing "athen" *and* "istanbul" i.e. the no images selected page. Are you getting all the images instead of none? I think there might be a need for some brackets around all the tags clauses and all the location clauses because of my point below

To get images from both "athen" and "istanbul" I would use
```text
location = "athen or istanbul", tags = "playmobil"
```
did you try that? The problem might be that the resultant SQL would be
```fname LIKE '/home/pi/Pictures%' AND location LIKE '%athen%' OR location LIKE '%istanbul%' AND tags LIKE '%playmobil%'```
which is wrong. It needs to be
```fname LIKE '/home/pi/Pictures%' AND (location LIKE '%athen%' OR location LIKE '%istanbul%') AND (tags LIKE '%playmobil%')```

I will fix that ASAP but will wait for you to clarify what results you get first.

Paddy

helgeerbe, 2021-03-31:
Sorry, might be a bit late yesterday

````
location = "athen" => 12 images
location = "istanbul" => 10 images
location = "athen and istanbul" => 0 images (as expected)
location = "athen or istanbul" => 22 images (as expected)
location = "athen or istanbul" and tags = "playmobil" => 13 images (12 of athen and 1 of istanbul with tag = 'playmobil')
loacation = "istanbul or athen"  and tags = "playmobil" => 11 images (10 of istanbul and 1 of athen with tag = 'playmobil')
````
So I think it is indeed the missing brackets. AND seems to bind stronger than OR

paddywwoof, 2021-03-31:
Helge, yes the more I looked at it the more I could see random results. With dates and subfolders in there as well it would have been quite easy to get very muddled results i.e. where the date only filtered the pictures taken in instanbul

Boolean algebra has a precedence defined as with normal algegbra. Just checking for sqlite makes me think that the evaluation of the NOT clauses might go awry too. As it stands setting `location` to `not germany or berlin` will convert to
`NOT location LIKE '%germany%' OR location LIKE '%berlin%'`
The precedence list I found implies that it will be evaluated NOT before LIKE before AND before OR:
`((NOT location) LIKE '%germany%') OR (location LIKE '%berlin%')`
However that would be very counter-intuitive syntax and, after doing a bit of testing with sqlitebrowser, it certainly seems that it produces the expected interpretation
`(NOT (location LIKE '%germany%')) OR (location LIKE '%berlin%')`
Maybe the parser spots the pattern `NOT colname LIKE..` and converts it to `colname NOT LIKE..` which is a special operator that inverts the selection process i.e. `NOT(colname LIKE..)

helgeerbe, 2021-03-31:
I'll give it a try

jgodfrey, 2021-07-05:
I think this can be closed, right?

sapnho, 2021-07-05:
It does now show more characters but oftentimes with non-Latin languages, the characters are still not shown. In Home Assistant, all the characters are shown but not in PictureFrame. But I guess, it's too hard to change that so we might as well close this issue.

paddywwoof, 2021-07-05:
I believe that the issue has been moved outside ```pi3d``` and ```picframe```. i.e. if the font that is referenced from ```configuration.yaml``` has the character in it then it should show on the screen.

@sapnho is this not the case?

sapnho, 2021-07-06:
I probably misunderstand how this is supposed to work. If I have an image with Japanese GPS data, the Japanese characters are shown correctly in Home Assistant but not on the info display in Pi3D. So my hope was since I don't have to specify a particular font in Home Assistant, it might work the same in Pi3D. Is there any way to make this happen? Thanks!

helgeerbe, 2021-07-06:
@sapnho You can play around with google fonts and try to find one that meets the requirements. [https://fonts.google.com/specimen/Open+Sans?subset=japanese&preview.text=彼らの機器や装置はすべて生命体だ%E3%80%82dfasdfas&preview.text_type=custom#standard-styles](https://fonts.google.com/specimen/Open+Sans?subset=japanese&preview.text=彼らの機器や装置はすべて生命体だ%E3%80%82dfasdfas&preview.text_type=custom#standard-styles)
Not all fonts support all characters. But I don't now how Home Assistant is dealing with that.

sapnho, 2021-07-06:
Indeed that is a possibility but unfortunately incomplete. I wonder how this is handled in non-Latin character languages in general.

helgeerbe, 2021-07-06:
Try to install google noto fonts.

> Beautiful and free fonts for all languages

> When text is rendered by a computer, sometimes characters are displayed as “tofu”. They are little boxes to indicate your device doesn’t have a font to display the text.

> Google has been developing a font family called Noto, which aims to support all languages with a harmonious look and feel. Noto is Google’s answer to tofu. The name noto is to convey the idea that Google’s goal is to see “no more tofu”. Noto has multiple styles and weights, and is freely available to all. The comprehensive set of fonts and tools used in our development is available in our GitHub repositories.

sapnho, 2021-07-06:
That sounds intruiging!

helgeerbe, 2021-07-06:
[https://www.google.com/get/noto/#sans-lgc](https://www.google.com/get/noto/#sans-lgc)

paddywwoof, 2021-07-06:
I switched to noto in pi3d_demos as soon as they were announced (can't believe it's **five** years ago!). And these are the font files brought across into picframe... but I haven't updated the files since I first added them so maybe there were some missing characters that have now been filled in. I will check out the latest download.

paddywwoof, 2021-07-06:
Hmm, just downloaded the zip and NotoSans-Regular.ttf was last updated in 2017. However it now looks to be 455.2kB and the version in picframe is 399kB so there might be some missing character there. I will update and push to dev branch.

helgeerbe, 2021-07-06:
582 languages, but no Japanese in NotoSans.
[image]

sapnho, 2021-07-06:
@paddywwoof Outstanding! Why they left out the few kana characters in Japanese is a miracle to me as the rest should look like the Chinese Kanji ones, shouldn't they?

paddywwoof, 2021-07-06:
Maybe the chinese characters in NotoSans-Regular is very incomplete. And you should really use https://www.google.com/get/noto/help/cjk/

Not sure what the most efficient way to get all characters without having to change fonts mid-way through the text. Probably basic ASCII will included at the beginning of **all** fonts. Obviously it would be possible to test for the existence of a character and look through different fonts but that's beyond the scope of the picframe specification!

jgodfrey, 2021-07-11:
Related... I have the following string in a `location` table record:

`Wuzhen West Scenic Zone, 桃源镇, Wujiang District, China`

When shown on the frame, those 3 Chinese chars show up as hollow boxes - even with the latest Noto font updates...

paddywwoof, 2021-07-11:
When I copy paste your text into libreoffice and set the font to NotoSans I see that those three characters actually get `Noto Sans Mono CJ` applied to them. The WP seems to have an algorithm in to check what installed fonts have the appropriate unicode glyphs in. I don't think we will be trying to do anything like that in picframe, however all the standard European as well as katagana, kiragan and anything else I could think of seem to be included in that font...

So maybe we should bundle `Noto Sans Mono CJ` with picframe and set the default to that

jgodfrey, 2021-07-11:
It looks like `Noto Sans Mono CJK` comes in 4 default language variations (times 2 serif variations). The site states the following:

_Each font sets one language as the default. Note that each language-specific font does support all four languages and includes the complete set of glyphs. However, you need an application program that can invoke an OpenType locl GSUB feature (e.g. Adobe InDesign) to access language-specific variants other than the default language._

I assume nothing in the picframe dependency chain supports the mentioned OpenType locl GSUB feature (?). Which one would you install by default?

jgodfrey, 2021-07-11:
I just installed the `NotoSansCJKsc-Regular` font and have verified that it does indeed display Chinese characters in the string mentioned above correctly. Nice.

sapnho, 2021-07-11:
> I just installed the `NotoSansCJKsc-Regular` font (the simplified Chinese variation) and have verified that it does indeed display Chinese characters in the string mentioned above correctly. Nice.

That is great news!

closed 2021-11-25 by helgeerbe

sapnho, 2024-11-30:
I know that this thread has rested in peace for three years but just like with unsolved murder cases, there are new insights every now and then that merit a revisit...

I just had a photo from Myanmar on my frame and the geodata was shown correctly in Home Assistant but only boxes appeared on PictureFrame.

If it is doable on Home Assistant, can't it be done on PicFrame? Hope never dies..

[image]
[image]