helgeerbe/picframe #87: subdirectory: ""                        # default="", subdir of pic_dir - can be changed by MQTT"

sapnho, 2021-03-22:
Does 
`subdirectory: ""  # default="", subdir of pic_dir - can be changed by MQTT" `
in configuration.yaml still have a use case?

jeffalperin, 2021-03-24:
I like being able to specify a specific subdirectory, but I'd also like to be able to specify a new **root** directory via MQTT.

helgeerbe, 2021-03-25:
subdirectory in configuration.yam is a initial starting point, when PictureFrame starts up. So I would say, that this is usefull. 

Changing root directory might not be trivial. Because you have to deal with the db. Does changing root mean to clear the database, and the cache has to be rebuild. Or do you want to have a 'temporary' root change? E. g. using a usb stick. And then return to the origin root, when unconnecting the stick.

jgodfrey, 2021-03-25:
FWIW - Today, a "root-level" directory swap (assuming we provided the ability to do it) would clear the cache and start the rebuild process again with info from the new folder...

helgeerbe, 2021-03-25:
Just thinking loud. Changing root, should rebuild the cache. This is for me  a not regular task. But using sticks or everything under `/media/usbdrive` will create a separate temporary database and cache. When you pull the stick, the original cache and root will continue to work.

jgodfrey, 2021-03-25:
Agreed, it needs to rebuild the cache on root directory change. However, maybe we don't need to be so aggressive with clearing out the previous data. For instance, the stale records could just be *marked* as having been deleted (or "not current"), rather than physically removing them. That'd make returning back to that initial folder less painful as the cache would still exist, and would just have to be marked as "current" (or similar)...

Just thinking out loud - I haven't given any real thought to any of this...

sapnho, 2021-03-25:
@jeffalperin What is the use case behind changing the root folder?

jeffalperin, 2021-03-25:
Fair question!  It would allow a less-organized user (me!) to be a little more casual in how they store their images on the rPi.  At the moment while I'm playing with setting up the picframe, I have images in multiple "root" directories rather than a single, nested directory structure.  Of course, I could (and perhaps should) spend some time to rationalize this sloppy structure, but it would be nice to be able to access any image folder on the rPi via MQTT even if the image directory structure is messy.

A second use case is if I'm using SAMBA to access image folders on another computer in my network (i.e. my Win 10 desktop where I do my image editing/processing).  I sometimes lose the SAMBA connection to the remote image folder, so if that happened, it would be nice to be able to specify a new folder on the rPi via MQTT instead of having to stop picframe and restart it with the "replacement" directory.

helgeerbe, 2021-03-27:
Haven't tested it. But couldn't you just mount your Samba share below 'Pictures'? Something like this `mount -t smbfs -o username=<your_username>,password=<your_password> //server/share /home/pi/Pictures/images`?

So if keep old folders in the cache, you would have instantly access to the images, if the folder appears again.

jeffalperin, 2021-03-27:
@helgeerbe Yes, that sounds like a better organizational scheme.

helgeerbe, 2021-03-28:
Have a look here, for an [automount example](https://github.com/helgeerbe/picframe/issues/76#issuecomment-808919644). Of course you could also mount your smb device.

helgeerbe, 2021-04-08:
Guess we are done here. 
I tested auto mount on my UHD TV. It's fun to have a quick slide show on 65'' UHD from a stick. Only backdraw is that the aspect ratio is only 16:9.

closed 2021-04-08 by helgeerbe