helgeerbe/picframe #88: for integration with Home Assistant, ignore "hidden" subdirectories of root image folder

jeffalperin, 2021-03-23:
I'm using SyncThing to synchronize image folders to my picture frame.  SyncThing creates an empty subfolder named `.stfolder` in the root image folder used by picframe. This subfolder is shown in Home Assistant's drop-down list of possible image directories.  Is it possible to exclude this (and any other hidden directories)?

sapnho, 2021-03-23:
Hi Jeff, that's interesting. I am also using SyncThing but no `.stfolder` is being shown in my Home Assistant dropdown list, only the folders under `/Pictures`. Anything different from the default settings that you may have done?

jeffalperin, 2021-03-23:
@sapnho, I have a different root photos directory `/home/pi/4kpix`, but I wouldn't think that cause this behavior.  I manually deleted the `.stfolder` from the dropdown list (from the Helpers section of Home Assistant) and then manually re-executed the `picframe_directory_get` automation and .stfolder was once again added to the drop-down list.

If you do an `ls -all` inside `/Pictures`, do you have an `.stfolder`?

sapnho, 2021-03-23:
> If you do an `ls -all` inside `/Pictures`, do you have an `.stfolder`?

No, I don't.

jeffalperin, 2021-03-23:
Hmmm.  From: SyncThing FAQ at https://docs.syncthing.net/users/faq.html

**How do I serve a folder from a read only filesystem?**

Syncthing requires a “folder marker” to indicate that the folder is present and healthy. By default this is a directory called .stfolder that is created by Syncthing when the folder is added. If this folder can’t be created (you are serving files from a CD or something) you can instead set the advanced config Marker Name to the name of some file or folder that you know will always exist in the folder.
**I really hate the .stfolder directory, can I remove it?**

See the previous question.

sapnho, 2021-03-23:
> I really hate the .stfolder directory, can I remove it?

I think you can't.

However: My structure is that I have a folder below /Pictures which is called Photos. Inside Photos is the .stfolder.

I guess that's why it doesn't show in Home Assistant.

sapnho, 2021-03-23:
And to add, my synchronization is on /Pictures/Photos.

helgeerbe, 2021-03-23:
Hi, actual all folders are used. To be honest, haven't thought about hidden folders. But it should be easy to exclude them. Don't know what best how to deal with subfolders.
Actual only first level subfolders are send do homeassistant (But all images below this subfolder structure are displayed). But there might be user who have a deep nested folder structure. Although I don't know how to handle that in a selection list.

E. G.
- Pictures
  - subdir1
  - subdir 2
    - subsubdir1
    - subsubdir2

At the end this could result in a selection like this:
- Pictures
- Pictures/subdir1
- Pictures/subdir 2
- Pictures/subdir 2/subsubdir1
- Pictures/subdir 2/subsubdir2

Could become long and nasty for huge structures.

helgeerbe, 2021-03-23:
@sapnho Only first level subfolders are displayed in home assisstant.

jgodfrey, 2021-03-23:
It looks like the folder collection you're working with is coming from `model.get_directory_list()`. There, it's using the standard `os.walk()` method to build up the list. While there's no option to skip hidden files/folders in `os.walk()`, it be trivial to not include them at that stage.

@paddywwoof / @helgeerbe - is there any legitimate reason we'd ever want to include hidden folders here (I assume not), or can we simply eliminate them in `get_directory_list()`?

helgeerbe, 2021-03-23:
@jgodfrey no, we should delete them

paddywwoof, 2021-03-23:
We already filter any hidden image files or ones with `.AppleDouble` in the path so maybe we can generalise it to any file or folder starting with `.`

helgeerbe, 2021-03-27:
I removed hidden folders from the directory list [5adca41 ](https://github.com/helgeerbe/picframe/commit/09ec5986b93dcb7f79a3139bab9677c098f38715) as a workaround. In the moment I don't have a good idea how to display a complex  folder structure in home assistant. So at the moment I would stay with the first level below root.

closed 2021-04-02 by helgeerbe