helgeerbe/picframe #89: picframe showing errors in the console with IPTC descriptions (captions) longer than 199 characters

jeffalperin, 2021-03-23:
When running picframe, the console shows errors like this:  
`ERROR:pi3d.util.TextBlock:'Pinkas Synagogue, Prague.  The walls of the synagogue are covered with names of about 78 000 Bohemian and Moravian Jewish victims of Shoah. The names are arranged by communities where the victims came from and complemented with their birth and death date. • Europe 2019_20190702_7348.jpg • Jul 02, 2019 • Pinkas Synagogue, Jewish quarter, Prague, Prague, Czechia • Europe_2019 ' needs 377 characters but only space for 199`
for images with long IPTC descriptions/captions.
The full caption seems to be displayed onscreen and in Home Assistant, but MQTT commands (e.g. advance to next image) are delayed in execution for several seconds.

Is it possible to increase the size available for captions?

paddywwoof, 2021-03-23:
Yes this can be set to any number for the PointText object here https://github.com/helgeerbe/picframe/blob/6236fa48ad95e41bd4aadaff1053c97fc2aaf4e6/picframe/viewer_display.py#L390 and each TextBlock here https://github.com/helgeerbe/picframe/blob/6236fa48ad95e41bd4aadaff1053c97fc2aaf4e6/picframe/viewer_display.py#L401

I'm not sure what you would set it to. Is 377 top end? In which case 1000 and 499 in each TextBlock might do. However the advantage of using the PointText system is that it allows great flexibility for changing character and moving individual letters about - none of which is really used in picframe. On that basis, dicussed https://github.com/helgeerbe/picframe/issues/86#issuecomment-804369504 it might give more resilience and cope with other alphabets if we modified FixedString and used that.

paddywwoof, 2021-04-17:
This problem fixed by switching to pi3d.FixedString

closed 2021-04-17 by paddywwoof