helgeerbe/picframe #90: Matting color different depending on mat type or random?

sapnho, 2021-03-27:
@jgodfrey This is not a bug report but just a question out of curiosity: Is the matting color depending on the mat type or is it sometimes randomized?

I am asking because I got two very different results for the same image and the only difference was the mat type:

[image]
[image]

paddywwoof, 2021-03-27:
The colour is determined by taking three randomly selected pixels initially then grouping all the pixels by how close they are to each of those three. The the three RGB values are taken as the average of each group and the process repeated. After a few cycles the colors partition themselves into groups of similar colours which are then sorted according to how non-grey they are. Obviously it does depend on what the initial pixels are but quite often the mean of each cluster 'migrates' towards the same value.

It might be more consistent if we started with more than three points - I don't think it would have much impact on processing time. Also we don' really take account of colour perception in judging the non-greyness so [230, 255, 230] would score 25 higher than [50, 74, 50] scoring 24
[image]

jgodfrey, 2021-03-27:
Yep, Paddy's description is exactly correct. It's that initial "random" selection of 3 pixels that causes the solution to sometimes differ for the same image. Really, the algorithm used here (kmeans clustering) is intended to be iterative. That is, it's meant to be run over and over and over again until the solution stabilizes (pixels are no longer shifted between the color buckets). While that would provide a much more repeatable solution, it's also much more expensive. Because of that, many implementations (including ours) just run some predetermined number of iterations and call the solution "good enough". If it were allowed to run longer, the final color selection should be more repeatable - so really, it's a _solution quality_ versus _time_ tradeoff. 

@paddywwoof - perhaps with the `numpy` implementation, we could afford a few more iterations? Though, I'm not sure it's necessary.

Another option to stabilize the color for a given image would be to store it in the cache and not calculate it again. Originally, I intended to do exactly that because the original implementation was _really_ expensive (and slow). But, with the `numpy` implementation, it was fast enough to just run _on-the-fly_, so no value storage necessary.

sapnho, 2021-03-27:
Thanks, guys, for the great explanation. It's great as it is already!

jgodfrey, 2021-04-02:
Seems like we're all in agreement here. Closing this one.

closed 2021-04-02 by jgodfrey