helgeerbe/picframe #91: Can Date values be "none"?

sapnho, 2021-03-27:
When I want to see all my images, I need to specify the dates in HA that will make sure that everything is included. I remember from a previous version that the date values could just be empty?

[image]
[image]

paddywwoof, 2021-03-27:
You can either send a float value representing seconds since 1970/1/1 or a string with year_num, month_num, day_num separated by "/" or ":" or "-" or "," or "." or a zero length string.

Just looking at the code I see that zero length sting is converted to 1970/1/1 whereas the init value is 1901/12/15 the oldest value that home assistant can cope with. I've fixed that now to make it consistent.

NB sending a string "None" or "null" will probably cause a ValueError. Sending "nan" or "NaN" will probably cause odd results, if not errors. Just tried that now and had to kill the program and restart it!

helgeerbe, 2021-03-27:
In home assistant, you have to set the dates. Initial it's get the values from PictureFrame when it starts up. But I can check if it would possible to set the input text fields to maximum and minimum dates, when you make an empty input (clear text and hit enter)

sapnho, 2021-03-27:
> But I can check if it would possible to set the input text fields to maximum and minimum dates, when you make an empty input (clear text and hit enter)

That would be perfect!

helgeerbe, 2021-03-31:
Wolfgang this seems to work for me:

`picframe.date_from_set`
````YAML
alias: picframe.date_from_set
description: ''
trigger:
  - platform: state
    entity_id: input_text.picframe_date_from
condition:
  - condition: template
    value_template: >-
      {{
      as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y"))
      | int != states("sensor.picframe_date_from")|int }}
action:
  - service: mqtt.publish
    data:
      topic: picframe/date_from
      payload: |-
        {% if states("input_text.picframe_date_from") != "" -%}   
          {{ as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y")) |     int }} 
        {%- else -%}     
          {{ as_timestamp(strptime("15.12.1901","%d.%m.%Y")) |     int }} 
        {%- endif %}
mode: single
````

`picframe.date_from_set`
````YAML
alias: picframe.date_to_set
description: ''
trigger:
  - platform: state
    entity_id: input_text.picframe_date_to
condition:
  - condition: template
    value_template: >-
      {{
      as_timestamp(strptime(states("input_text.picframe_date_to"),"%d.%m.%Y")) |
      int != states("sensor.picframe_date_to")|int }}
action:
  - service: mqtt.publish
    data:
      topic: picframe/date_to
      payload: |-
        {% if states("input_text.picframe_date_to") != "" -%}   
          {{ as_timestamp(strptime(states("input_text.picframe_date_to"),"%d.%m.%Y")) |     int }} 
        {%- else -%}     
          {{ as_timestamp(strptime("01.01.2038","%d.%m.%Y")) |     int }} 
        {%- endif %}
mode: single
````

Empty input will published as minimum/maximum date to PictureFrame. The PictureFrame sends the set date back to home assistant. So maximum/minimum date will appear in the text input field.

sapnho, 2021-03-31:
That works! I'll change it in the blog post as well.

paddywwoof, 2021-03-31:
What happens if you just send an empty string? That's what I do with my MQTT client on my phone or the HTTP (which ends up calling the same Controller method) the method interprets empty string as min or max

sapnho, 2021-03-31:
> That works! I'll change it in the blog post as well.

Ah, after some playing around, it doesn't work. The empty value is accepted but when I add back dates in, it doesn't accept it.

helgeerbe, 2021-03-31:
The given example is using German date format. Not sure, if you are using in the input_text American format? Regex?

helgeerbe, 2021-03-31:
I will test, to send empty "" string. In that way home assistant doesn't have to know max/min date.

helgeerbe, 2021-04-01:
Empty payload seems to work. @sapnho in the solution above there was a typo. Right behind `Payload` I uses `|` instead of `>`

`picframe.date_from_set`
````YAML
alias: picframe.date_from_set
description: ''
trigger:
  - platform: state
    entity_id: input_text.picframe_date_from
condition:
  - condition: template
    value_template: >-
      {{
      as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y"))
      | int != states("sensor.picframe_date_from")|int }}
action:
  - service: mqtt.publish
    data:
      topic: picframe/date_from
      payload: |-
        {% if states("input_text.picframe_date_from") != "" -%}   
          {{ as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y")) |     int }}  
        {%- endif %}
mode: single
````

`picframe.date_from_set`
````YAML
alias: picframe.date_to_set
description: ''
trigger:
  - platform: state
    entity_id: input_text.picframe_date_to
condition:
  - condition: template
    value_template: >-
      {{
      as_timestamp(strptime(states("input_text.picframe_date_to"),"%d.%m.%Y")) |
      int != states("sensor.picframe_date_to")|int }}
action:
  - service: mqtt.publish
    data:
      topic: picframe/date_to
      payload: |-
        {% if states("input_text.picframe_date_to") != "" -%}   
          {{ as_timestamp(strptime(states("input_text.picframe_date_to"),"%d.%m.%Y")) |     int }} 
        {%- endif %}
mode: single
````

Guess, this is the better solution, because PictureFrame should know its min/max date. Does anybody knows, if the linux time period is extended with a 64 bit linux?

paddywwoof, 2021-04-01:
Python 64 can cope with dates that go back earlier than py32 but I don't know what the range of the algorithm (that converts to Y/M/D) is (I don't suppose anyone worried about the number of days in February back in the Jurassic)

I don't know what the sqlite constraints are. I will check that now.

helgeerbe, 2021-04-01:
Paddy, I was looking forward in time. 2038 is not so far :smile:

helgeerbe, 2021-04-01:
@sapnho When you delete the date string in home assistant. Date is set to max/min date. If you delete the string again, the text input will be empty. Reason is, that the date state sensor for PictureFrame isn't updated (still min/max). So the automation is not triggered.

helgeerbe, 2021-04-01:
This works for me now

`picframe.date_from_set`
````YAML
alias: picframe.date_from_set
description: ''
trigger:
  - platform: state
    entity_id: input_text.picframe_date_from
condition:
  - condition: template
    value_template: >-
      {{
      as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y"))
      | int != states("sensor.picframe_date_from")|int }}
action:
  - service: mqtt.publish
    data:
      topic: picframe/date_from
      payload: |-
        {% if states("input_text.picframe_date_from") != "" -%}   
          {{ as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y")) |     int }} 
        {%- endif %}
  - condition: template
    value_template: '{{ states("input_text.picframe_date_from") == "" }}'
  - service: input_text.set_value
    data:
      value: >-
        {{ states('sensor.picframe_date_from') | int |
        timestamp_custom('%d.%m.%Y') }}
    entity_id: input_text.picframe_date_from
mode: single
````

`picframe.date_to_set`
````YAML
alias: picframe.date_to_set
description: ''
trigger:
  - platform: state
    entity_id: input_text.picframe_date_to
condition:
  - condition: template
    value_template: >-
      {{
      as_timestamp(strptime(states("input_text.picframe_date_to"),"%d.%m.%Y")) |
      int != states("sensor.picframe_date_to")|int }}
action:
  - service: mqtt.publish
    data:
      topic: picframe/date_to
      payload: |-
        {% if states("input_text.picframe_date_to") != "" -%}   
          {{ as_timestamp(strptime(states("input_text.picframe_date_to"),"%d.%m.%Y")) |     int }} 
        {%- endif %}
  - condition: template
    value_template: '{{  states("input_text.picframe_date_to") == "" }}'
  - service: input_text.set_value
    data:
      value: >-
        {{ states('sensor.picframe_date_to') | int |
        timestamp_custom('%d.%m.%Y') }}
    entity_id: input_text.picframe_date_to
mode: single
````

paddywwoof, 2021-04-01:
Reading a bit more on this it seems that linux since 5.6 (Jan 2020) can cope with datetime using int64 on 32 bit systems i.e. the current Raspbery Pi OS should do. Presumably it then depends on how the python and sqlite source was cross compiled on the debian repository. When I look at the python3 on the Pi3 it shows me Python 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0] on linux so I think it could be compiled to use int64 but it hasn't been. Presumably someone will fix that over the next few years.

sapnho, 2021-04-02:
> This works for me now

Great. I changed it in the article.

helgeerbe, 2021-04-08:
Updated the yaml files in the wiki.

closed 2021-04-08 by helgeerbe