helgeerbe/picframe #92: Brightness Slider does not work reliably in Home Assistant

sapnho, 2021-03-28:
I noticed that the Brightness Slider in Home Assistant is often unresponsive, especially if you change the brightness quickly a few times.

Did anyone notice the same thing? MQTT hick-up?

helgeerbe, 2021-03-28:
I just tested it, and it seems to work fine. But I could think, that there might be a timing issue on slow networks. When you moved the slider the value is send to PictureFrame. Changed values are send back to home assistant.
In that way home assistant will be in syc with the web-interface. 
But if you change the values too fast (you have to release the slider), home assistant might receive an  older value and thinks it's a remotely set value and will update the value to an old value.

sapnho, 2021-03-28:
> But if you change the values too fast (you have to release the slider), home assistant might receive an older value and thinks it's a remotely set value and will update the value to an old value.

Yes, I think that's it. However, my network is super fast, so maybe there is a limit of MQTT commands per second.

helgeerbe, 2021-04-08:
By the way. You can change the slider to input field. That might 'slow' you down. Sorry, but for now, I don't have a better solution.

closed 2021-04-15 by helgeerbe