helgeerbe/picframe #95: Sorting when shuffle is False

sapnho, 2021-03-31:
I have a reader whose images are 001.jpg, 002.jpg etc. but despite Shuffle: False, he still doesn't get the right order of the images.

Is there another logic to non-shuffled playback than sorting on the file names?

helgeerbe, 2021-03-31:
Well, shuffle is my default and I never changed it. But sometimes in the past I used `recent_n` after a restart, so see  images of my last trip over an evening. So I'm absolutely fine, in the way it is. But I'm also fine with sorting by filename. Because, I can filter now by start and end date and mostly my images have an image counter in its filename.

sapnho, 2021-03-31:
I only use shuffle mode, too, so I never looked at non-shuffle until today when I received the email.

I advocate for using sort by filename. Sorting by Exif date probably doesn't have a use case. People who use it commercially, or even if you want to have some special effects for parties, etc., will need the sort by file name.

jgodfrey, 2021-03-31:
@paddywwoof - what are you considering doing here? Ideally, it might be nice for a User to specify the `field` on which to sort and sort order (`ASC` or `DESC`) of the _sorted_ play list. However, as you eluded to above, we don't have the bare `filename` in the `all_data` table as it's been merged with the full path at that point. I guess we could split the filename off of the path for sorting purposes, though someone may well want the path as part of the sort criteria...

sapnho, 2021-03-31:
> what are you considering doing here? Ideally, it might be nice for a User to specify the `field` on which to sort and sort order (`ASC` or `DESC`) of the _sorted_ play list.

That would be ideal, of course!

paddywwoof, 2021-04-01:
Yes, it makes the config even more complicated but the alternative is to simply change from `exif_datetime` to `fname` and someone might want to sort on something different (i.e. `latitude ASC, longitude ASC`) The easiest way would be to put a hard code of likely column names in but that's seems like an extra maintenance thing to remember. I think it's better to add a method to ImageCache
```python
    def get_column_names(self):
        sql = "PRAGMA table_info(all_data)"
        rows = self.__db.execute(sql).fetchall()
        return [row['name'] for row in rows]
```
then something like this in Model.__get_files()
```python
        if self.shuffle:
            sort_list.append("RANDOM()")
        else:
            col_names = self.__image_cache.get_colum_names() # or do this once in __init__
            for col in self.sort_cols.split(","):
                colsplit = col.split()
                if colsplit[0] in col_names and (len(colsplit) == 1 or colsplit[1] in ('ASC', 'DESC')):
                    sort_list.append(col)
        sort_clause = ",".join(sort_list)
```
I'll try it out and see what the problems are.

paddywwoof, 2021-04-01:
Well that seems to work nicely. Something odd happens with portrait pairs though. Because the pairs are constructed by 'scanning ahead' for the next portrait image in the sequence they become increasingly forward shifted. I doubt there is any simple solution to this.

The way it's set up here means that if the user specifies
```yaml
  recent_n: 0
...
  shuffle: False
  sort_cols: '' # i.e. they have deleted the default value
...
```
Then it will generate SQL `SELECT ... ORDER BY` giving an incomplete input error.

PS I have switched the default order to `fname` from `exif_datetime`. And, to preempt anyone asking if it would be possible to generalize the where clause in a similar way: I think it would be significantly more difficult

jgodfrey, 2021-04-01:
| `PRAGMA table_info(all_data)`

Yeah, this is definitely the right approach for getting the field names. As you said, we don't want to maintain that independently from the db schema.

The sort clause builder you posted looks reasonable to me, though I might split it into its own method for clarity (if it's not already). 

| Then it will generate SQL SELECT ... ORDER BY giving an incomplete input error.

I guess we can just define a default value for `sort_cols` if its left undefined. Or, maybe I'm missing the point?

paddywwoof, 2021-04-01:
Yes, easy enough to do. Actually, when watching the slides I decided that this should go on the end anyway. i.e. if sorting on lat and lon then all the images with no data to sort on get ordered by fname which is more what you might expect: it's a logical sequence as opposed to `shuffle = True`.

pushed changes to dev https://github.com/helgeerbe/picframe/commit/ac7e29683471b926ab80f559f7518a98edbf2743

helgeerbe, 2021-04-02:
Guess I should release a new version, or are there some more improvements in the pipe?

paddywwoof, 2021-04-02:
The only thing I can remember off-hand is the system to stop ImageCache clearing all the records from the database as soon as the file path no longer exists... And I suppose some mechanism to make the that clean-up run, either once every x days defined in config (store the 'last-clean-up' info on the db?) or (and) via a Controller method that can be called via mqtt or http.

I think you got the auto-mounting with systemctl working so all that's needed for that are some docs.

PS the idea of setting jobs to be done at x frequency made me think that ought to be done by systemctl too (rather than setting up an internal timer), including changing brightness and turning the display on and off. Is this best done with mqtt command line (assuming homeassistant isn't being used - presumably that can do all this automatically)

sapnho, 2021-04-02:
So, we'll have a Good Friday release today? :-)

helgeerbe, 2021-04-02:
I'll do my best. Hope I have time in the evening. 
We have new configuration items. So I will also update the wiki.

sapnho, 2021-04-02:
That was quick! :-)
One question regarding the update with "`(sudo) pip3 install picframe --upgrade`": Must the initialisation also be run because the configuration.yaml may have changed?

sapnho, 2021-04-02:
I think I can answer that: Rename the old configuration.yaml to "_old", run `picframe -i .` and manually copy the old settings from configuration.yaml.

helgeerbe, 2021-04-02:
Done. Version `2021.04.02` is released. See [changelog](https://github.com/helgeerbe/picframe/wiki/Changelog).

Yes backup `configuration.yaml` and run `picframe -i`

sapnho, 2021-04-02:
And updated here: https://www.thedigitalpictureframe.com/pi3d-parameters-directory-config/

jgodfrey, 2021-04-02:
> The only thing I can remember off-hand is the system to stop ImageCache clearing all the records from the database as soon as the file path no longer exists... And I suppose some mechanism to make [...]

Yeah, I think that makes sense. I recently added a `db_info` table, which currently stores only a single `schema_version` record. We could add some sort of clean-up timestamp there. Though, we really should start to make use of the recently added `update_schema()` method to make these kinds of changes. That way, existing db's can be auto-migrated without requiring the cache to be rebuilt. 

I'll try to look at some of that over the weekend unless someone beats me to it...

paddywwoof, 2021-04-02:
Wolfgang, that's a *really* quick update to documentation! I notice you still mention `auto_resize` which is now no longer used (though I've commented it out in configuration.yaml rather than deleting it) Basically pi3d checks the maxiumum texture size when it starts and that gets used whatever platform you run on. The maximum size seems to always be a power of 2 rather than the maxium display size so I suspect it's tied up with the[ mipmapping functionality](https://en.wikipedia.org/wiki/Mipmap) which probably caused the seg faults before when widths were used that produced *no* mipmaps.

Also `recent_n` now represents the number of days ago that the file changed rather than the number of images.

Jeff, it would be good to get the update_schema system - I've no idea about db schema so will leave it to you! Probably not a rush for this now that there is a new update out - though we probably should stop the `ImageCache.__purge_missing_files_and_folders()` running automatically - maybe have a new class variable `self.__purge_files = False` and a public method `purge_files()` that sets it to True and can be called from Controller, and have an MQTT and HTTP way of calling that. I'm happy to do these latter if you want to look at the db side.

Paddy

jgodfrey, 2021-04-02:
> though we probably should stop the `ImageCache.__purge_missing_files_and_folders()` running automatically

@paddywwoof - I assume we'll want to incorporate some notion of _known to be missing_ folders at the same time. That is, mark the missing folders and use that when building the _playlist_ from within `query_chache()`. Technically, we shouldn't _have_ to do that at the same time (since the slideshow should simply skip missing files), but it could potentially lead to _lots_ of cache misses in the interim.

So, seems like it'd be better to do both at the same time, though probably not absolutely required. Do you agree?

paddywwoof, 2021-04-02:
Yes, sensible. Hadn't really thought that through. I've patched in the few lines to stop the purging automatically and pushed to dev https://github.com/helgeerbe/picframe/commit/445bcb1eebba1fcd595cf501d928abfa61240c4f I will look at record flagging next - though that will presumably need a db change so we should tie that in with the update_schema system.

sapnho, 2021-04-02:
Thanks, @paddywwoof, I made those changes in the article.

paddywwoof, 2021-04-02:
Hmm I wondering if this can be done by deleting the entry from the folder table. i.e. the first part of the `__purge_missing_files_and_folders()` method, but leaving the file and meta entries until the __purge_files has been set. The select in `query_cache()` would then just need to not include any records without an entry in folder - I think that would happen anyway by normal `inner join` logic. Doing it this way would mean no need for the extra column but it would mean that it would only work when whole folders had disappeared - which is what the change is aimed at so probably OK.

jgodfrey, 2021-04-02:
> Hmm I wondering if this can be done by deleting the entry from the folder table. i.e. the first part of the __purge_missing_files_and_folders() method, but leaving the file and meta entries until the __purge_files has been set.

Potentially "yes" (I think) though it would cause some havoc in the existing `trigger` mechanism that automatically clears file records for missing folders. Specifically, this: https://github.com/helgeerbe/picframe/blob/de3710a1ce38ace8f15666807912ff768af1607a/picframe/image_cache.py#L235

For that reason, I'd still probably opt to add another field to the `folder` table to indicate its current availability...

paddywwoof, 2021-04-02:
Yes, in actual fact the second part of purge_missing_files etc only deletes files that *aren't* in folders that still *do* exist in the `folder` table as the `DELETE FROM .. all_data` will miss the files. That's probably why you use the trigger.

Another field it is then.

paddywwoof, 2021-04-02:
Added a new branch 017e2615c30486feaaed061bb1d3588fbc7fd97f with a modified db flagging missing folders but not deleting them until purge flag set. Seems to run OK but bit of a palaver adding the new column - the only way seems to be to create a new table with the additional column then INSERT from old one then DROP old one and rename new one. I kept getting errors trying to rename the table because at that stage 'it' decided to check the all_data view which was now referring to a table that had been deleted. The view needed to be deleted and re-created anyway but obviously the order of doing it is critical.

Presumably the schema system will need to have a series of scripts to convert from each previous schema to the latest one. Is the idea to change the required_db_schema_version in model? A default value is also set in ImageCache.__init__ argument. In some ways, as the db structure is 'hard coded' into the ImageCache it might be more logical to set it in the __init__ I'm not sure why it would need to be set to something different externally.

jgodfrey, 2021-04-02:
> the only way seems to be to create a new table with the additional column then INSERT from old one then DROP old one and rename new one.

It should be possible to do something like `ALTER TABLE folder ADD COLUMN missing INTEGER DEFAULT 0 NOT NULL` with the existing table.

> Presumably the schema system will need to have a series of scripts to convert from each previous schema to the latest one.

Correct. The idea is to always create the same initial schema in `__create_open_db()` - even though it's now outdated. Then, simply incrementally migrate from the current schema through each additional schema change via independent steps in `update_schema()`. 

So, rather than adding that new `missing` column in `__create_open_db()`, it'd be added by the migration step that goes from schema_version 1 to schema_version 2. 

> Is the idea to change the required_db_schema_version in model?

That's where it is now, though it could be specified elsewhere. Really, the important part is that any given version of the application should specify what schema version it expects to work with. And, if the db it accesses is older than that expected version, it should be capable of migrating the db to the expected version _on the fly_.

I've used a similar db migration mechanism in several (SQLite-based) commercial applications and it's worked really well...

jgodfrey, 2021-04-02:
Also, I wonder if the `all_data` view should simply not contain files where the containing folder is currently missing - rather than `query_cache()` culling those files later in the game? Is there any advantage to including the missing files in the `all_tables` view? I can't think of any...

paddywwoof, 2021-04-02:
Ah doing an incremental system like that makes the external version setting much more logical. The errors might be a limitation of sqlitebrowser - it appears to allow some tweaking of database structure and lets you run SQL but when I tried to modify the table (and now when I retry it after dropping the view) I get
```text
Result: near "NUL": syntax error
At line 2:
ALTER TABLE folder ADD COLUMN missing INTEGER DEFAULT 0 NOT NUL
```
Oddly it will let me run it if I remove the 'NOT NUL' constraint!

I think you're right that all_data doesn't need to list the files at all - before you pointed out the auto delete trigger I thought the purge process selecting the files from all_data was the mechanism for deleting them. Presumably there isn't a way of altering a VIEW -  just have to DROP and create a new one I should think.

jgodfrey, 2021-04-02:
> Presumably there isn't a way of altering a VIEW - just have to DROP and create a new one I should think.

Yeah, I think that's correct (no `ALTER VIEW`). So, the view should be dropped and recreated as part of a migration step...

jgodfrey, 2021-04-02:
> Result: near "NUL": syntax error
At line 2:
ALTER TABLE folder ADD COLUMN missing INTEGER DEFAULT 0 NOT NUL

Is that due to using `NUL` instead of `NULL`? There are some unusual limitations in SQLite's ALTER command, but for this case, the limitation should just be...

_If the new column has a NOT NULL constraint, you must specify a default value for the column other than a NULL value._

paddywwoof, 2021-04-02:
Yes that's exactly it!

helgeerbe, 2021-04-15:
Can we close this issue?

paddywwoof, 2021-04-17:
Helge, I will try to add something to the missing_folder branch to do the schema migration then test it a bit. Maybe we should also think about how and when Controller.purge_files() gets called

jgodfrey, 2021-04-17:
> Helge, I will try to add something to the missing_folder branch to do the schema migration then test it a bit.

Apologies for not finding the time to work on this recently. I seem to have a few too many irons in the fire right now. Some work in this area is still on my list if you don't get to it @paddywwoof...

paddywwoof, 2021-04-17:
Jeff, no problem - I've run it for a while on the laptop and Pi3 and it seems to behave OK. Sometimes it takes a wile to realize that the folder has been restored, not sure what causes the delay.

Also I've put a comment in https://github.com/helgeerbe/picframe/compare/missing_folder#diff-c2f08010d2b50292d75ccba60b4e63df2814ee75912b971e8f306b4f64cdd74cR26 questioning whether we should allow Model to ask for a specific schema number as the methods of ImageCache (i.e. __get_modified_folders(), __insert_file() and __purge_missing_files_and_folders()) have hard coded references to the column added in the latest schema. Maybe the value should be hard coded in ImageCache.__init__()

jgodfrey, 2021-04-17:
> Also I've put a comment in https://github.com/helgeerbe/picframe/compare/missing_folder#diff-c2f08010d2b50292d75ccba60b4e63df2814ee75912b971e8f306b4f64cdd74cR26 questioning whether we should allow [...]

The idea is that _the application_ should know what schema it expects. Currently, I've done that by defining the one, single, required application schema version in the `model.py` and then passing that value to the `ImageCache` object. Certainly that value could simply be stored in the `ImageCache` object itself. It's really a matter of taste I guess... 

Then `ImageCache` armed with 1) the schema it should be working with and 2) the schema of the current db file should be able to handle the (possible) migration on its own. Really, there are 3 ways it could play out...

1. The application's required schema matches the database schema, so no migration is necessary.
2. The database schema is older than the required application schema. In that case, the app should take whatever migration steps are necessary and then update the db's recorded schema to the latest.
3. The application's required schema is older than the db, which indicates the application is too old. In this case, there's nothing the app can really do, so it throws an error asking the user to update the app.

Also, looking at the schema upgrade step in the `missing_folder` branch, I don't think we always want to drop the `all_data` view as is being done now. This particular step (from schema version 1 to schema version 2) does indeed need to drop `all_data` and rebuild it. But, when we move to schema version 3 (for whatever reason), there may be no reason to rebuild `all_data`, so there'd be no reason to drop it.

For instance, if the db was at version 2 and the app now required version 3, it looks to me like the current migration code would break the db, as you'd end up with no `all_data` table. The point is the each migration step should be discreet, and do only the necessary work to get to the _next_ schema version. Each step will incrementally do the same thing, resulting in a fully migrated db at the end of the process.

This does, however, require that the db migration be allowed to fully complete prior to anything trying to access the db for purposes of running the slideshow.

jgodfrey, 2021-04-17:
Maybe some additional clarity here... The db migration system needs to handle the case where a user doesn't upgrade their picture frame software for an extended period of time. Let's say they're using a version that works with schema version 2. Now, a year later, they upgrade the software, and it now requires schema version 8. That should just work. The app should see that it needs version 8 and the db is at version 2. Via the code in `update_schema`, it should incrementally migrate the db through each successive schema version as follows:

`2 -> 3`, `3 -> 4`, `4 -> 5`, `5 -> 6`, `6 -> 7`, and `7 -> 8`.

At the end of the migration process, the version 2 db will be at version 8, and everything should just work...

paddywwoof, 2021-04-17:
Yes, I wasn't entirely sure (and still not) about the best way to do the incremental steps. I found that, when manually adding the column to folder previously, I got a slightly opaque error message that only went away after I dropped the view. I assumed that it was some safety feature that meant that *any* changes to tables referenced by a view would error unless the view was dropped so thought (as all the tables in the db are linked in all_data) we would always have to drop and rebuild it. And yes that 2 would have to be changed to 3 etc etc as the current version advances.. So yes it shouldn't be inside a conditional block.

However it might be that I was doing something else incorrectly when I was messing around before and the view dropping isn't necessary at all. I will do some experiments to check.

On the schema version being passed in by Model, my thinking is that all the db related code and definitions are in ImageCache so it would be neater to keep it self contained, without any variables being passed in (that have to be checked back like the nightmare exercise of debugging C code with its web of included header files).

paddywwoof, 2021-04-17:
Yes, after a few tests I can confirm that previously I was almost certainly doing something incorrectly, and there isn't a requirement to drop the view in order to alter tables. I will modify the code. Presumably it would be along the lines
```python
if schema_version <= 1:
    # do jobs
if schema_version <= 2:
    # next increment etc etc
```

jgodfrey, 2021-04-17:
> all the db related code and definitions are in ImageCache so it would be neater to keep it self contained

Sure, I'm good with that.

jgodfrey, 2021-04-17:
Also, is there any reason to carry the `missing` field in the `all_data` view? It looks like you're only including files for folders that are not missing (as previously discussed), so probably no reason to actually carry the field?

paddywwoof, 2021-04-17:
No. Well spotted. That will tidy it up further.

paddywwoof, 2021-04-18:
Pushed a couple more minor changes https://github.com/helgeerbe/picframe/compare/dev...missing_folder (and some interface_http that should have gone into their own branch!) I will checking a bit more then merge with dev

jgodfrey, 2021-04-18:
@paddywwoof - While you're making changes to `update_schema()`, those last few lines that update the db's schema to the current application level don't need to execute unless the db version was originally less than the required version. So, they really belong inside this `if` block:

`if schema_version < required_db_schema_version:`

They've been out of place since I originally added the `update_schema()` method...   :^)

paddywwoof, 2021-04-18:
Fixed.

helgeerbe, 2021-07-05:
I assume this is fixed.

closed 2021-07-05 by helgeerbe