helgeerbe/picframe #96: Scan for meta data changes?

sapnho, 2021-04-02:
Does our database recognize a metadata change?

I just updated two images on my test Pi that contained new metadata (title & caption). However, they were only shown once I had deleted the pictureframe.db3.

Is there a way to scan for metadata updates or is it just a matter of time?

jgodfrey, 2021-04-02:
Hmmm... It should pick them up automatically as long as your change updated the file's modified time - which I assume it should have...

sapnho, 2021-04-03:
I ran another test with images to which I added a title and caption and swapped them out. In one instance, the new title/caption was displayed in all other cases it wasn't.

Then I deleted pictureframe.db3 and started again. All titles/captions were displayed.

sapnho, 2021-04-03:
I noticed another thing: I usually treat all my images in Lightroom. However, when I export pictures from Apple Photos after having added title/caption, it is not shown in PictureFrame.

Maybe someone else with an iPhone can confirm or refute this.

jgodfrey, 2021-04-03:
> I ran another test with images to which I added a title and caption and swapped them out. In one instance, the new title/caption was displayed in all other cases it wasn't.

Yeah, that's not expected. Do you have access to a SQLite db browser tool? On Windows, I can highly recommend [SQLite Expert](http://www.sqliteexpert.com/download.html) (the personal edition is free). You can open the db on the RPi directly via a Samba connection. 

I'm curious if the updated meta data is in the db or not for this case? I assume it's not, but verification might be helpful here.

sapnho, 2021-04-03:
I ran another test with six very similar images, taken on an iPhone and treated with Lightroom. Out of the 6, 5 were recognised without deleting pictureframe.db3. One was not. Upon db deletion and restart, all 6 files were shown with title/caption.
[image]
[image]

sapnho, 2021-04-03:
And I just did another test, changing these 6 images again. The result: All 6 show up correctly. 
So for some reason, the scan is somewhat unreliable. Sometimes it works, sometimes it doesn't. Don't we all love debugging errors like this?.. :-)

helgeerbe, 2021-04-03:
> I noticed another thing: I usually treat all my images in Lightroom. However, when I export pictures from Apple Photos after having added title/caption, it is not shown in PictureFrame.
> 
> Maybe someone else with an iPhone can confirm or refute this.

@sapnho I don't have Apple Photos. But title, and caption are extracted from IPTC Tags. I had to learn, that there is no strict rule where the programs store this info. Most applications use IPTC and other fields.

Maybe you can install [exiftool](https://exiftool.org/install.html) and examine the image. It really dumps out every tag you can think off.

helgeerbe, 2021-04-03:
https://discussions.apple.com/thread/252235816 here is someone complaining, that jpgs exported from Photos for Mac don't contain the meta data. But I'm not sure, if this is your program?

sapnho, 2021-04-03:
> https://discussions.apple.com/thread/252235816 here is someone complaining, that jpgs exported from Photos for Mac don't contain the meta data. But I'm not sure, if this is your program?

Yes, that's probably the issue. Amazing that Apple allows this to happen.

paddywwoof, 2021-04-03:
The file that didn't get updated has the latest last_modified time in the list, and after the db was deleted the last_modified time (also for the first file) change to a lower value. The second version must show the actual values on the files returned by `os.path.getmtime()` as that's being filled into an empty database and files haven't been edited. So the reason the file wasn't updated was because the getmtime from when it was first put into the db was *later* than the run after editing - I've no idea how that can happen but there isn't a clock built into the RPi so maybe there's a delay between starting up and checking what the time is from the internet connection. It also doesn't explain why the first file was updated as that also had a getmtime on the db later than the value from the os method. Very odd.

On the Apple software side - I think Apple have decided that heic should replace jpeg as the new standard so they probably have prioritized getting that fully working and moved any jpeg work to a lower priority. It is a bit annoying that heic isn't really fully implemented on every other platform yet!

jgodfrey, 2021-09-06:
I just ran into a variation of this issue. This morning, I noticed that a group of my photos contained incorrect meta-data on the frame. Looking in my photo editing software, it was wrong there too. To fix it, I had to re-import the meta-data from my original RAW files into my editing software and then export a subset of those photos back onto the frame to get the correct data in the images on the frame.

However, after doing that, the frame did not pick up the changes and continued to display the wrong meta-data. I verified that the modified times of the updated images were correctly reflected on the frame, but it was still oblivious to the changes.

I figured the containing folder's time stamp hadn't been updated by the file  modifications, so I manually updated the stamp via:

`touch <the_folder>`

... and, with that, the frame immediately picked up the changes for all of the  modified images.

So, it looks like we have a chink in our caching armor when an existing file is modified in some way. Because that operation doesn't update the containing folder's timestamp, the changes are not seen by the frame.

I'm not sure what the best solution here might be. I guess we could switch to a filesystem monitoring mechanism like [watchdog](https://pypi.org/project/watchdog/), but I don't know much about it ATM...

paddywwoof, 2021-09-06:
Actually this has reminded me that I did have a recursive crawler background job in the pi3d_demos version to do this file change checking. It's probably not very elegant but I think it was only a few lines of code https://github.com/pi3d/pi3d_demos/blob/d15f0c544d46038157d4626b3f44bf200b32e0f9/PictureFrame2020.py#L212

helgeerbe, 2021-09-07:
A Directory is considered as changed when there is any addition or deletion of file/directory inside it. If existing files/directories are just getting updates then the parent directory timestamp will not be changed.

So we could use:
- Paddy's crawler running once a day, every hour, on demand
-  touch <the_folder>
- create and delete additional dummy file in <the_folder>

I could live with just the touch command. But to make it more user friendly we could start the crawler on demand via the web interface im combination with the cache update (delete unneeded cache entries). I wouldn't do it on a time schedule, because I expect changing images is not a regular task.

helgeerbe, 2022-03-24:
Should be fixed with 2022.03.24.

closed 2022-03-24 by helgeerbe