helgeerbe/picframe #97: Location data missing in database

sapnho, 2021-04-04:
I had a look at my (digital picture frame in the living room) database with the DB Browser. There are about 1,250 photos in there. All, except for a few exceptions, have GPS coordinates. Yet about 300 of them don't have an identified location.

I was trying to identify a pattern, but couldn't find one. I wonder if anybody else has come across this.

[image]

paddywwoof, 2021-04-04:
That is strange, assuming they've all been displayed. (To start with no images have location info) Did you notice it from seeing photos on the display with no location?

Is the internet connection reliable for the picframe? If it's intermittent then I would expect the locations to be filled in eventually.

sapnho, 2021-04-04:
> That is strange, assuming they've all been displayed. (To start with no images have location info) Did you notice it from seeing photos on the display with no location?

I will pull these photos over to my Test Pi and see what happens. And yes, the playlist has been circulating for quite some time.

> Is the internet connection reliable for the picframe? If it's intermittent then I would expect the locations to be filled in eventually.

Very much so. We have 100 M/bit internet and the router is 4 meters away from the frame. Signal strength is strong.

sapnho, 2021-04-04:
Ok, so I imported the images on my Test Pi and let it run for a while. All 1250 images were recognized with the location.
Strange.

sapnho, 2021-04-04:
Is the DB actually deleted when upgrading PictureFrame?

paddywwoof, 2021-04-04:
I don't think it is. That certainly wasn't the idea.

sapnho, 2021-04-04:
Ok, good to know. Just trying to understand the root cause of the issue.

helgeerbe, 2021-04-15:
Is there anything left to do?

helgeerbe, 2022-01-05:
@sapnho Can we close this?

sapnho, 2022-01-05:
Yes.

closed 2022-01-05 by helgeerbe