helgeerbe/picframe #98: dual screen support on pi4

dadokkio, 2021-04-05:
Hello,
can picframe be used on raspberry pi 4 on two screen simultaneously? 
For example with 2 instance showing images from different folder on each screen

paddywwoof, 2021-04-05:
I don't see why two instances of python couldn't be started each pointing to a different screen. I think the processing load on the GPU (and CPU, but that's not significant) would be twice as high as for single instance.

Other people have asked similar questions and I think they have managed but I don't have two micro HDMI plugs and leads to test it.

dadokkio, 2021-04-05:
I was thinking if that was possible in configuration to keep them running as a service.. but I'll try run them manually to test it

paddywwoof, 2021-04-06:
Yes, that's what I would try first. ssh into the same Pi with two different sessions and try to start picframe on each but sending output to different monitor. See also conversation here from last year - not sure if relevant now. https://www.raspberrypi.org/forums/viewtopic.php?t=270358

helgeerbe, 2021-04-15:
Something for the wiki?