helgeerbe/picframe #99: Question regarding location elements

sapnho, 2021-04-06:
I remember from previous conversations with @paddywwoof that the structure of the Nominatim location data is not necessarily consistent. In the new PF we now have this key list

`  key_list: [
    ["tourism","amenity","isolated_dwelling"],
    ["suburb","village"],
    ["city","county"],
    ["region","state","province"],
    ["country"]]
`

Can we modify this by leaving away parts of these elements? I have the issue that sometimes the town is not displayed but the country is  - which I don't really understand how this can be?

I wonder what other people's experience with this is and if any solutions to location issues that have been found.

paddywwoof, 2021-04-06:
Good idea to build up a list of what works where. It's *very* country dependent so you might have to tweak the structure to find something that works for your pictures. I know you say in your instructions to just leave it as it is but I think, eventually, we should give people a tool to help them tune it for themselves. I will post a little python program like the one I did before that people can run to see what they get for a given lat, lon.

The way the key_list works is that it will try each list of keys until it finds one and use that, then move on to the next. i.e. you will only see "isolated_dwelling" if it looked for "amenity" and didn't find anything and it will only try looking for "amenity" if it failed to find "tourism". Basically you only ever see *one* of the categories in each list

```python
import json
import urllib.request
URL = "https://nominatim.openstreetmap.org/reverse?format=geojson&lat={}&lon={}&email={}"

lat = float(input("enter the latitude for the lookup "))
lon = float(input("enter the longitude for the lookup "))

# NB THIS WILL THROW AN ERROR UNTIL YOU REPLACE THE XXXX BELOW WITH YOUR EMAIL ADDRESS IN QUOTES
# i.e. XX.. => "me@gmail.com"
with urllib.request.urlopen(URL.format(lat, lon, XXXXXXXXXXXXX)) as req:
    data = json.loads(req.read().decode())
address = data['features'][0]['properties']['address']
for (key, val) in address.items():
    print("'{}':'{}'".format(key, val))
```
gives
```
enter the latitude for the lookup 54
enter the longitude for the lookup -2
'natural':'Deer Gallows Plain'
'road':'Embsay Pasture Road'
'suburb':'Embsay with Eastby'
'village':'Embsay'
'city':'Craven'
'county':'North Yorkshire'
'state_district':'Yorkshire and the Humber'
'state':'England'
'postcode':'BD23'
'country':'United Kingdom'
'country_code':'gb'

enter the latitude for the lookup 50.08
enter the longitude for the lookup 8.76
'house_number':'65'
'road':'Gravenbruchweg'
'suburb':'Rosenhöhe'
'city':'Offenbach am Main'
'state':'Hessen'
'postcode':'63069'
'country':'Deutschland'
'country_code':'de'

enter the latitude for the lookup 35
enter the longitude for the lookup -100
'county':'Harmon County'
'state':'Oklahoma'
'country':'United States'
'country_code':'us'

enter the latitude for the lookup 49.6696
enter the longitude for the lookup -124.927
'leisure':'Comox Bay Marina'
'road':'Breakwater Boardwalk'
'town':'Comox'
'county':'Comox Valley Regional District'
'state':'British Columbia'
'postcode':'V9M 3M2'
'country':'Canada'
'country_code':'ca'
```

sapnho, 2021-04-06:
Thanks, @paddywwoof! That thing is amazingly versatile! 

Finetuning might work but most people will have images from all over the world, won't they? But the Python script could be a great way to cater to individual preferences.

So the idea is, the let the Python script run on your /Pictures library and then change parameters to achieve the ideal outcome? That could be neat.

closed 2022-01-05 by helgeerbe