helgeerbe/picframe #100: Issue on a Pi 2

sapnho, 2021-04-13:
Just received this from reader Richard:

> I'd thought I'd touch base to find out if you have any tips on how to address a problem I have, where the fading and picture display are erratic.  I'm using a RaspPi model 2 (2011.12) with a small number of images stored on the local SD drive and I consistently have this problem.  An example might be where an image may be visible for perhaps 60 seconds (rather than the 30 I have set) and then, rather than fade into the next photo, it'll quickly display the next image for a few seconds only and then fade into the next :)  Another example is fading will take alot longer than the set 7 seconds sometimes - perhaps 30 seconds. Other times it works perfectly.  I followed your instructions to the letter so everything is fresh install, and config is only as suggested in your guide.  Anyway, any pointers on what the issue might be would be gratefully received.

Any thoughts? I don't have a Pi 2 anymore so I can't test it myself.

paddywwoof, 2021-04-13:
Not entirely sure without details of the actual pictures causing the issue, but: This sounds like the Pi struggling to process the images at the rate asked for. When you specify a frame rate of 20 and a fade time of 7s the pictureframe divides that into 140 steps of alpha change and that's the amount of change each frame. However if the frame rate is only getting up to 5fps for some images then the fade time will take 28s. The timing of new pictures is driven entirely by elapsed time from the Controller instance so that might make the fade stop or start at an odd time but shouldn't add an extra 30s to the display time. However with this being a lower power Pi it might be one of the CPU based operations that is holding up the whole process. i.e. resizing an image to one of the sizes the GPU can cope with. And/or the new mat and portrait pairs functionality are very CPU intensive so could be relevant.

So TLDR; a) Try reducing the FPS to 5 and increasing the display time to 60s to see if it is still erratic. b) Check the size of the images. If there are some wider than 1920 resize them (make backup first) then increase the FPS and reduce the delay time in stages to see if and when the problem recurs.

Whatever happens please let us know.

Paddy

sapnho, 2021-04-13:
Thanks, @paddywwoof. I'll report back from the reader.

jgodfrey, 2021-04-13:
> When you specify a frame rate of 20 and a fade time of 7s the pictureframe divides that into 140 steps of alpha change and that's the amount of change each frame. However if the frame rate is only getting up to 5fps for some images then the fade time will take 28s.

Why can't we change the per-frame transition timing to work more like a game engine, which calculates a live delta time for each frame and adjusts all motion accordingly? So, for example, if the current frame happens to have taken twice as long as expected, the alpha value (in our case) would be shifted by twice as much in that frame. So, the transition *time* becomes the consistent factor, and the *step* between frames becomes the variable - based on hardware performance?

I think it should be a relatively small change to the main looping mechanism, but might make a considerable difference in transition consistency across varying hardware.

Thoughts?

paddywwoof, 2021-04-13:
That's probably sensible. Originally there wasn't a mechanism to force all the CPU load into the "static" part of the display (and that is still an issue for the kenburns where the image moves the whole time) Having equal steps made it less obvious that there had been a large delay which would otherwise have caused a sudden change in alpha.

I will try it out and let you know.

sapnho, 2021-04-14:
> I changed the settings as suggested by Paddy and I suddenly got a clean transition between the photos. The display time was generally 60 seconds but varied slightly. I noticed the file size varied between 12mb and 25mb though. 
I next tried using a set of photos generally < 2mb and within the width requested by Paddy. I'm now seeing a consistent display time as well.
So I think it's time to upgrade the Pi. 
Thanks to you and Paddy for your help. Much appreciated.

paddywwoof, 2021-04-14:
That's useful info. It's tedious to resize images but I think 25MB is the kind of resolution required for cinema screen high resolution images! I might post an app here that will resize a set of files to an appropriate width. Generally the width needed isn't the screen width but the size specified by the GPU i.e. 768, 800, 960, 1024, 1080, 1920, 2048 with the v4 also coping with 4096

sapnho, 2021-04-14:
Yes, that what I also told Richard. The max file size for 1920x1200 (or 1080) images is 4MB if properly sized at 100% jpg quality.

closed 2021-04-15 by helgeerbe